zephyrproject-rtos/zephyr #53455: GATT: Deadlock while sending GATT notification from system workqueue thread

MariuszSkamra, 2023-01-03:
**Describe the bug**
The notifications sent from system workqueue thread cause deadlock.
The `bt_att_chan_create_pdu` is blocking operation, where timeout is set to `K_FOREVER` blocking for net_buf and `struct bt_att_tx_meta_data data` allocation. The problem is that `bt_att_tx_meta_data` is free'd from `conn.c: tx_complete_work` that is ran from System WQ as well.

bt_att_chan_create_pdu:
```
	buf = bt_l2cap_create_pdu_timeout(NULL, 0, timeout);
	if (!buf) {
		LOG_ERR("Unable to allocate buffer for op 0x%02x", op);
		return NULL;
	}

	LOG_DBG("buf %p allocated", buf);

	data = tx_meta_data_alloc(timeout);    // <- deadlock
	if (!data) {
		LOG_WRN("Unable to allocate ATT TX meta");
		net_buf_unref(buf);
		return NULL;
	}

	LOG_DBG("data %p allocated", data);
```

**To Reproduce**
Use the branch: https://github.com/zephyrproject-rtos/zephyr/commit/70b36e36169501dea3437d13ffb3aedf21e054f1

Steps to reproduce the behavior:
Flash nrf5340 network core as described here https://github.com/nrfconnect/sdk-nrf/tree/main/applications/nrf5340_audio

Build `samples/bluetooth/unicast_audio_client`
1. mkdir -p build/unicast_audio_client; cd build/unicast_audio_client
2. west build -b nrf5340dk_nrf5340_cpuapp -s ../../samples/bluetooth/unicast_audio_client/ -- -DOVERLAY_CONFIG="rtt.conf" 
3. west flash

Build `samples/bluetooth/unicast_audio_server`
1. mkdir -p build/unicast_audio_server; cd build/unicast_audio_server
2. west build -b nrf5340dk_nrf5340_cpuapp -s ../../samples/bluetooth/unicast_audio_server/ -- -DOVERLAY_CONFIG="rtt.conf" 
3. west flash

**Expected behavior**
The notification is sent

**Impact**
showstopper

**Logs and console output**
unicast_audio_server:
```
QoS: stream 0x20004718 qos 0x2000ba78
QoS: interval 10000 framing 0x00 phy 0x02 sdu 40 rtn 2 latency 10 pd 40000
QoS: stream 0x200046c8 qos 0x2000bc08
QoS: interval 10000 framing 0x00 phy 0x02 sdu 40 rtn 2 latency 10 pd 40000
QoS: stream 0x200046f0 qos 0x2000bb40
QoS: interval 10000 framing 0x00 phy 0x02 sdu 40 rtn 2 latency 10 pd 40000
[00:00:06.941,000] <dbg> bt_att: bt_att_chan_create_pdu: chan 0x20007908 op 27 size 13
[00:00:06.941,000] <dbg> bt_att: bt_att_chan_create_pdu: buf 0x2000b3dc allocated
[00:00:06.941,000] <dbg> bt_att: bt_att_chan_create_pdu: data 0x200048e4 allocated
[00:00:06.941,100] <dbg> bt_att: bt_att_chan_send: chan 0x20007908 flags 4 code 0x1b
[00:00:06.941,100] <dbg> bt_att: chan_send: code 0x1b
[00:00:06.941,300] <dbg> bt_att: bt_att_chan_create_pdu: chan 0x20007908 op 27 size 19
[00:00:06.941,300] <dbg> bt_att: bt_att_chan_create_pdu: buf 0x2000b3dc allocated
[00:00:06.941,300] <dbg> bt_att: bt_att_chan_create_pdu: data 0x200048f8 allocated
[00:00:06.941,400] <dbg> bt_att: bt_att_chan_send: chan 0x20007908 flags 4 code 0x1b
[00:00:06.941,400] <dbg> bt_att: chan_send: code 0x1b
[00:00:06.941,600] <dbg> bt_att: bt_att_chan_create_pdu: chan 0x20007908 op 27 size 19
[00:00:06.941,600] <dbg> bt_att: bt_att_chan_create_pdu: buf 0x2000b3dc allocated
[00:00:06.941,600] <dbg> bt_att: bt_att_chan_create_pdu: data 0x2000490c allocated
[00:00:06.941,600] <dbg> bt_att: bt_att_chan_send: chan 0x20007908 flags 4 code 0x1b
[00:00:06.941,600] <dbg> bt_att: chan_send: code 0x1b
[00:00:06.941,800] <dbg> bt_att: bt_att_chan_create_pdu: chan 0x20007908 op 27 size 19
[00:00:06.941,900] <dbg> bt_att: bt_att_chan_create_pdu: buf 0x2000b3dc allocated
```

Observe 2 out of 3 expected notifications are sent
```
< ACL Data TX: Handle 0 flags 0x00 dlen 24                                                                                                                                                                                                     #207 10.940400
      ATT: Handle Value Notification (0x1b) len 19
        Handle: 0x001b
          Data: 0302000010270000022800020a00409c00
        ASCS: Audio Stream Source Endpoint (Handle: 0x001b)
        ASE ID: 3
        QoS configured
          CIG ID: 0
          CIS ID: 0
          SDU interval: 10000
          Framing: Unframed
          LE PHY (0x02)
            LE 2M PHY
          Maximum SDU size: 40
          Retransmission number: 2
          Transport latency: 10 ms
          Presentation delay: 40000
< ACL Data TX: Handle 0 flags 0x00 dlen 24                                                                                                                                                                                                     #208 10.940700
      ATT: Handle Value Notification (0x1b) len 19
        Handle: 0x0015
          Data: 0102000010270000022800020a00409c00
        ASCS: Audio Stream Source Endpoint (Handle: 0x0015)
        ASE ID: 1
        QoS configured
          CIG ID: 0
          CIS ID: 0
          SDU interval: 10000
          Framing: Unframed
          LE PHY (0x02)
            LE 2M PHY
          Maximum SDU size: 40
          Retransmission number: 2
          Transport latency: 10 ms
          Presentation delay: 40000
= bt: QoS: stream 0x20004718 qos 0x2000ba78                                                                                                                                                                                                         10.939600
= bt: QoS: interval 10000 framing 0x00 phy 0x02 sdu 40 rtn 2 latency 10 pd 40000                                                                                                                                                                    10.939700
= bt: QoS: stream 0x200046c8 qos 0x2000bc08                                                                                                                                                                                                         10.939700
> HCI Event: Number of Completed Packets (0x13) plen 5                                                                                                                                                                                         #209 10.972400
        Num handles: 1
        Handle: 0
        Count: 1
> HCI Event: Number of Completed Packets (0x13) plen 5                                                                                                                                                                                         #210 10.972900
        Num handles: 1
        Handle: 0
        Count: 1
= bt: QoS: interval 10000 framing 0x00 phy 0x02 sdu 40 rtn 2 latency 10 pd 40000                                                                                                                                                                    10.939800
= bt: QoS: stream 0x200046f0 qos 0x2000bb40                                                                                                                                                                                                         10.939800
= bt: QoS: interval 10000 framing 0x00 phy 0x02 sdu 40 rtn 2 latency 10 pd 40000                                                                                                                                                                    10.939900
> HCI Event: Number of Completed Packets (0x13) plen 5                                                                                                                                                                                         #211 11.004600
        Num handles: 1
        Handle: 0
        Count: 1
```

unicast_audio_client:
```
Configuring streams
Audio Stream 0x20003544 configured
Configured sink stream[0]
Audio Stream 0x20003578 configured
Configured sink stream[1]
Audio Stream 0x200034e0 configured
Configured source stream[0]
Stream configured
Setting stream QoS
Audio Stream 0x200034e0 QoS set
Audio Stream 0x20003544 QoS set
```

**Environment (please complete the following information):**
Use the branch: https://github.com/zephyrproject-rtos/zephyr/commit/70b36e36169501dea3437d13ffb3aedf21e054f1

**Additional context**
Used with network core controller firmware provided here: https://github.com/nrfconnect/sdk-nrf/blob/main/applications/nrf5340_audio/bin/ble5-ctr-rpmsg_3310.hex

Thalley, 2023-01-03:
Try increasing `CONFIG_BT_L2CAP_TX_BUF_COUNT` and see if that helps. The QoS operation will trigger 1 notification per stream, so if `CONFIG_BT_L2CAP_TX_BUF_COUNT=2` and you have 3 streams, that might cause some issue perhaps.

hermabe, 2023-01-04:
I see three different solutions to this:
1. Add a similiar check as the one in `conn.c` to ATT and L2CAP as well so that we do not block on the system workqueue.
https://github.com/zephyrproject-rtos/zephyr/blob/58c924ec569afceaa96d8e771a91c797b978c81b/subsys/bluetooth/host/conn.c#L380-L386
2. Create a separate workqueue that the Bluetooth stack can use internally and remove all uses of the system workqueue.
3. Document that the Bluetooth stack shall not be used from the system workqueue, as it is used by the stack internally.

MariuszSkamra, 2023-01-04:
> 1. Add a similiar check as the one in `conn.c` to ATT and L2CAP as well so that we do not block on the system workqueue.

Yes, that works actually and fixes this issue.

> 2. Create a separate workqueue that the Bluetooth stack can use internally and remove all uses of the system workqueue.

I like this idea. The disadvantage of system workqueue I see is that it's exposed with public API, meaning that the user can disturb the work of Bluetooth stack internals. Having dedicated internal workqueue wold make the stack more robust.

closed 2023-01-04 by carlescufi

jori-nordic, 2023-01-09:
> Yes, that works actually and fixes this issue.
>
>
> I like this idea. The disadvantage of system workqueue I see is that it's exposed with public API, meaning that the user can disturb the work of [...]

Yeah I'd prefer the internal BT workqueue as well, but we have some memory-conscious users that wouldn't like that. 
For now solution 1. is better as you implemented.

PR #53482 (2023-01-04): Bluetooth: Fix deadlock on meta data context allocation
This fixes deadlock that happened waiting for meta data in system
workqueue.
The meta data always get freed in the system workqueue,
so if we're in the same workqueue but there are no immediate
contexts available, there's no chance we'll get one by waiting.

Fixes: #53455
commit 9148fb65 (2023-01-04): Bluetooth: att: Fix deadlock on meta data context allocation

This fixes deadlock that happened waiting for meta data in system
workqueue.
The meta data always get freed in the system workqueue,
so if we're in the same workqueue but there are no immediate
contexts available, there's no chance we'll get one by waiting.

Fixes: #53455