zephyrproject-rtos/zephyr #53458: IPv4 address autoconfiguration leaks TX packets/buffers

tatuylonen, 2023-01-03:
If I enable CONFIG_NET_IPV4_AUTO in my prj.cfg, the system will start leaking TX packets/buffers as shown by  "net mem" on the console shell (the Avail field of the "TX" row goes to zero over a couple of minutes).  Within a few minutes, the system runs out of TX packets/buffers and all network operations will fail.  I had CONFIG_NET_PKT_TX_COUNT=20, CONFIG_NET_BUF_TX_COUNT=30.  The leak also seems to happen if the application doesn't actually initiate anything network-related, even DHCP.  The problem goes away if I disable CONFIG_NET_IPV4_AUTO.  [I did also have DHCP4 enabled; I had no statically configured addresses.]

I haven't debugged this extensively and don't actually need CONFIG_NET_IPV4_AUTO, but it still took three days to figure out what was causing the network to fail in my application after initially working for a couple of minutes after boot.

I'm using a custom STM32F407ZG board using its native ethernet (CONFIG_ETH_STM32_HAL) (network aspects of the board are similar to olimex_stm32_e407).  I build using Ubuntu 22.4.

Since I don't actually need CONFIG_NET_IPV4_AUTO, this won't impact my work further, but losing packet buffers makes the whole network subsystem look bad and not ready for use (especially in combination with Zephyr apparently still answering PINGs with incorrect checksums! - cf. https://github.com/zephyrproject-rtos/zephyr/issues/7989), and made me almost ready to give up using Zephyr.  I think enabling CONFIG_NET_IPV4_AUTO is an an error Zephyr novices might easily make, and the error just makes Zephyr look bad more than being critical functionality.  [Incorrect checksums in PING replies still look very bad!]

rlubos, 2023-01-04:
There seems to be a regression related to the recent ARP rework (https://github.com/zephyrproject-rtos/zephyr/commit/771668cf91392ffba172102c6fadd68e75705a91) causing net_pkt leaks if IPv4 autoconfiguration feature is enabled. I'll provide a fix soon.

> especially in combination with Zephyr apparently still answering PINGs with incorrect checksums! - cf. https://github.com/zephyrproject-rtos/zephyr/issues/7989

Can you please elaborate on the problem you see here? The 4-year-old issue you mention was provided with a fix after two days, and on my side I see no issues with ICMP checksum calculation with current `main`.

tatuylonen, 2023-01-04:
The ping problem manifested when I tried to ping the zephyr board at its IP address from my workstation on the same local network.  Ping on the workstation was not receiving replies, but tcpdump showed reply packets arriving.  I checked with Wireshark, which said the ICMP checksum was wrong in the received echo replies.  When I saw the 4-year-old issue I stopped analyzing it further.  However, I still see the same problem when I try to ping the board.

I'm running Zephyr 3.2.0-2838-g02b83892d2da on my board and Ubuntu 22.04 on my workstation.  TCP/IP traffic works on the board and DHCP, TCP connect, HTTP GET/POST work fine, so I assume hardware works.  Custom STM32F407ZG board.

thedjnK, 2023-01-04:
> The ping problem manifested when I tried to ping the zephyr board at its IP address from my workstation on the same local network. Ping on the workstation was not receiving replies

Have used nordic boards and stm32 board with non-st MACs and I have not seen this issue, pings have worked fine for me.

rlubos, 2023-01-05:
I've submitted a fix for the net pkt leak in IPv4 autoconf module: https://github.com/zephyrproject-rtos/zephyr/pull/53525.

The ICMP checksum calculation error I simply cannot reproduce, even with commit you reference.

nordicjm, 2023-01-05:
Just checked an stm32 (nucleo f746zg) using the ST MAC with ping, even with autoconfig enabled, and it works fine, checksum is valid.

PR #53525 (2023-01-05): net: ipv4: Fix packet leak with IPv4 autoconf
The IPv4 autoconfiguration feature relies on the fact, that autoconf ARP packets are always prepared by the ARP module. After recent ARP refactoring though that could no longer be the case due to packet queueing mechanism. This could lead to net pkt leaks in the autoconf module.

Fix this by skipping the pending packet queue for autoconf packets. Since for autoconf ARP requests there's no really a pending packet to queue, it can be safely avoided. This results in the ARP request being always sent for the autoconf case, preventing the packet leak.

 [... cut]
closed 2023-01-05 by fabiobaltieri