zephyrproject-rtos/zephyr #53466: Nucleo F413ZH CAN bus support

tomstix, 2023-01-03:
Altough the STM32F413 supports up to 3 CAN buses, they aren't added to the nucleo_f413zh board's device tree so I tried adding it but can't get it to work.
First of, I am very new to Zephyr so there might be things I didn't understand fully/correctly.

I added the following to the stm32f412.dtsi under soc (as the stm32f413.dtsi includes this and the f412 has 2 CAN buses with the same memory mapping):

```
can1: can@40006400 {
	compatible = "st,stm32-can";
	reg = <0x40006400 0x400>;
	interrupts = <19 0>, <20 0>, <21 0>, <22 0>;
	interrupt-names = "TX", "RX0", "RX1", "SCE";
	clocks = <&rcc STM32_CLOCK_BUS_APB1 0x02000000>;
	status = "disabled";
	sjw = <1>;
	sample-point = <875>;
};

can2: can@40006800 {
	compatible = "st,stm32-can";
	reg = <0x40006800 0x400>;
	interrupts = <63 0>, <64 0>, <65 0>, <66 0>;
	interrupt-names = "TX", "RX0", "RX1", "SCE";
	/* also enabling clock for can1 (master instance) */
	clocks = <&rcc STM32_CLOCK_BUS_APB1 0x06000000>;
	master-can-reg = <0x40006400>;
	status = "disabled";
	sjw = <1>;
	sample-point = <875>;
};
```

Then I added to nucleo_f413zh.dts (I chose the pin configuration to not interfere with what is already set):
```
&can1 {
	pinctrl-0 = <&can1_rx_pd0 &can1_tx_pd1>;
	pinctrl-names = "default";
	bus-speed = <250000>;
	status = "okay";
};
```
As well as under chosen:
`zephyr,canbus = &can1;`

And in nucleo_f413zh.yaml I added `  - can` under supported:.

With this, `west build -b nucleo_f413zh .\samples\drivers\can\babbling` compiles and uploads fine, but it fails to transmit  any messages with the follwing output:
```
21:23:07.606 -> *** Booting Zephyr OS build zephyr-v3.2.0-2914-g82da499d5d93 ***
21:23:07.606 -> babbling on can@40006400 with standard (11-bit) CAN ID 0x010, RTR 0, CAN-FD 0
21:23:07.606 -> abort by pressing User button
21:23:07.606 -> failed to enqueue CAN frame (err -114)
21:23:07.606 -> failed to enqueue CAN frame (err -114)
21:23:07.639 -> failed to enqueue CAN frame (err -11)
```

Is there something I've forgotten/overseen?

henrikbrixandersen, 2023-01-04:
A better test would be `tests/drivers/can/api/`. Please post the output of running that test.

tomstix, 2023-01-04:
Okay, nevermind. The problem was actually my USB-CAN adapter, which seems not to work properly and didn't ACK the CAN frames. The `tests/drivers/can/api/` runs fine and when I connect a different USB-CAN adapter the sample also works.
Thank you @henrikbrixandersen for the hint and sorry for the inconvenience!

Here is the test output of someone is interested:
```
09:58:28.509 -> *** Booting Zephyr OS build zephyr-v3.2.0-2914-g82da499d5d93 ***
09:58:28.509 -> Running TESTSUITE can_api
09:58:28.543 -> ===================================================================
09:58:28.543 -> START - test_add_filter
09:58:28.543 ->  PASS - test_add_filter in 0.001 seconds
09:58:28.543 -> ===================================================================
09:58:28.543 -> START - test_filters_added_while_stopped
09:58:28.543 ->  PASS - test_filters_added_while_stopped in 0.001 seconds
09:58:28.543 -> ===================================================================
09:58:28.577 -> START - test_filters_preserved_through_bitrate_change
09:58:28.577 ->  PASS - test_filters_preserved_through_bitrate_change in 0.002 seconds
09:58:28.577 -> ===================================================================
09:58:28.577 -> START - test_filters_preserved_through_mode_change
09:58:28.577 ->  PASS - test_filters_preserved_through_mode_change in 0.003 seconds
09:58:28.577 -> ===================================================================
09:58:28.611 -> START - test_get_capabilities
09:58:28.611 ->  PASS - test_get_capabilities in 0.001 seconds
09:58:28.611 -> ===================================================================
09:58:28.611 -> START - test_get_core_clock
09:58:28.611 ->  PASS - test_get_core_clock in 0.001 seconds
09:58:28.611 -> ===================================================================
09:58:28.611 -> START - test_get_state
09:58:28.611 ->  PASS - test_get_state in 0.001 seconds
09:58:28.645 -> ===================================================================
09:58:28.645 -> START - test_max_ext_filters
09:58:28.645 -> W: No free filter left
09:58:28.645 ->  PASS - test_max_ext_filters in 0.003 seconds
09:58:28.645 -> ===================================================================
09:58:28.645 -> START - test_max_std_filters
09:58:28.645 -> W: No free filter left
09:58:28.645 ->  PASS - test_max_std_filters in 0.003 seconds
09:58:28.645 -> ===================================================================
09:58:28.680 -> START - test_receive_timeout
09:58:28.783 ->  PASS - test_receive_timeout in 0.101 seconds
09:58:28.783 -> ===================================================================
09:58:28.783 -> START - test_recover
09:58:28.783 ->  PASS - test_recover in 0.001 seconds
09:58:28.783 -> ===================================================================
09:58:28.783 -> START - test_recover_while_stopped
09:58:28.783 ->  PASS - test_recover_while_stopped in 0.001 seconds
09:58:28.783 -> ===================================================================
09:58:28.817 -> START - test_send_and_forget
09:58:28.817 ->  PASS - test_send_and_forget in 0.002 seconds
09:58:28.817 -> ===================================================================
09:58:28.817 -> START - test_send_callback
09:58:28.817 ->  PASS - test_send_callback in 0.002 seconds
09:58:28.817 -> ===================================================================
09:58:28.817 -> START - test_send_fd_format
09:58:28.817 -> E: unsupported CAN frame flags 0x04
09:58:28.851 ->  PASS - test_send_fd_format in 0.004 seconds
09:58:28.851 -> ===================================================================
09:58:28.851 -> START - test_send_invalid_dlc
09:58:28.851 -> E: DLC of 9 exceeds maximum (8)
09:58:28.851 ->  PASS - test_send_invalid_dlc in 0.003 seconds
09:58:28.851 -> ===================================================================
09:58:28.851 -> START - test_send_receive_ext_id
09:58:28.885 ->  PASS - test_send_receive_ext_id in 0.004 seconds
09:58:28.885 -> ===================================================================
09:58:28.885 -> START - test_send_receive_ext_id_masked
09:58:28.885 ->  PASS - test_send_receive_ext_id_masked in 0.004 seconds
09:58:28.885 -> ===================================================================
09:58:28.885 -> START - test_send_receive_ext_id_rtr
09:58:29.091 ->  PASS - test_send_receive_ext_id_rtr in 0.204 seconds
09:58:29.126 -> ===================================================================
09:58:29.126 -> START - test_send_receive_msgq
[... 61 lines cut ...]
09:58:29.743 ->  - PASS - [can_api.test_send_invalid_dlc] duration = 0.003 seconds
09:58:29.743 ->  - PASS - [can_api.test_send_receive_ext_id] duration = 0.004 seconds
09:58:29.743 ->  - PASS - [can_api.test_send_receive_ext_id_masked] duration = 0.004 seconds
09:58:29.743 ->  - PASS - [can_api.test_send_receive_ext_id_rtr] duration = 0.204 seconds
09:58:29.743 ->  - PASS - [can_api.test_send_receive_msgq] duration = 0.011 seconds
09:58:29.778 ->  - PASS - [can_api.test_send_receive_std_id] duration = 0.004 seconds
09:58:29.778 ->  - PASS - [can_api.test_send_receive_std_id_masked] duration = 0.004 seconds
09:58:29.778 ->  - PASS - [can_api.test_send_receive_std_id_rtr] duration = 0.204 seconds
09:58:29.778 ->  - PASS - [can_api.test_send_receive_wrong_id] duration = 0.102 seconds
09:58:29.778 ->  - PASS - [can_api.test_send_while_stopped] duration = 0.001 seconds
[... cut ...]
09:58:29.812 ->  - PASS - [can_api.test_set_state_change_callback] duration = 0.001 seconds
09:58:29.812 ->  - PASS - [can_api.test_set_timing_while_started] duration = 0.001 seconds
09:58:29.846 ->  - PASS - [can_api.test_start_while_started] duration = 0.001 seconds
09:58:29.846 ->  - PASS - [can_api.test_stop_while_stopped] duration = 0.001 seconds
09:58:29.846 -> 
09:58:29.846 -> ------ TESTSUITE SUMMARY END ------
09:58:29.846 -> 
09:58:29.846 -> ===================================================================
09:58:29.846 -> PROJECT EXECUTION SUCCESSFUL

```

closed 2023-01-04 by tomstix