zephyrproject-rtos/zephyr #53468: STM32 single-wire UART not working when poll-out more than 1 char

cooked, 2023-01-03:
**Describe the bug**
When the STM32 single-wire UART sample is modified such that to poll out more than just one byte, no characters are received by the polling in side.

I'm currently testing on a ST Nucleo F103RB using USART1 (sending) and USART3 (receiving).

For reference, I tested the [default single-wire sample](https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/drivers/uart/stm32/single_wire) (1 char sent and received) on the Nucleo F103RB and it works just fine. 

**To Reproduce**
Build the sample using the code for **main.c** and **nucleo_f103rb.overlay** in the Additional Context below

**Expected behavior**
Data sent from USART1 shall be received at USART3

**Impact**
Showstopper

**Environment (please complete the following information):**
 - OS: Linux
 - Zephyr SDK 0.15.0
 - Zephyr 3.2.0

**Additional context**

main.c
```c
#include <zephyr/kernel.h>
#include <zephyr/device.h>
#include <zephyr/devicetree.h>
#include <zephyr/drivers/uart.h>

#define UART_NODE1 DT_ALIAS(uart1)
#define UART_NODE2 DT_ALIAS(uart2)

const struct device *const sl_uart1 = DEVICE_DT_GET(UART_NODE1);
const struct device *const sl_uart2 = DEVICE_DT_GET(UART_NODE2);

void main(void)
{
	unsigned char recv;

	if (!device_is_ready(sl_uart1) || !device_is_ready(sl_uart2)) {
		printk("uart devices not ready\n");
		return;
	}

	while (true) {

		uint8_t i;

		// TX
		//uart_poll_out(sl_uart1, 'c');
		unsigned char tx_buf[4] = { 0x0A, 0x0B, 0x0C, 0x0D };
		for(i=0; i<sizeof(tx_buf); i++) {
			uart_poll_out(sl_uart1, tx_buf[i]);
		}

		/* give the uarts some time to get the data through */
		k_msleep(50);

		int ret = uart_poll_in(sl_uart2, &recv);
		if (ret < 0) {
			printk("Receiving failed. Error: %d\n", ret);
		} else {
			printk("Received %c\n", recv);
		}

		// RX
		/*i=0;
		while(i<4) {
			ret = uart_poll_in(sl_uart2, &recv[i]);
			if( ret < 0 ) {
				k_usleep(10);
				//printk("err\n");
			} else {
				printk(" %02X ", &recv[i]);
				i++;
			}
		}
		printk("\n");*/

		k_sleep(K_MSEC(2000));
	}
}
```

nucleo_f103rb.overlay
```
/ {
	aliases {
	   uart1 = &usart1;
	   uart2 = &usart3;
   };
};

&usart1 {
	single-wire;
	pinctrl-0 = <&usart1_tx_pa9>;
};

&usart2 {
	// !!! default ST-LINK / debug port (zephyr console),
	// do not use here
};

&usart3 {
	single-wire;
	pinctrl-0 = <&usart3_tx_pb10>;
	status = "okay";
};
```

cooked, 2023-01-04:
Forgot to mention that I had to add an external pull-up 10k resistor to my  "loopback fixture" (to use the same jargon used in the existing sample docs) for my test to work, according to what stated in the [F103](https://www.st.com/resource/en/reference_manual/cd00171190-stm32f101xx-stm32f102xx-stm32f103xx-stm32f105xx-and-stm32f107xx-advanced-arm-based-32-bit-mcus-stmicroelectronics.pdf) reference manual sec. 27.3.10

gautierg-st, 2023-01-05:
I tried your code and I can confirm that no data is received in your case.
I pushed things a bit further to try to understand what is happening.

First thing I saw is that when sending two bytes in a row, an overrun error occurs on both uart. It can be checked by calling the uart_err_check function after poll_out.
In your case, after the second poll_out, the overrun error bits of both uart are raised, and never cleared, so it is stuck in error.

In any case, since we are using polling functions without dma or interrupts, writing 4 bytes in a row necessarily results in lost data, even with your commented code, since the uart does not have access to anything to store the received data.

cooked, 2023-01-22:
@gautierg-st I took some time to go through the docs and run more tests.
I was familiar with the mcu RM and knew about the pins being connected together. I assumed the chip logic, when put in single-wire, would prevent local echo of characters sent at register level, which I understand from you it's not the case.

with that out of the way, and since half-duplex is a mandatory requirement, I perfected the polling example I've got to prevent overrun readings all characters sent out. doing so I can poll multiple chars on a command-response pattern no problem.

here's the snippet to do that:

`
        // command
	int i;
	for(i=0; i<N_CMD; i++) {
		uart_poll_out(cfg->uart_dev, tx_buf[i]);
		while( uart_poll_in(cfg->uart_dev, &buf)) {
		}
	}

	// response
	for(i=0; i<N_RSP; i++) {
		while( uart_poll_in(cfg->uart_dev, &data->rd_data[i])) {
		}
	}
`

closed 2023-01-22 by cooked