zephyrproject-rtos/zephyr #53470: Unable to build using Adafruit Trinket M0

Dark1886, 2023-01-04:
**Describe the bug**
The DTSI file for the Adafruit Trinket M0 references "#include <dt-bindings/pinctrl/samd21-da1gXabcd-pinctrl.h>" but this file does not seem to exist anywhere in the zephyr repo. 

**To Reproduce**
Steps to reproduce the behavior:
Building any software while attempting to use the stock Trinket M0 device tree. 

**Expected behavior**
Code compile without a Device tree error

**Impact**
Inability to build without finding some other workaround. 

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

\`\`\`
blake@DesktopPC:~/shredlink-workspace/rtos/zephyr/tests/drivers/i2c/i2c_api$ west build -b arduino_zero -p always
-- west build: making build dir /home/blake/shredlink-workspace/rtos/zephyr/tests/drivers/i2c/i2c_api/build pristine
-- west build: generating a build system
Including boilerplate (Zephyr base): /home/blake/shredlink-workspace/rtos/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/blake/shredlink-workspace/rtos/zephyr/tests/drivers/i2c/i2c_api
-- Zephyr version: 3.0.0 (/home/blake/shredlink-workspace/rtos/zephyr)
-- Found Python3: /usr/bin/python3.10 (found suitable exact version "3.10.6") found components: Interpreter
-- Found west (found suitable version "0.14.0", minimum required is "0.7.1")
-- Board: arduino_zero
-- Cache files will be written to: /home/blake/.cache/zephyr
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.15.2 (/home/blake/zephyr-sdk-0.15.2)
-- Found dtc: /home/blake/zephyr-sdk-0.15.2/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
-- Found toolchain: zephyr 0.15.2 (/home/blake/zephyr-sdk-0.15.2)
-- Found BOARD.dts: /home/blake/shredlink-workspace/rtos/zephyr/boards/arm/arduino_zero/arduino_zero.dts
In file included from /home/blake/shredlink-workspace/rtos/zephyr/boards/arm/arduino_zero/arduino_zero.dts:9,
                 from <command-line>:
/home/blake/shredlink-workspace/rtos/zephyr/boards/arm/arduino_zero/arduino_zero-pinctrl.dtsi:6:10: fatal error: dt-bindings/pinctrl/samd21-da1gXabcd-pinctrl.h: No such file or directory
    6 | #include <dt-bindings/pinctrl/samd21-da1gXabcd-pinctrl.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
CMake Error at /home/blake/shredlink-workspace/rtos/zephyr/cmake/dts.cmake:166 (message):
  command failed with return code: 1
Call Stack (most recent call first):
  /home/blake/shredlink-workspace/rtos/zephyr/cmake/app/boilerplate.cmake:542 (include)
  /home/blake/shredlink-workspace/rtos/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
  /home/blake/shredlink-workspace/rtos/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:35 (include_boilerplate)
  CMakeLists.txt:4 (find_package)

\`\`\`

**Environment (please complete the following information):**
 - OS: Windows
 - Zephyr SDK
 - Commit

henrikbrixandersen, 2023-01-04:
Did you run `west update`?

nandojve, 2023-01-04:
I noted that you are using Zephyr 3.0.0. Since Atmel is driven by community there is no regular backport from newer additions. I recommend you sync to latest Zephyr and Zephyr SDK versions. The files that you want are at hal_atmel.

```
$ west build -b arduino_zero samples/hello_world

-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /home/gfbudke/zephyr/mainline/zephyr/samples/hello_world
-- CMake version: 3.22.1
-- Found Python3: /home/gfbudke/zephyr/mainline/zephyr/.venv-3.10/bin/python3.10 (found suitable exact version "3.10.9") found components: Interpreter 
-- Cache files will be written to: /home/gfbudke/.cache/zephyr
-- Zephyr version: 3.2.99 (/home/gfbudke/zephyr/mainline/zephyr)
-- Found west (found suitable version "0.14.0", minimum required is "0.7.1")
-- Board: arduino_zero
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.15.2 (/home/gfbudke/zephyr-sdk-0.15.2)
-- Found toolchain: zephyr 0.15.2 (/home/gfbudke/zephyr-sdk-0.15.2)
-- Found Dtc: /home/gfbudke/zephyr-sdk-0.15.2/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /home/gfbudke/zephyr/mainline/zephyr/boards/arm/arduino_zero/arduino_zero.dts
-- Generated zephyr.dts: /home/gfbudke/zephyr/mainline/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/gfbudke/zephyr/mainline/zephyr/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /home/gfbudke/zephyr/mainline/zephyr/build/zephyr/dts.cmake
Parsing /home/gfbudke/zephyr/mainline/zephyr/Kconfig
Loaded configuration '/home/gfbudke/zephyr/mainline/zephyr/boards/arm/arduino_zero/arduino_zero_defconfig'
Merged configuration '/home/gfbudke/zephyr/mainline/zephyr/samples/hello_world/prj.conf'
Configuration saved to '/home/gfbudke/zephyr/mainline/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/gfbudke/zephyr/mainline/zephyr/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 12.1.0
-- The CXX compiler identification is GNU 12.1.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/gfbudke/zephyr-sdk-0.15.2/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Configuring done
-- Generating done
-- Build files have been written to: /home/gfbudke/zephyr/mainline/zephyr/build
-- west build: building application
[1/138] Preparing syscall dependency handling

[3/138] Generating include/generated/version.h
-- Zephyr version: 3.2.99 (/home/gfbudke/zephyr/mainline/zephyr), build: zephyr-v3.2.0-3005-g9ed7f055b838
[128/138] Linking C executable zephyr/zephyr_pre0.elf

[132/138] Linking C executable zephyr/zephyr_pre1.elf

[138/138] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:        9570 B       256 KB      3.65%
             RAM:        3720 B        32 KB     11.35%
        IDT_LIST:          0 GB         2 KB      0.00%

```

closed 2023-01-04 by nandojve