zephyrproject-rtos/zephyr #53487: west: flash: stm32cubeprogrammer: Reset command line argument is wrong

nordicjm, 2023-01-04:
**Describe the bug**
The tests for the stm32cubeprogrammer control reset by specifying `--reset <mode>`, however this is invalid, the actual argument is `--reset-mode <mode>` therefore one can assume that the tests are not properly checking this argument as if they were, the tests would be failing. @erwango I will leave it to you to decide if you think this is a deficiency in the tests that need adding, however this bug is just for fixing the argument itself (which is failing due to a PR I have)

**Expected behavior**
Command line argument should be correct (this bug)
Tests should actually validate the reset mode (not part of this bug)

**Impact**
Showstopper

**Environment (please complete the following information):**
 - Commit SHA or Version used: 58c924ec569afceaa96d8e771a91c797b978c81b

erwango, 2023-01-04:
> The tests for the stm32cubeprogrammer control reset by specifying `--reset <mode>`

Can you verify ?

Here is a test sample (skipping parameters that are not of interest for current issue)
```
        "port": "swd",
        "frequency": None,
        "reset_mode": "core",
        [...]
        "calls": [
            [
               [...]
                "--connect",
                "port=swd reset=Crst",
                [...]
            ],
        ],
```

So test is configuring `"reset_mode": "core",` as west runner command parameter.
Then it verifies it is correctly translated into `reset=Crst`, which is the correct way to provide arguments to `STM32_Programmer_CLI`.

However, I can see that all board descriptions which provide stm32cubeprogrammer runner configuration are put as:
```
board_runner_args(stm32cubeprogrammer "--port=swd" "--reset=hw")
```
which seems indeed incorrect at first sight.

Though, I've just made some tests and it seems to be a python courtesy to accept shorten parameter names:
- `--reset-mode=hw`, `--reset=hw` `--reset-m=hw`, `--res=hw`: all are seen correct by python and allows to generate expected `-connect 'reset=HWrst'` configuration
- `--resetmode` generates an error

So: 
- STM32CubeProgrammer runner test looks correct to me.
- Even if `reset` were used instead of `reset-mode`, it would not be strictly incorrect.

Did I miss something ?

nordicjm, 2023-01-04:
@mbolivar this seems like a bug, why is python allowing the command line arguments to be shortened?

nordicjm, 2023-01-04:
Seems you are right, this flag is enabled by default: https://docs.python.org/3/library/argparse.html#allow-abbrev
I will submit an issue and patch to disable it globally, the mdb test case also falls over when it's disabled.

erwango, 2023-01-05:
@nordicjm Closing as this is not the bug you're looking for.

closed 2023-01-05 by erwango