zephyrproject-rtos/zephyr #53495: zephyr-wide: python: argparse default configuration allows shortened command arguments

nordicjm, 2023-01-04:
**Describe the bug**
This seems to be a whole zephyr-wide issue with any python code making use of argparse, this library decided that defaulting to allowing shortened command line arguments was a good idea, unlike traditional commands where you have a short and long argument e.g. `-h` and `--help`, this python library will, assuming you have a `--help` command accept this as `--h`, `--he`, `--hel` and `--help`. This is a pretty huge oversight as if a command exists, is wrongly called with a short argument and then a new command is added which uses the shortened argument, this will silently change the whole behaviour of the script, and indeed 2 whole test scripts for west flash commands start failing when this option is disabled and one starts failing because a new command has been added which the test script was wrongly using for the command previous.

Therefore, this is a tree-wide change request for fixing this issue so that argparse is set to not allow it to generate its own shortened commands, when commands are added they can manually specify the shortened command to allow alongside the long argument.

**Expected behavior**
Shortened arguments should not be accepted

**Impact**
Showstopper, this is silently hiding bugs and fault in the code and test scripts.

**Logs and console output**
Example after disabling feature for one of the "west flash" runners:
```
tests/test_mdb.py FFF.......                                                                                                  [ 71%]

============================================================= FAILURES ==============================================================
____________________________________________________ test_flash_nsim[test_case0] ____________________________________________________

require = <MagicMock name='require' id='140064864486976'>, cc = <MagicMock name='popen_ignore_int' id='140064867421152'>
t = <MagicMock name='sleep' id='140064865197200'>, gcp = <MagicMock name='get_cld_pid' id='140064864890752'>
test_case = {'i': ['--cores=1', '--nsim=test_nsim.args'], 'o': ['mdb', '-nooptions', '-nogoifmain', '-toggle=include_local_symbols=1', '-nsim', '@/test/zephyr/boards/test-arch/test-board/support/test_nsim.args', ...]}
mdb_nsim = <function mdb.<locals>._factory at 0x7f6364cef490>

    @pytest.mark.parametrize('test_case', TEST_NSIM_FLASH_CASES)
    @patch('runners.mdb.get_cld_pid', return_value=(False, -1))
    @patch('time.sleep', return_value=None)
    @patch('runners.mdb.MdbNsimBinaryRunner.popen_ignore_int')
    @patch('runners.core.ZephyrBinaryRunner.require', side_effect=require_patch)
    def test_flash_nsim(require, cc, t, gcp, test_case, mdb_nsim):
>       mdb_nsim(test_case['i']).run('flash')

/tmp/aa/zephyr/scripts/west_commands/tests/test_mdb.py:157: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
/tmp/aa/zephyr/scripts/west_commands/tests/test_mdb.py:126: in _factory
    arg_namespace = parser.parse_args(args)
/usr/lib/python3.10/argparse.py:1829: in parse_args
    self.error(msg % ' '.join(argv))
/usr/lib/python3.10/argparse.py:2587: in error
    self.exit(2, _('%(prog)s: error: %(message)s\n') % args)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

self = ArgumentParser(prog='__main__.py', usage=None, description=None, formatter_class=<class 'argparse.HelpFormatter'>, conflict_handler='error', add_help=True)
status = 2, message = '__main__.py: error: unrecognized arguments: --nsim=test_nsim.args\n'

    def exit(self, status=0, message=None):
        if message:
            self._print_message(message, _sys.stderr)
>       _sys.exit(status)
E       SystemExit: 2

/usr/lib/python3.10/argparse.py:2574: SystemExit
------------------------------------------------------- Captured stderr call --------------------------------------------------------
usage: __main__.py [-h] [--elf-file FILE] [--hex-file FILE] [--bin-file FILE]
                   [--cores CORES] [--nsim_args NSIM_ARGS]
__main__.py: error: unrecognized arguments: --nsim=test_nsim.args

..
```

**Environment (please complete the following information):**
 - Commit SHA or Version used: 58c924ec569afceaa96d8e771a91c797b978c81b

nordicjm, 2023-01-04:
Default description: https://docs.python.org/3/library/argparse.html#allow-abbrev

sjg20, 2023-01-06:
How do we still allow abbreviations on the command line?

nordicjm, 2023-01-09:
> How do we still allow abbreviations on the command line?

You would add a long and short argument for a command when adding the command, e.g. for `-h` and `--help` it would look similar to: `parser.add_argument("-h", "--help", help="Show help text")`. Any sort of automatic abbreviation is prohibited.

PR #53498 (2023-01-26): treewide: Disable automatic argparse argument shortening
Disables allowing the python argparse library from automatically shortening command line arguments, this prevents issues whereby a new command is added and code that wrongly uses the shortened command of an existing argument which is the same as the new command being added will silently change script behaviour.

Fixes #53495

- [X] Scripts updated
- [X] Script invocations updated
- [x] Release notes added
- [X] RFC treewide change email sent
- [X] Architecture working group discussed
- [X] pylint check added to ensure additional python c [... cut]
closed 2023-01-26 by stephanosio