zephyrproject-rtos/zephyr #53505: Some device tree integers may be signed or unsigned depending on their value

marc-hb, 2023-01-04:
**Describe the bug**

Among other integer insanities, the sign of literals in C depends on their value _and_ on the C version used, the compiler used and probably the phase of the moon too.

As luck has it, some addresses like the one added in commit d76419973a081 fall between 2^31 and 2^32 and can end up being signed in some conditions.

According to https://github.com/devicetree-org/devicetree-specification/releases/tag/v0.4-rc1 has u32 and u64 but no signed type. https://elinux.org/Device_Tree_Mysteries#Signed_Property_Values and notably http://lists.infradead.org/pipermail/linux-arm-kernel/2013-April/159682.html explain how to indirectly define signed values using integer _expressions_, e.g.: `(0 - 5)`.

The demo in #53467 has a hack that appends `u` and silences this warning. An earlier and naive attempted to add a `u` suffix unconditionally but that breaks linker scripts.

https://faultlore.com/blah/rust-layouts-and-abis/#the-c-integer-hierarchy
https://faultlore.com/blah/c-isnt-a-language/

**Impact**

It does not seem to cause any real issue so far but the static analyzer "sparse" is catching this and complaining about it. Because that register is used several times a .h file, sparse logs end up being spammed 1200 times.

Also, integer complexities and "cast abuse" have been known to cause hard to debug issues and vulnerabilities so maybe better safe than sorry.

**Expected behavior**

Device tree integers are always unsigned.

**To Reproduce**

To reproduce the 1200 sparse warnings:

git clone https://github.com/thesofproject/sof/commits/f32b6ac0fea3 in a new `west` workspace, initialize it and update it and then run sparse like this:

```
./scripts/xtensa-build-zephyr.py  --pristine --cmake-args=-DSPARSE=y mtl
```

You must also install sparse from https://github.com/marc-hb/sparse/commits/main and uninstall `ccache`. More details in https://github.com/thesofproject/sof/actions/runs/3831594561/jobs/6520904053

Uninstall ccache and `--pristine` are required because sparse currently breaks the incremental build in various obscure ways, hopefully fixed by #52671.

**Logs and console output**

https://github.com/thesofproject/sof/actions/runs/3831594561/jobs/6520904053

**Environment (please complete the following information):**

sparse static analyzer on Linux.

mbolivar-nordic, 2023-01-06:
> It does not seem to cause any real issue so far but the static analyzer "sparse" is catching this and complaining about it.

Gonna mark low, then. I don't think the juice is worth the squeeze fixing this right now given people workloads.

marc-hb, 2023-01-06:
You just reminded me this checkpatch warning: "A patch subject line should describe the change not the tool that found it"

marc-hb, 2023-03-12:
Still spamming sparse output massively:
https://github.com/thesofproject/sof/actions/runs/4389646297/jobs/7687421819

lyakh, 2023-04-17:
Having too many sparse warnings has a real negative effect on our ability to use sparse output effectively: sparse stops producing warnings after reaching a certain umber of them for each individual file. It prints a message "warning: too many warnings" and stops printing any further one for that file. Currently we have large numbers of "dubious x & !y" and "Variable length array is used." sparse warnings in SOF builds, both generated by logging calls like `comp_info()` and those often exceed that threshold. That leads to warnings that we are actually interested in like "different address spaces" being lost.

marc-hb, 2023-04-17:
Raising priority based on @lyakh's last comment.

@galak is this something you or someone else could take a look at? (@mbolivar-nordic has no time for this right now)

marc-hb, 2023-06-21:
Still happening https://github.com/thesofproject/sof/actions/runs/5324802896/jobs/9644627642?pr=7838

```
/zep_workspace/zephyr/soc/xtensa/intel_adsp/common/include/cpu_init.h:100:15: warning: decimal constant 3187671040 is between LONG_MAX and ULONG_MAX. For C99 that means long long, C90 compilers are very likely to produce unsigned long (and a warning) here
/zep_workspace/zephyr/soc/xtensa/intel_adsp/cavs/sram.c:114:24: warning: decimal constant 3187671040 is between LONG_MAX and ULONG_MAX. For C99 that means long long, C90 compilers are very likely to produce unsigned long (and a warning) here
/zep_workspace/zephyr/soc/xtensa/intel_adsp/cavs/sram.c:146:24: warning: decimal constant 3196059648 is between LONG_MAX and ULONG_MAX. For C99 that means long long, C90 compilers are very likely to produce unsigned long (and a warning) here
```

marc-hb, 2023-09-10:
Still happening https://github.com/thesofproject/sof/actions/runs/6134027541/job/16646452352

carlescufi, 2023-10-24:
Architecture WG:

- @gmarull suggest an approach like here: https://github.com/zephyrproject-rtos/zephyr/pull/56684
- @henrikbrixandersen remarks that the DT spec only supports u32 and u64 natively
- @gmarull points out that Linux has accessors for signed values: https://elixir.bootlin.com/linux/latest/source/include/linux/of.h#L1388
- @henrikbrixandersen points out that our current macros are "wrong". Eg. `DT_REG_ADDR` does not add a `U`. One option would be to add _U and _S macros or to change the existing ones to append `U`
- @galak and @mbolivar-nordic what's your take?

galak, 2023-10-24:
I would suggest something along the lines of what @gmarull suggested.  Introduce some macro variants, I'm assuming this is really only needed for DT_*_REG_* type macros.

marc-hb, 2023-12-26:
Still happening, still the most verbose "sparse" warning by far:
https://github.com/thesofproject/sof/actions/runs/7325312100/job/19949765369

stephanosio, 2024-02-20:
[We have encountered this issue in the main CI while compiling for C++98 (i.e. C90), now in the GCC itself](https://github.com/zephyrproject-rtos/zephyr/pull/69226#issuecomment-1954195768):

```
In file included from /__w/zephyr/zephyr/include/zephyr/arch/arm/arch.h:50,
                 from /__w/zephyr/zephyr/include/zephyr/arch/cpu.h:19,
                 from /__w/zephyr/zephyr/include/zephyr/kernel_includes.h:37,
                 from /__w/zephyr/zephyr/include/zephyr/kernel.h:17,
                 from /__w/zephyr/zephyr/tests/lib/cpp/cxx/src/main.cpp:18:
/__w/zephyr/zephyr/include/zephyr/arch/arm/cortex_a_r/timer.h:40:1: error: this decimal constant is unsigned only in ISO C90 [-Werror]
   40 | DEVICE_MMIO_TOPLEVEL_STATIC(timer_regs, ARM_TIMER_NODE
```

Given that C++98/C90 support is not too absurd, we will need to resolve this one way or another.

marc-hb, 2024-04-22:
Still  responsible for the most spam in sparse analysis, by far:
https://github.com/thesofproject/sof/actions/runs/8776716191/job/24080631356

swift-tk, 2024-09-21:
Does this still happens if the generated macro value is already in hex form?

fabiobaltieri, 2024-10-02:
Hey @marc-hb, https://github.com/zephyrproject-rtos/zephyr/pull/78766 should have fixed the

`/zep_workspace/sof/xtos/include/sof/lib/mailbox.h:84:72: warning: decimal constant 2684485632 is between LONG_MAX and ULONG_MAX. For C99 that means long long, C90 compilers are very likely to produce unsigned long (and a warning) here`

right? Can we call this done (I'm doing the SIZE macro as well btw)

marc-hb, 2024-10-02:
Thanks for the fix! I haven't tested it. We'll know as soon as SOF uprevs Zephyr. This happens at least every couple weeks.

fabiobaltieri, 2024-10-02:
Good stuff let's check it in a couple weeks then.

marc-hb, 2024-10-02:
It didn't take that long so I updated Zephyr and ran the check locally. It does get rid of that sparse warning, thanks!

This gets the sparse build log from 4200 lines down to 1700 lines, great job!

PS: now the most common (spurious?) warning is by far `warning: symbol 'X' was not declared. Should it be static?`

closed 2024-10-02 by marc-hb

fabiobaltieri, 2024-10-02:
> This gets the sparse build log from 4200 lines down to 1700 lines, great job!

Ah! Hurray? :-)