zephyrproject-rtos/zephyr #53506: Use recommended word separator (`-`) in devicetree property names

mbolivar-nordic, 2023-01-04:
## Introduction

Devicetree properties, by convention, use dash (`-`) as a word separator. This RFC proposes making this a requirement for upstream bindings.

### Problem description

There are currently no automated checks in CI that ensure that devicetree properties use dashes as separators. As a result, many properties use underscore (`_`) as a separator. This is inconsistent with the style adopted by the devicetree specification for all standard properties (with the exception of `device_type`, but this property is deprecated and has limited usage).

It would be better style to use dashes instead, and would make things consistent for users typing properties. It also would avoid situation like having a `foo-bar` and `foo_bar` property defined on the same node due to typos.

### Proposed change

1. Make all in-tree bindings and DTS files use dashes
2. Add CI checks to make sure we don't regress this

## Detailed RFC

This is a straightforward treewide change.

### Proposed change (Detailed)

See pull request #53502. This does what is described above, taking the time to leave other compliance check code a bit cleaner than I found it.

### Dependencies

Out of tree users of existing bindings will need to migrate to the new property names.

### Concerns and Unresolved Questions

None.

## Alternatives

An approach which allowed both styles for backwards incompatibility was considered, but this is too ripe for introducing simultaneous use of `foo-bar` vs `foo_bar` in a single devicetree.

henrikbrixandersen, 2023-01-04:
I welcome this policy, however we need to consider existing "upstream" (Linux kernel) bindings using `_` separators for properties as a valid exemption for this.

mbolivar-nordic, 2023-01-05:
> we need to consider existing "upstream" (Linux kernel) bindings using `_` separators for properties as a valid exemption for this.

Indeed, this would make for a good exception -- could you give me some pointers? I did a search in our existing bindings for anything using underscores that had a corresponding binding doing the same in Linux, and I could not find any.

mbolivar-nordic, 2023-01-05:
@henrikbrixandersen I added some language about this potential exception. I think the CI results should still be a failure which require admin override, the same way we handle additions of new typedefs.

henrikbrixandersen, 2023-01-05:
> Indeed, this would make for a good exception -- could you give me some pointers? I did a search in our existing bindings for anything using underscores that had a corresponding binding doing the same in Linux, and I could not find any.

I am not aware of any current "violations" - I just want to make sure that we do not create a policy like this without an exception for upstream/well-established bindings.

cfriedt, 2023-01-10:
Makes sense to me

carlescufi, 2023-01-10:
Architecture WG:

- @henrikbrixandersen is in favor but asks about how the CI check would work. @mbolivar-nordic mentions that the tooling would issue a warning and CI would upgrade the warning to an error
- @galak mentions that out of tree users would benefit from a script that helps with the transition from `_` to `-`, otherwise he's in favor. The script might not be trivial and is unclear whether it is worthwhile. @mbolivar-nordic to investigate further 
-

nashif, 2023-04-13:
can this be closed? AFAIK it was done already.

cfriedt, 2023-04-16:
> can this be closed? AFAIK it was done already.

https://github.com/zephyrproject-rtos/zephyr/pull/53502 is still open

rruuaanng, 2024-12-19:
Is this still an ongoing effort? We should continue pushing it forward?

cfriedt, 2024-12-21:
@mbolivar - do you think you would consider continuing with #53502 or would you be ok if someone else were to continue it (preserving copyright of course)?

@rruuaanng - is that something you would be interested in working on?

rruuaanng, 2024-12-22:
@cfriedt Yes, I will push it forward if it is allowed.

rruuaanng, 2024-12-25:
@cfriedt @henrikbrixandersen Work is in progress. Should we add a reference to the device tree coding style in the contributor guide?
(I have already completed the patch, which provides a link to the upstream Linux device tree coding style, if the request is approved, I will issue a PR.)

@mbolivar If you have time, feel free to take a look at it ;--). I'd be more than happy to add a `Co-developed-by` tag or express my gratitude in the commit message to acknowledge the handoff of the work.

VynDragon, 2025-01-07:
Hello, I am having a issue in a PR that is directly related to this, namely i used the format <pin name><delimiter>gpios, but it ends up being i2c_en-gpio, as the pin name in datasheet is literally i2c_en. What is the way to handle this?

rruuaanng, 2025-01-08:
> Hello, I am having a issue in a PR that is directly related to this, namely i used the format <pin name><delimiter>gpios, but it ends up being i2c_en-gpio, as the pin name in datasheet is literally i2c_en. What is the way to handle this?

The proposal has not been implemented yet, so we can try to trace the source of the error from the existing modules. By the way, what do you expect it to look like? Typically, bindings are written by the user, and the naming is entirely up to them.

> but it ends up being i2c_en-gpio

I noticed that the name `i2c_en-gpios` is defined in `tests/drivers/build_all/sensor/i2c.dtsi`, and the same name is used in the bindings as well, which doesn’t seem to pose any issues.

VynDragon, 2025-01-08:
I meant it as the same issue as we were talking on the PR, if the pin name contains a _, should it be replaced with - regardless of consistency, as in what is the official stance on _ when not being used as delimiter but part of the name.

rruuaanng, 2025-01-08:
> if the pin name contains a _, should it be replaced with - regardless of consistency

Yes, we should use hyphens(`-`) instead of underscores(`_`). Even if its name in the manual is `i2c_en`, we should still use `i2c-en` for naming it.

As it stands, upstream typically uses a hyphen for "en" rather than an underscore, as seen in the case of this.
```
- row-output-enabled: An optional property indicating whether the row or
  column is being used as output. If specified the row is being used
  as the output. Else defaults to column.
```

> as in what is the official stance on _ when not being used as delimiter but part of the name referencing external documents.

The only basic exception for underscores is when they are used in upstream Linux naming conventions; otherwise, we should use hyphens.

The only special case at the moment is `mmc-hs400-1_8v`, which appears in that document:
https://www.kernel.org/doc/Documentation/devicetree/bindings/mmc/mmc-controller.yaml

VynDragon, 2025-01-08:
Alright, that is clear now,  I will change the PR to match, thanks.

mbolivar, 2025-01-08:
> @mbolivar If you have time, feel free to take a look at it ;--). I'd be more than happy to add a `Co-developed-by` tag or express my gratitude in the commit message to acknowledge the handoff of the work.

@rruuaanng thanks for taking this on. Please preserve any Signed-off-by: tags that I added on anything you adapt from my commits, and add your own Signed-off-by: to them -- that's all you need to do to preserve attribution; see the DCO docs:

https://docs.zephyrproject.org/latest/contribute/guidelines.html#developer-certification-of-origin-dco

rruuaanng, 2025-07-29:
#83352

closed 2025-07-29 by rruuaanng