zephyrproject-rtos/zephyr #53520: tests: drivers: gpio: gpio_api_1pin: peripheral.gpio.1pin fails

nordic-piks, 2023-01-05:
**Describe the bug**
The tests/drivers/gpio/gpio_api_1pin/peripheral.gpio.1pin fails.

Observed for
- nrf5340dk_nrf5340_cpunet

**To Reproduce**
Steps to reproduce the behavior:
1. have nrf5340dk_nrf5340_cpunet connected
2. go to your zephyr dir
3. call ./scripts/twister -T tests/drivers/gpio/gpio_api_1pin -p nrf5340dk_nrf5340_cpunet --device-testing --device-serial /dev/ttyACM1 -v --inline-logs
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v3.2.0-3053-g42b33382f765 ***
Running TESTSUITE gpio_api_1pin_conf
===================================================================
START - test_gpio_pin_configure_push_pull
Running test on port=gpio@418c0500, pin=28
Assertion failed at ../../../../../../../../../../zephyr/zephyr/tests/drivers/gpio/gpio_api_1pin/src/test_config.c:24: pin_get_raw_and_verify: (val_expected not equal to val_actual)
Test point 3: invalid pin get value
FAIL - test_gpio_pin_configure_push_pull in 0.052 seconds
===================================================================
START - test_gpio_pin_configure_single_ended
Running test on port=gpio@418c0500, pin=28
Board configuration does not allow to run the test
SKIP - test_gpio_pin_configure_single_ended in 0.109 seconds
===================================================================
TESTSUITE gpio_api_1pin_conf failed.
Running TESTSUITE gpio_api_1pin_int
===================================================================
START - test_gpio_int_edge_both
Running test on port=gpio@418c0500, pin=28
Assertion failed at ../../../../../../../../../../zephyr/zephyr/tests/drivers/gpio/gpio_api_1pin/src/test_pin_interrupt.c:99: test_gpio_pin_interrupt_edge: (cb_count not equal to cb_count_expected)
Test point 0: Pin interrupt triggered invalid number of times on rising/to active edge
FAIL - test_gpio_int_edge_both in 0.064 seconds
===================================================================
START - test_gpio_int_edge_falling
Running test on port=gpio@418c0500, pin=28
Assertion failed at ../../../../../../../../../../zephyr/zephyr/tests/drivers/gpio/gpio_api_1pin/src/test_pin_interrupt.c:107: test_gpio_pin_interrupt_edge: (cb_count not equal to cb_count_expected)
Test point 0: Pin interrupt triggered invalid number of times on falling/to inactive edge
FAIL - test_gpio_int_edge_falling in 0.065 seconds
===================================================================
START - test_gpio_int_edge_rising
Running test on port=gpio@418c0500, pin=28
Assertion failed at ../../../../../../../../../../zephyr/zephyr/tests/drivers/gpio/gpio_api_1pin/src/test_pin_interrupt.c:99: test_gpio_pin_interrupt_edge: (cb_count not equal to cb_count_expected)
Test point 0: Pin interrupt triggered invalid number of times on rising/to active edge
FAIL - test_gpio_int_edge_rising in 0.064 seconds
===================================================================
START - test_gpio_int_edge_to_active
Step 1: Configure pin as active high
Running test on port=gpio@418c0500, pin=28
Assertion failed at ../../../../../../../../../../zephyr/zephyr/tests/drivers/gpio/gpio_api_1pin/src/test_pin_interrupt.c:99: test_gpio_pin_interrupt_edge: (cb_count not equal to cb_count_expected)
Test point 0: Pin interrupt triggered invalid number of times on rising/to active edge
FAIL - test_gpio_int_edge_to_active in 0.074 seconds
===================================================================
START - test_gpio_int_edge_to_inactive
Step 1: Configure pin as active high
Running test on port=gpio@418c0500, pin=28
Assertion failed at ../../../../../../../../../../zephyr/zephyr/tests/drivers/gpio/gpio_api_1pin/src/test_pin_interrupt.c:107: test_gpio_pin_interrupt_edge: (cb_count not equal to cb_count_expected)
Test point 0: Pin interrupt triggered invalid number of times on falling/to inactive edge
FAIL - test_gpio_int_edge_to_inactive in 0.075 seconds
===================================================================
[... 159 lines cut ...]
- FAIL - [gpio_api_1pin_pin.test_gpio_pin_set_get_active_high] duration = 0.064 seconds
- FAIL - [gpio_api_1pin_pin.test_gpio_pin_set_get_active_low] duration = 0.063 seconds
- FAIL - [gpio_api_1pin_pin.test_gpio_pin_set_get_raw] duration = 0.059 seconds
- FAIL - [gpio_api_1pin_pin.test_gpio_pin_toggle] duration = 0.052 seconds
- PASS - [gpio_api_1pin_pin.test_gpio_pin_toggle_visual] duration = 6.015 seconds
SUITE FAIL -   0.00% [gpio_api_1pin_port]: pass = 0, fail = 9, skip = 0, total = 9 duration = 0.526 seconds
- FAIL - [gpio_api_1pin_port.test_gpio_port_set_bits_clear_bits] duration = 0.060 seconds
- FAIL - [gpio_api_1pin_port.test_gpio_port_set_bits_clear_bits_raw] duration = 0.054 seconds
- FAIL - [gpio_api_1pin_port.test_gpio_port_set_clr_bits] duration = 0.053 seconds
- FAIL - [gpio_api_1pin_port.test_gpio_port_set_clr_bits_raw] duration = 0.054 seconds
- FAIL - [gpio_api_1pin_port.test_gpio_port_set_masked_get] duration = 0.053 seconds
- FAIL - [gpio_api_1pin_port.test_gpio_port_set_masked_get_active_high] duration = 0.065 seconds
- FAIL - [gpio_api_1pin_port.test_gpio_port_set_masked_get_active_low] duration = 0.065 seconds
- FAIL - [gpio_api_1pin_port.test_gpio_port_set_masked_get_raw] duration = 0.061 seconds
- FAIL - [gpio_api_1pin_port.test_gpio_port_toggle] duration = 0.061 seconds
------ TESTSUITE SUMMARY END ------
===================================================================
RunID: 6e473436b4717d91340279fd4451c1fe
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain Zephyr SDK 0.15.1
 - Commit SHA or Version used: zephyr-v3.2.0-3053-g42b33382f765

anangl, 2023-01-13:
@nordic-piks This behavior is not caused by a bug in something related to the nrf5340dk_nrf5340_cpunet board. The thing is that if the Network core on nRF5340 is supposed to use a GPIO, the firmware on the Application core needs to assign this GPIO to the Network core. Hence, this is a problem in the test environment setup. If you use e.g. https://github.com/nrfconnect/sdk-nrf/tree/main/samples/nrf5340/empty_app_core for the Application core, the test will pass.

closed 2023-01-13 by anangl

nordic-piks, 2023-01-16:
@anangl Agree with your explanation, but I propose to make test code more explicit.
If the test cannot be executed on HW with twister, then in testcase.yaml there should be stated that for platform nrf5340dk_nrf5340_cpunet this sample is build_only.
Or the same can be achieved by adding harness_config with some special label when executing on nrf5340dk_nrf5340_cpunet, so it will build but not execute as usually such custom harness is not provided (here it means that App core is not flashed with proper content).
Last, least preferable, platform nrf5340dk_nrf5340_cpunet can be excluded from supported platforms for this test. 

What do you think?

nordic-piks, 2023-01-16:
Please reopen this issue.

nordic-piks, 2023-04-21:
Test was renamed to drivers.gpio.1pin.
Issue still occurs

katgiadla, 2023-07-20:
Please reopen. Issue still occurs