zephyrproject-rtos/zephyr #53537: TFM-M doesn't generate tfm_ns_signed.bin image for FOTA firmware upgrade

nandojve, 2023-01-05:
**Describe the bug**
When building a TFM image with b_u585i_iot02a_ns board which has TFM_PARTITION_FIRMWARE_UPDATE enabled Zephyr build system does not generate the tfm_ns_signed.bin.

**To Reproduce**
Follow steps of samples/tfm_integration/psa_firmware using the b_u585i_iot02a_ns board. After build a search in the build directory by *.bin files return:

```shell
$ find build -name *.bin

build/zephyr/zephyr.bin
build/zephyr/isrList.bin
build/update-header.bin
build/update-image.bin
build/tfm/bl2/ext/mcuboot/tfm_s_signed.bin
build/tfm/bin/tfm_s_signed.bin
build/tfm/bin/bl2.bin
build/tfm/bin/tfm_s.bin
build/tfm/install/outputs/tfm_s_signed.bin
build/tfm/install/outputs/bl2.bin
build/tfm/install/outputs/tfm_s.bin
```

**Expected behavior**

The Firmware Upgrade can work updating 3 image types: `FWU_IMAGE_TYPE_FULL`, `FWU_IMAGE_TYPE_SECURE` and `FWU_IMAGE_TYPE_NONSECURE`. Those are selected by `MCUBOOT_IMAGE_NUMBER`. In this case when `MCUBOOT_IMAGE_NUMBER=1` a `FWU_IMAGE_TYPE_FULL` signed  image in the binary form should be available. Without that a FOTA may not be possible. In the same way when `MCUBOOT_IMAGE_NUMBER=2` both `FWU_IMAGE_TYPE_SECURE` and `FWU_IMAGE_TYPE_NONSECURE` signed images in the binary form should be provided. The binary image should be the signed one to allow MCUboot perform the upgrade and that must include headers in the correct position, if necessary.

I noted that when `-DMCUBOOT_IMAGE_NUMBER=1` and both images are bigger then S slot, for instance, 524k, the final binary won't be generated. As far I understand for this board the final combined image could be up to 832k.

**Impact**
Besides sample of firmware upgrade be a valid one it not show too many details about how to generated those signed binaries image manually. Currently I can not move forward in the project.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK 0.15.2
 - Commit SHA: Current mainline e8f0e66bc4

CC: @erwango

erwango, 2023-01-06:
@nandojve My understanding is hat it is not specific to `b_u585i_iot02a_ns`, is it ?

nandojve, 2023-01-06:
> @nandojve My understanding is hat it is not specific to `b_u585i_iot02a_ns`, is it ?

Yes, I believe it affects all besides I'm currently more focused on the `b_u585i_iot02a_ns` and `stm32l562_ek_ns`.

microbuilder, 2023-01-25:
@nandojve The signed NS file is set in the TF-M cmake file via: `set(NS_SIGNED_FILE ${CMAKE_BINARY_DIR}/zephyr_ns_signed.hex)`

```bash
$ find . -name '*.hex'                
./tfm/install/outputs/tfm_s.hex
./tfm/install/outputs/bl2.hex
./tfm/bin/tfm_s.hex
./tfm/bin/bl2.hex
./zephyr_ns_signed.hex
./tfm_merged.hex
./tfm_s_signed.hex
./zephyr/zephyr.hex
```

This is then merged with the S hex file to produce `tfm_merged.hex`. Is converting the .hex file to .bin an option? I probably won't have the bandwidth to dig into this until after the feature freeze, though, sorry.

nandojve, 2023-05-27:
I managed to have a NS binary working on `ST B_U585I_IOT02A Discovery kit` using PSA API from Zephyr 3.2 and TFM 1.6.
The `objcopy -I ihex -O binary build/zephyr_ns_signed.hex build/zephyr_ns_signed.bin` command is enough.

joerchan, 2023-07-27:
@nandojve Please raise a feature-request for the zephyr build system to output signed TF-M images as .bin files.
Currently it only supports output of .hex files.

closed 2023-07-27 by joerchan