zephyrproject-rtos/zephyr #53548: net: ip: igmp: Mechanism to add MAC address for IGMP all systems multicast address to an ethernet multicast hash filter

cperera-aud, 2023-01-06:
**Is your enhancement proposal related to a problem? Please describe.**
Currently I'm looking into getting a multicast hash filter working for the STM32H7 by extending the work started in the PR https://github.com/zephyrproject-rtos/zephyr/pull/39089. When the multicast hash filter is enabled the STM32H7 stops receiving IGMP membership queries from the all systems multicast address, i.e. 224.0.0.1. As a result the STM32H7 doesn't respond back with IGMP membership reports with the multicast addresses that have joined. 

**Describe the solution you'd like**
I opened this case to get some ideas from the Zephyr community on how to go about and make IGMP signal that the MAC address for the all systems multicast address be added to the multicast hash filter. Once there's consensus on how to go about and implement it, I can make the enhancement myself.

One suggestion I had was to have a init function or something equivalent in the IGMP module which invokes net_if_mcast_monitor with the all systems multicast address. LWIP appears to have done something similar to this where they have a igmp_start function which sets up the multicast hash filter with the all systems multicast address.

Another suggestion was to have a separate callback mechanism for multicast filter add/remove which is invoked by the IGMP for startup, join and leave. 

**Describe alternatives you've considered**
I've just put a hack directly in the Zephyr STM32 ethernet driver to set the multicast hash filter with the MAC address of the IGMP all systems multicast address. With this hack, the SMT32H7 was able to receive IGMP membership queries from a Cisco SG300 switch which has IGMP snooping and a IGMP querier enabled.

jukkar, 2023-01-06:
IGMP should work same way as MLD is working for IPv6. The igmp.c will call `net_if_mcast_monitor()` when multicast address is registered/unregistered. The Ethernet driver needs to call `net_if_mcast_mon_register()` in its interface init function so that the driver can catch these multicast address changes. The mcux driver already does this so you can see how it does things.
In the upper layers the system then needs to join the needed multicast group(s), either by application or automatically by the network stack (perhaps according to some configuration data).

cperera-aud, 2023-01-09:
Hi @jukkar,

Thank you and much appreciated for getting back to me. Based on what was done in PR https://github.com/zephyrproject-rtos/zephyr/pull/39089, all what you described has been implemented. i.e. the STM32 ethernet driver calls `net_if_mcast_mon_register()` in its init function. The callback which is registered with `net_if_mcast_mon_register()` updates the multicast hash filter. All of this works fine... except when IGMP snooping and an IGMP querier is enabled. 

Where things start to break is where messages from 224.0.0.1 gets filtered by the ethernet MAC and they don't get passed up to the network stack. The 224.0.0.1 address is the multicast address where an IGMP querier sends membership queries. When the device receives the membership queries it replies with the membership reports. 

I acknowledge what you mentioned where the application can join the 224.0.0.1 address but I wasn't sure with this approach if whether any additional changes are required for IGMP because it probably doesn't make sense to send membership reports for 224.0.0.1? 

Hopefully I'm not violating a policy by sharing links to opensource projects but in lwip they add the 224.0.0.1 internally within IGMP at the beginning: https://github.com/lwip-tcpip/lwip/blob/master/src/core/ipv4/igmp.c#L145. In the case of Linux network stack, they appear to have a similar approach: https://github.com/torvalds/linux/blob/master/net/ipv4/igmp.c#L1789

In light of these do you still recommend using `net_if_ipv4_maddr_join()` to join 224.0.0.1 by the application?

jukkar, 2023-01-09:
> Hopefully I'm not violating a policy by sharing links to opensource projects but in lwip they add the 224.0.0.1 internally within IGMP at the beginning:

We could certainly do the same in zephyr. Probably best place to do that is when network interface comes up. Could you send a PR for it?

cperera-aud, 2023-01-09:
> We could certainly do the same in zephyr. Probably best place to do that is when network interface comes up. Could you send a PR for it?

Yes certainly I can @jukkar, feel free to assign the case to me as well.

cperera-aud, 2023-01-12:
Hi @jukkar Sorry for the delay. I've created the PR for the proposed change in https://github.com/zephyrproject-rtos/zephyr/pull/53733

PR #53733 (2023-01-13): net: ipv4: Added mechanism to add 224.0.0.1 address to a multicast filter
The purpose of the change was to have the Zephyr network stack internally add the IGMP all systems 224.0.0.1 multicast address to a multicast membership list so that multicast hash filter implementations can add this address to the hash filter.

Fixes #53548
commit b71b514e (2023-01-13): net: ipv4: Added mechanism to add 224.0.0.1 address to a multicast filter

The purpose of the change was to have the Zephyr network stack
internally add the IGMP all systems 224.0.0.1 multicast address to a
multicast membership list so that multicast hash filter implementations
can add this address to the hash filter.

Fixes #53548
closed 2023-01-13 by carlescufi