zephyrproject-rtos/zephyr #53555: ESP32-C3 Is RV32IMA, Not RV32IMC?

vindicatorr, 2023-01-06:
**Describe the bug**
https://github.com/zephyrproject-rtos/zephyr/blob/main/soc/riscv/esp32c3/Kconfig.soc is showing "rv32ima_zicsr_zifencei", but my understanding is that it ought to just be "rv32imc".

Full disclosure, I'm building zephyr in my own "special" way, but that should(?) be irrelevant to this issue(?)...

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using? esp32-c3
 - What have you tried to diagnose or workaround this issue? I encountered `Fatal error: -march=rv32ima_zicsr_zifencei: Invalid or unknown z ISA extension: 'zifencei`. I also saw https://github.com/zephyrproject-rtos/zephyr/pull/49459, and am also seeing https://github.com/zephyrproject-rtos/zephyr/commit/5fece03d7dfaa55ddc5598692420f4127b5d2fdd in the soc commit history.
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit? I see it was introduced in https://github.com/zephyrproject-rtos/zephyr/commit/10061efdc4d2d5a327a6c40affe43b6fcff4a7d2#diff-d77971293986a018c935dfa646540bbdeefc91315e07e80c9e3b3b8b033ae943

**To Reproduce**
Steps to reproduce the behavior:
N/A

**Expected behavior**
Use the correct instruction set?

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)
This along with using the older esp-idf 4.3 hal makes me wonder if this is viable for me, or if I should just stick with the latest esp-idf.

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows): Linux
 - Toolchain (e.g Zephyr SDK, ...): https://github.com/espressif/crosstool-NG
 - Commit SHA or Version used: Always the latest master of anything.

**Additional context**
https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/about.html says `The ESP32-C3 is a 2.4 GHz Wi-Fi Bluetooth Low Energy combo SoC, which integrates a 32-bit RISC-V RV32IMC single-core processor.`

sylvioalves, 2023-01-09:
@vindicatorr, I am still off this.. Why have you been talking about esp-idf 5.0 or Espressif's toolchain? If you are building app for Zephyr, you have go to stick with Zephyr's SDK instead, which contains proper ESP32 toolchain.

vindicatorr, 2023-01-09:
Because zephyr uses [`hal_esp](https://github.com/zephyrproject-rtos/hal_espressif/)?

But like I said from the start, that version is old (albeit receiving updates), which is why I went with doing alterations so it'd work with the upstream esp-idf.
Same with  the toolchain... If building for an Espressif-made chip, why not use THEIR toolchain for THEIR chip?

You said `stick with Zephyr's SDK instead, which contains proper ESP32 toolchain.`, but I'd think if any toolchain for an Espressif board is going to be `proper`, I'd think it'd be theirs.

Hence the origin of this `issue`, where `-march=rv32ima_zicsr_zifencei` is being used by Zephyr when Espressif documents the instruction capabilities for their board to be `rv32im`.

sylvioalves, 2023-01-09:
`hal_espressif` is a modified version of esp-idf for Zephyr. We will also clean that up to remove all unused content soon. It should not be used for as `esp-idf` environment.

> Same with the toolchain... If building for an Espressif-made chip, why not use THEIR toolchain for THEIR chip?  
The basic explanation is that Zephyr has its own toolchain to build its own environment. newlib, c++ and other stuff provided in Zephyr's OS requires its toolchain/SDK. Same as esp-idf, it needs its own toolchain..  

Regarding riscv capabilities, that I agree with you, I need to check that.  

Thanks.

vindicatorr, 2023-01-09:
I don't use `hal_espressif`. I was trying to shove a round peg (upstream `esp-idf`) into a square hole (`zephyr` module with some things pulled from `hal_espressif` and altered).
After all it's what you all did to get v4.3 to work with zephyr (`<...> is a modified version`), and it's what I figure you'll eventually be doing with v5.#+.
I was just trying to get ahead of the curve.

The way I see projects like zephyr is that you use the SDKs from the companies (ST/Nordicsemi/Espressif/...), and create a common interface for much of it.
Like ST may have a function `<...> st_spi_tx(<...>)`, Espressif `esp_spi_tx`, ... `<company>_spi_tx`, and then zephyr provides a singular all-encompassing function `zephyr_spi_tx` with some `ifdef`s in there to call the respective companie's SDK function.

While I love the idea of projects like Zephyr and Matter, combining various things into a single common thing (cross-platform), there are aspects I don't care much for like:
- Older underlying SDKs (again, v4.3 vs v5.1 for Espressif)
- Large third-party clones that may already exist elsewhere

What I like about Zephyr for the latter one is the option to use `ZEPHYR_MODULES`, as well as the ability to forego using `west`.
Yes, there are headaches when I try to force that round peg into the square hole by using upstream third-party repositories, but I'd like the ability to have a common framework to work with. Eg. Using upstream esp-idf for esp or zephyr projects (by making a small changes in `CMakeLists.txt` and setting the environment variables). As opposed to essentially replicating what already exists. I'm sure a lot of what you have in `hal_espressif` already exists in the upstream version of it.

It's a similar issue with Matter (I think I brought it up there as well), where they just automatically clone third-party stuff.
I saw 2.2GB had been downloaded before I "noped" out of there, and started looking at the scripts and stuff and selectively cloning only what was required.
Yes, it takes a bit of time to flesh out the inner workings, but I at least have control. I was able to build the `chip-tool` for linux and android. I think I was about to incorporate it with zephyr for my esp32, but ended up bowing out. I'll likely go back to it at some point, but am sticking with upstream SDKs for now (st, esp, nrf for me).

lsiudut, 2023-01-11:
@sylvioalves it seems that Zephyr SDK (at least 0.15.2) doesn't contain riscv32 compiler, which is required for ESP32-C3 builds?

As an additional note, before I ran into the same issue as @vindicatorr I was following tips that are printed out when building for esp32c3 and compiler path is not set properly. As toolchain couldn't be found I got suggestion to:
```
Run `west espressif install` to download it.  Then, export to path
  accordingly.
```
It seems that this is not valid anymore for Zephyr 3.2.0 though.

sylvioalves, 2023-01-11:
@lsiudut, sorry, I did not get it. I have just checkout v3.2.0, run `west update` and successfully build esp32c3 sample code.  

`west build -p -b esp32c3_devkitm`  

1) Have you properly configured and env as document in v3.2.0 release?  
2) Can you post the log output?

commit f18d811c (2023-02-28): soc: esp32c3: fix soc architecture definition

Update ESP32-C3 architecture as IMC instead IMA.

Although not documented, ESP32-S3 supports CSR instructions.
It also needs to be enabled, otherwise build will fail.

Fixes #53555
closed 2023-02-28 by fabiobaltieri