zephyrproject-rtos/zephyr #53556: Cannot add multiple out-of-the-tree secure partitions

k8zalewska, 2023-01-06:
Zephyr is not handling correctly the parameter -DTFM_EXTRA_PARTITION_PATHS when it contains multiple paths. According to documentation, path should be inside double quote and separated by ;.
```
-DTFM_EXTRA_MANIFEST_LIST_FILES="<Absolute-path-part-1-folder/manifes_list.yaml>;<Absolute-path-part-2-folder/manifes_list.yaml>"
-DTFM_EXTRA_PARTITION_PATHS="<Absolute-path-part-1-folder>;<Absolute-path-part-2-folder>"
```
When I try to use it in my code with:
```
set_property(TARGET zephyr_property_target
  APPEND PROPERTY TFM_CMAKE_OPTIONS
  ...
  -DTFM_EXTRA_PARTITION_PATHS="${MY_PARTITION_DIR};${DUMMY_PARTITION_DIR}"
)
```
I get the error because two paths are treated as one path, with space inside. 
```
CMake Error at secure_fw/partitions/CMakeLists.txt:58 (add_subdirectory):
  add_subdirectory given source
  "C:/Users/kasia/zephyrproject/zephyr/zephyr-integration-tests/my_partition
  C:/Users/kasia/zephyrproject/zephyr/zephyr-integration-tests/tfm_secure_partition/dummy_partition"
  which is not an existing directory.
```
In CMakeCache.txt I also see the paths are separated by space:
```
TFM_EXTRA_PARTITION_PATHS:PATH=C:/Users/kasia/zephyrproject/zephyr/zephyr-integration-tests/my_partition C:/Users/kasia/zephyrproject/zephyr/zephyr-integration-tests/tfm_secure_partition/dummy_partition
```
Same issue happens to multiple manifest file.

microbuilder, 2023-01-06:
I just had a quick look at a project where we have multiple paths, and this seems to be what we use:

```cmake
set_property(TARGET zephyr_property_target
  APPEND PROPERTY TFM_CMAKE_OPTIONS
  -DTFM_EXTRA_MANIFEST_LIST_FILES=${CMAKE_CURRENT_BINARY_DIR}/tfm_secure_partitions/tfm_manifest_list.yaml
  -DTFM_EXTRA_PARTITION_PATHS="${CMAKE_CURRENT_LIST_DIR}/tfm_secure_partitions/tfm_huk_deriv_srv\\\;${CMAKE_CURRENT_LIST_DIR}/tfm_secure_partitions/tfm_tflm_service\\\;${CMAKE_CURRENT_LIST_DIR}/tfm_secure_partitions/tfm_utvm_service"
  )
```

microbuilder, 2023-01-10:
@stephanosio https://github.com/zephyrproject-rtos/zephyr/pull/53557 should solve this if you have time to review/merge?

PR #53557 (2023-01-10): doc: tfm: Document how to pass list arguments to the TF-M build system
Document how to pass list arguments to the TF-M build system.

fixes: #53556
commit cc8493c3 (2023-01-10): doc: tfm: Document how to pass list arguments to the TF-M build system

Document how to pass list arguments to the TF-M build system.

fixes: #53556
closed 2023-01-10 by stephanosio