zephyrproject-rtos/zephyr #53559: mgmt: mcumgr: explore why UART interface is so slow (possible go application fault)

nordicjm, 2023-01-06:
**Is your enhancement proposal related to a problem? Please describe.**
With the rework from https://github.com/zephyrproject-rtos/zephyr/pull/53558 - the fs_mgmt transfer speed for bluetooth has jumped up to about 13KBps, whilst the UART (note: via shell interface) is stuck on about 3KBps... What makes even less sense is that the UART speed does not increase with larger MTUs or faster baud rates, indeed with an MTU of 4KB and baud rate of 460800, the transfer speed is... 3KBps.

My suspicious are high that it's the go tool that is the culprit of the issues, to see no difference between 115200 baud with 2048 byte MTU and 460800 baud with 4096 byte MTU just seems absurd, regardless of the processing speed of the various layers in zephyr. 

**Describe the solution you'd like**
MCUmgr transfer speeds via UART to not be such in the 1980s.

**Describe alternatives you've considered**
There are no alternatives, at 3KBps the system is just woefully inadequate.

xudongzheng, 2023-01-07:
Thanks for this patch. With fd00ebc, MCUmgr file transfer over USB CDC ACM has gotten significantly faster when used with [this patch](https://github.com/apache/mynewt-newtmgr/issues/185#issuecomment-894907940) and [this fix](https://github.com/apache/mynewt-newtmgr/issues/185#issuecomment-920868644).

Uploading 32KB decreased from from 10.249s to 2.120s. Downloading 32KB decreased from 2.110s to 0.696s

nordicjm, 2023-01-09:
Tested with custom tool, 896KB uploaded in 63 seconds = 14.2KBps, at 460800 baud, 512 byte MTU, therefore concurring that the fault is with the go application at the 3KBps limitation, which has an appalling design. At this point the only logical thing to do is dump it entirely.

However the problem is not entirely with the go application, I'm maxing out at about 15.5KBps over UART with 1Mbps baud and 2048 byte MTU, which is about 6% utilisation of the available baud rate - absolutely terrible. Might need to see about larger shell MCUmgr frames e.g. 1KB instead of 128 byte sizes and see if that helps improve throughput.

CC @de-nordic

nordicjm, 2023-02-06:
No point keeping this open as the issue is with the go application, which now has a note that it is not recommended to be used.

Maximum transfer speeds for fs_mgmt observed with custom UART tool were ~18KBps write speed and ~23KBps read speed (external QSPI memory in high-speed mode on nRF52840 with QSPI operating at 32MHz)

closed 2023-02-06 by nordicjm