zephyrproject-rtos/zephyr #53568: Kconfig search has white on white text

decsny, 2023-01-06:
**Describe the bug**
The Kconfig search on the documentation site has white text on the white background of the search bar, so you can only see entered text by highlighting as seen in the screenshot below:
[image]

**Expected behavior**
Perhaps the text is meant to be white, but the background of the search bar should be black? I don't know, but I just expected the text to be readable

**Impact**
Annoyance 

**Environment (please complete the following information):**
Other people have the same issue as me, and the common factors seem to be 1) Firefox 2) Windows 10 3) Windows has the dark color theme. Switching the windows color theme to light results in black on white text in the Kconfig search bar, as would be expected

beriberikix, 2023-01-07:
Was just going to report this. I'm on Windows 11 and latest Chrome (108.0.5359.125 (Official Build) (64-bit))

decsny, 2023-01-29:
I have also confirmed the issue exists on linux mint on firefox as well

PR #54451 (2023-02-07): docs: kconfig: restore appearance of kconfig search
This PR restores the appearance of kconfig search, with rounded corners and black background. This should resolve visibility issues encountered on some browsers when running in dark mode

Fixes #53568

The output Kconfig search from this PR can be viewed here: https://builds.zephyrproject.io/zephyr/pr/54451/docs/kconfig.html

Some sample screenshots (collected on Firefox 109.0.1, running on Windows 10):

OS Light mode:

[image]

OS Dark mode:

[image]

I've also verified this change on Chrome 109.0.5414.120, on Windows 10.

@gmarull, I'd appr [... cut]
commit 83d5ed58 (2023-02-07): docs: kconfig: restore appearance of kconfig search

Restore appearance of kconfig search, with rounded corners and
black background. This should resolve visibility issues encountered
on some browsers when running in dark mode

Fixes #53568
closed 2023-02-07 by carlescufi