zephyrproject-rtos/zephyr #53579: add_compile_defintions does not "propagate upwards" as supposed to when using west

erlingrj, 2023-01-07:
**Describe the bug**
I have this simple project

CMakeLists.txt
```
add_subdirector(foo)
target_sources(app PRIVATE main.c)
```

foo/CMakeLists.txt
```
add_compile_definitions(FOO=42)
```

The expected behavior is that main.c is compiled with -DFOO=42. However, this is not the case. For some reason the compile definitions does not propagate "upwards" as they are supposed to do. If I instead do

CMakeLists.txt
```
add_compile_definitions(FOO=42)
target_sources(app PRIVATE main.c)
```

Then everything works as expected. Any suggestions here?

thedjnK, 2023-01-07:
Invalid issue caused by using the wrong cmake command. `include()` should be used here which will share the same scope as the parent, `add_subdirectory()` will create a child scope inheriting the parent's variables as duplicates, but they are separate. Updating them from the child will not propagate them back to the parent.

erlingrj, 2023-01-07:
Thank you, you are right. Sorry for confusion

closed 2023-01-07 by erlingrj