zephyrproject-rtos/zephyr #53584: STM32F1 PWM Input Capture Issue

lainslie, 2023-01-08:
### Discussed in https://github.com/zephyrproject-rtos/zephyr/discussions/53514

<div type='discussions-op-text'>

<sup>Originally posted by **lainslie** January  5, 2023</sup>
Hi there,

I am using a custom STM32F103C8T6 board. Part of the design is to determine a fans speed by its cyclic period, an example signal has been attached:

[image]

This fan has been connected to TIM3 ch2 PA7. See the device tree setup for this device:

[image]

Within my proj.conf, I have enabled CONFIG_PWM and CONFIG_PWM_CAPTURE. 

I am currently experiencing an issue that occurs when I initialise the PWM capture for that pin. The PWM output of the fan seems to be 'driven' to low (0 voltage), seen by the logic analyser, which means I am unable to read a value from this pin, since no cycles are occurring. I'm unsure of what is causing this issue, the only idea in my head is that the pin by default is configured as an output pin, since PWM is typical an output signal, and is not reconfigured as an input pin when pwm_capture_usec, pwm_pin_configure_capture or pwm_pin_enable_capture is called. Any help would be greatly appreciated thanks</div>

gautierg-st, 2023-01-09:
Hello,

I have created an overlay for a Nucleo F103RB for the test pwm_loopback and the test is successful.
Here is the code of nucleo_f103rb.overlay:
```
#include <zephyr/dt-bindings/pwm/pwm.h>

/ {
	pwm_loopback_0 {
		compatible = "test-pwm-loopback";
		/* first index must be a 32-Bit timer */
		pwms = <&pwm1 1 0 PWM_POLARITY_NORMAL>,
			<&pwm3 2 0 PWM_POLARITY_NORMAL>;
	};
};

/* 32-Bit timers */
&timers1 {
	status = "okay";
	pwm1: pwm {
		status = "okay";
		pinctrl-0 = <&tim1_ch1_pwm_pa8>; /* CN8 PIN1 A0 */
		pinctrl-names = "default";
	};
};

&timers3 {
	st,prescaler = <255>;
	status = "okay";
	pwm3: pwm {
		status = "okay";
		pinctrl-0 = <&tim3_ch2_pwm_pa7>; /* CN5 PIN4 D11 */
		pinctrl-names = "default";
	};
};
```
The main difference I see is that you don't have a st,prescaler for timers3. I tested without and it fails. Maybe it's all you're missing?

lainslie, 2023-01-11:
Thanks for the reply! I decided to run the test_pwm_loopback tests the same way you did so there was no ambiguity on if I was using any functions incorrectly. As you suspected, it seems to be an issue with not using the pre-scaler! Thanks so much for your help!!!!!

gautierg-st, 2023-01-23:
Hi lainslie,
I come back to you to know if you succeeded to solve your problem. 
If so, could you please close this issue?
Thanks

lainslie, 2023-01-24:
Hi Gautierg,

I have found that although using the pwm test loopback works, when the tachometer feedback signal is used instead, the pin is still driven low and no signal is seen. I am unsure if this is a hardware issue on the fans end, or a pin setting still. Any ideas?

gautierg-st, 2023-01-25:
Sorry for asking maybe trivial questions, but I need to understand what we're facing.
- On your custom board, you executed the test pwm_loopback, connecting two pins, with the tachometer disconnected, and it passed?
- With the tachometer disconnected from the PWM pin, do you have a signal on the tachometer output pin?

If both answers are yes, I don't really understand what's going on here. One further test you could try is to connect the tachometer output pin to a simple input pin of the STM32F103 (without PWM capture) and see if you have a signal. If yes, we can look further on the pin configuration, if no, I'd lean on the HW side.

lainslie, 2023-01-31:
Sorry for the delay I have had to travel a bit recently. 

1. On my custom board, I connected the output of one pwm to the input of the other pwm, and this passed. The tachometer was disconnected.
2. When the tachometer is disconnected from the PWM pin, there is a signal on the tachometer output pin. 

So both answers are yes.

Further testing, when the output pin is connected to a simple input pin of the stm32f103 (I decided to just use a rising edge interrupt), the signal is unaffected and the interrupt fires correctly. 

From this, we should look further into pin configuration?

gautierg-st, 2023-02-01:
Could you run the pwm_loopback but instead of having the loopback, you connect the tachometer? The goal is not to have the test PASSED, but just to see if your signal is here or not. Maybe you should loop the pwm capture code in the test to have the time to look at the signal and be sure the pin is not reconfigured at the end.

If you see your signal, the problem probably comes from the way you handle things in your code, and I would suggest to try to reproduce the way it is done in the pwm_loopback test. I can't really say any more without seeing that part of your code unfortunately.

lainslie, 2023-02-01:
Thanks for the suggestion. I ran the test, ensuring that the fan was spinning, and the pwm loopback input was set to the fan tachometer. The results of this test was the tacho signal not being observed and instead was pulled to logic level 0, the same as was originally reported.

gautierg-st, 2023-02-02:
I did a little search and apparently there is a peculiarity in GPIO management in STM32F1xx family. The input alternate function should just be configured in input (contrary to other STM32 families).
See https://github.com/stm32duino/Arduino_Core_STM32/issues/812

Could you try changing the file modules\hal\stm32\dts\st\f1\stm32f103c(8-b)tx-pinctrl.dtsi by replacing this:
```
			/omit-if-no-ref/ tim3_ch2_pwm_pa7: tim3_ch2_pwm_pa7 {
				pinmux = <STM32F1_PINMUX('A', 7, ALTERNATE, TIM3_REMAP0)>;
			};
```
by this:
```
			/omit-if-no-ref/ tim3_ch2_pwm_pa7: tim3_ch2_pwm_pa7 {
				pinmux = <STM32F1_PINMUX('A', 7, GPIO_IN, TIM3_REMAP0)>;
			};
```
And let me know if that solves your problem.

lainslie, 2023-02-09:
Thanks again for all your help. I have thoroughly tested the change that you suggested and it now works. I am able to see the tachometer signal, and successfully capture this signal using pwm_capture_cycles. 

Thanks so much

gautierg-st, 2023-02-09:
I have pushed #54673. Could you please try it out to check if it correctly solves the issue? 
Note that you'll need to do a west update, and to rename your PWM gpios in your dts like that: tim1_ch1_pwm_out_pa8 or tim1_ch1_pwm_in_pa8.

PR #54673 (2023-02-20): Fix stm32f1xx pwm input capture
On STM32F1xx, pins used for PWM input must be configured as input, and not as alternate.
This PR uses the latest STM32 HAL where new entries were added for STM32F1xx PWM inputs.

Fixes #53584
closed 2023-02-10 by erwango