zephyrproject-rtos/zephyr #53605: tests: posix: common: portability.posix.common fails - posix_apis.test_clock_gettime_rollover

nordic-piks, 2023-01-09:
**Describe the bug**
The tests/posix/common/portability.posix.common... fails.

Tests:
- tests/posix/common/portability.posix.common.tls.newlib
- tests/posix/common/portability.posix.common.picolibc
- tests/posix/common/portability.posix.common.tls
- tests/posix/common/portability.posix.common

Observed for
- nrf52840dk_nrf52840
- nrf52dk_nrf52832
- nrf9160dk_nrf9160
- nrf9160dk_nrf9160_ns
- nrf5340dk_nrf5340_cpuapp
- nrf5340dk_nrf5340_cpuapp_ns
- nrf5340dk_nrf5340_cpunet

**To Reproduce**
Steps to reproduce the behavior:
1. have nrf52840dk_nrf52840 connected
2. go to your zephyr dir
3. call ./scripts/twister -T tests/posix/common -p nrf52840dk_nrf52840 --device-testing --device-serial /dev/ttyACM0 -v --inline-logs
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
[2023-01-05T23:39:30.794Z] [0mERROR   - *** Booting Zephyr OS build zephyr-v3.2.0-3121-g8e4137f14901 ***[0m
[2023-01-05T23:39:30.794Z] [0mRunning TESTSUITE posix_apis[0m
[2023-01-05T23:39:30.794Z] [0m===================================================================[0m
[2023-01-05T23:39:30.794Z] [0mSTART - test_clock_gettime_rollover[0m
[2023-01-05T23:39:30.794Z] [0mCONFIG_SYS_CLOCK_TICKS_PER_SEC: 32768[0m
[2023-01-05T23:39:30.794Z] [0mrollover_s: 562949953421311[0m
[2023-01-05T23:39:30.794Z] [0mt-1: {562949953421311, 999969482}[0m
[2023-01-05T23:39:30.794Z] [0mt+0: {0, 0}[0m
[2023-01-05T23:39:30.794Z] [0mAssertion failed at ../../../../../../../../../zephyr/zephyr/tests/posix/common/src/clock.c:177: posix_apis_test_clock_gettime_rollover: rollover_s not equal to ts[1].tv_sec[0m
[2023-01-05T23:39:30.794Z] [0mFAIL - test_clock_gettime_rollover in 0.027 seconds[0m
[2023-01-05T23:39:30.794Z] [0m===================================================================[0m
[2023-01-05T23:39:30.794Z] [0mSTART - test_nanosleep_0_1[0m
[2023-01-05T23:39:30.794Z] [0mPASS - test_nanosleep_0_1 in 0.001 seconds[0m
[2023-01-05T23:39:30.794Z] [0m===================================================================[0m
[2023-01-05T23:39:30.794Z] [0mSTART - test_nanosleep_0_1000000000[0m
[2023-01-05T23:39:30.794Z] [0mPASS - test_nanosleep_0_1000000000 in 0.001 seconds[0m
[2023-01-05T23:39:30.794Z] [0m===================================================================[0m
[2023-01-05T23:39:30.794Z] [0mSTART - test_nanosleep_0_1001[0m
[2023-01-05T23:39:30.794Z] [0mPASS - test_nanosleep_0_1001 in 0.001 seconds[0m
[2023-01-05T23:39:30.794Z] [0m===================================================================[0m
[2023-01-05T23:39:30.794Z] [0mSTART - test_nanosleep_0_500000000[0m
[2023-01-05T23:39:30.794Z] [0mPASS - test_nanosleep_0_500000000 in 0.501 seconds[0m
[2023-01-05T23:39:30.794Z] [0m===================================================================[0m
[2023-01-05T23:39:30.794Z] [0mSTART - test_nanosleep_0_n1[0m
[2023-01-05T23:39:30.794Z] [0mPASS - test_nanosleep_0_n1 in 0.001 seconds[0m
[2023-01-05T23:39:30.794Z] [0m===================================================================[0m
[2023-01-05T23:39:30.794Z] [0mSTART - test_nanosleep_1_0[0m
[2023-01-05T23:39:30.794Z] [0mPASS - test_nanosleep_1_0 in 1.001 seconds[0m
[2023-01-05T23:39:30.794Z] [0m===================================================================[0m
[2023-01-05T23:39:30.794Z] [0mSTART - test_nanosleep_1_1[0m
[2023-01-05T23:39:30.794Z] [0mPASS - test_nanosleep_1_1 in 1.001 seconds[0m
[2023-01-05T23:39:30.794Z] [0m===================================================================[0m
[2023-01-05T23:39:30.794Z] [0mSTART - test_nanosleep_1_1001[0m
[2023-01-05T23:39:30.794Z] [0mPASS - test_nanosleep_1_1001 in 1.001 seconds[0m
[2023-01-05T23:39:30.794Z] [0m===================================================================[0m
[2023-01-05T23:39:30.794Z] [0mSTART - test_nanosleep_NULL_NULL[0m
[2023-01-05T23:39:30.794Z] [0mPASS - test_nanosleep_NULL_NULL in 0.001 seconds[0m
[... 178 lines cut ...]
[2023-01-05T23:39:30.795Z] [0m- PASS - [posix_apis.test_posix_normal_mutex] duration = 0.103 seconds[0m
[2023-01-05T23:39:30.795Z] [0m- PASS - [posix_apis.test_posix_pthread_create_negative] duration = 0.001 seconds[0m
[2023-01-05T23:39:30.795Z] [0m- PASS - [posix_apis.test_posix_pthread_error_condition] duration = 0.001 seconds[0m
[2023-01-05T23:39:30.795Z] [0m- PASS - [posix_apis.test_posix_pthread_execution] duration = 2.260 seconds[0m
[2023-01-05T23:39:30.795Z] [0m- PASS - [posix_apis.test_posix_pthread_termination] duration = 1.014 seconds[0m
[2023-01-05T23:39:30.795Z] [0m- PASS - [posix_apis.test_posix_realtime] duration = 1.803 seconds[0m
[2023-01-05T23:39:30.795Z] [0m- PASS - [posix_apis.test_posix_recursive_mutex] duration = 0.003 seconds[0m
[2023-01-05T23:39:30.795Z] [0m- PASS - [posix_apis.test_posix_rw_lock] duration = 0.080 seconds[0m
[2023-01-05T23:39:30.795Z] [0m- PASS - [posix_apis.test_posix_semaphore] duration = 5.001 seconds[0m
[2023-01-05T23:39:30.795Z] [0m- PASS - [posix_apis.test_posix_single_thread_multiple_keys] duration = 0.017 seconds[0m
[2023-01-05T23:39:30.795Z] [0m- PASS - [posix_apis.test_posix_thread_attr_stacksize] duration = 0.001 seconds[0m
[2023-01-05T23:39:30.795Z] [0m- PASS - [posix_apis.test_posix_timer] duration = 2.111 seconds[0m
[2023-01-05T23:39:30.795Z] [0m- PASS - [posix_apis.test_pthread_descriptor_leak] duration = 0.007 seconds[0m
[2023-01-05T23:39:30.795Z] [0m- PASS - [posix_apis.test_sleep] duration = 4.001 seconds[0m
[2023-01-05T23:39:30.795Z] [0m- PASS - [posix_apis.test_usleep] duration = 1.501 seconds[0m
[2023-01-05T23:39:30.795Z] [0m------ TESTSUITE SUMMARY END ------[0m
[2023-01-05T23:39:30.795Z] [0m===================================================================[0m
[2023-01-05T23:39:30.795Z] [0mRunID: d8c8edf3a90b247d687dbb491807cda0[0m
[2023-01-05T23:39:30.795Z] [0mPROJECT EXECUTION FAILED[0m
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain Zephyr SDK 0.15.1
 - Commit SHA or Version used: zephyr-v3.2.0-3121-g8e4137f14901

hakehuang, 2023-01-30:
NXP platfrom(mimxrt1170_evk_cm7/mimxrt1160_evk_cm7) has the same issue

cfriedt, 2023-02-07:
@nordic-krch - have you made any progress on this? It would be best to get a fix in for v3.3
cc @carlescufi

nordic-krch, 2023-02-07:
sorry, unfortunately no and i won't get time for that in the near future.

cfriedt, 2023-02-07:
@carlescufi - is there someone else who can be assigned? I am happy to help, but would prefer to review since I might not have all of the hardware available.

cfriedt, 2023-02-26:
@hakehuang - was there a fix that went in for this for NXP platforms recently?

hakehuang, 2023-02-27:
> @hakehuang - was there a fix that went in for this for NXP platforms recently?

not yet, but I will work on a solution.

cfriedt, 2023-03-28:
I'm able to reproduce the issue with `nrf52840dk_nrf52840`. Will self-assign for now as well.

carlescufi, 2023-03-29:
> I'm able to reproduce the issue with `nrf52840dk_nrf52840`. Will self-assign for now as well.

Thanks @cfriedt. Let us know if we can help.

PR #56446 (2023-04-04): tests: posix: common: remove test_clock_gettime_rollover
The logic in this test only worked for a subset of configurations with 64-bit counters. Remove the test to ensure the suite passes on all platforms.

Fixes #53605
closed 2023-04-04 by carlescufi