zephyrproject-rtos/zephyr #53607: sysbuild: missing rom_report builds for targets

nordicjm, 2023-01-09:
**Describe the bug**
There are rom_report and ram_report builds for targets, however sysbuild seemingly does not create entries for these from the top level project - for zephyr-based images, these should be created

**To Reproduce**
`ninja smp_svr_rom_report` does not work

**Expected behavior**
The command to work

**Impact**
Annoyance

**Environment (please complete the following information):**
 - Commit SHA or Version used: 8d86fa012592e3c78f6489ff88966c23f780f25f

tejlmand, 2023-01-11:
I would like to open a discussion on expectations should be.

We can't maintain a mapping of every subtarget through sysbuild, because any Zephyr module can effectively add additional targets, so before proceeding we should discuss expectation to ensure when and why to do such mapping.

The reason that `menuconfig`, `guiconfig`, `hardenconfig` are mapped is beacuse sysbuild itself has those targets, meaning that the build target: `ninja menuconfig` is valid and opens menuconfig for sysbuild.

There exists no `ninja rom_report` or `ninja ram_report` targets.

Dedicated image build targets can be invoked using:
```
$ cd build
$ ninja -C <image> rom_report
```
for example to run `rom_report` for mcuboot in a sysbuild controlled build folder, one can do: `ninja -C mcuboot rom_report`.

With this, I have provided reason why the various `config` targets are mapped, because they natively exists in sysbuild itself as targets.
The report targets doesn't.

In my opinion it does make sense to allow the rom / ram reports to be mapped, but what would be the argument for exactly those targets and not others ?
Like `tfm_ram_report` or `bl2_ram_report` ?

We need to define exactly which, why, and when a target should be mapped, as to ensure some form of consistency.

I gave the reason for the existing mapping, but that reason cannot be used for the targets in this issue.
Feel free to propose a guideline as to when we should map other targets.

tejlmand, 2023-01-11:
changed to `feature request` as those targets are available using `ninja -C <image> <target>`.
As of now, only targets which also exists natively as sysbuild build targets are mapped, so this is actually a request to broaden the scope of what to map, and for that we need a discussion to have some form of consistency.

nordicjm, 2023-01-24:
From `cmake/usage/usage.cmake` it lists these:
menuconfig, guiconfig, flash, debug, debugserver, attach, ram_report, rom_report

Interestingly enough, hardenconfig is missing.

In `cmake/reports/CMakeLists.txt` it has:
ram_report, rom_report, footprint, puncover (if installed), pahole (if installed)
Then for tfm:
tfm_ram_report, tfm_rom_report, tfm_footprint, bl2_ram_report, bl2_rom_report, bl2_footprint

I don't know what footprint does, only used ram/rom report. Arguably the tfm_* ones can be dropped if tfm is fixed to properly use a secure and non-secure image so they would no longer be present, and arguably from that as well the bl2_* ones could be dropped if that was split out into its own image - which it should be, it is a second stage bootloader for trusted firmware, it should be its own target.

nordicjm, 2023-01-24:
Addding @joerchan for comment on TF-M

closed 2023-04-19 by nordicjm