zephyrproject-rtos/zephyr #53617: unicast_audio_client and unicast_audio_server example assert

JiaLWang, 2023-01-09:
**Describe the bug**
When using CIS example of LE audio will cause a crash. What is the reason for the above crash?

**To Reproduce**
Steps to reproduce the behavior:
1. Need two nrf52840 board
2. one board flash unicast_audio_client example and another flash unicast_audio_server example.
3. will cause crash

error log as below:

[image]

**Environment (please complete the following information):**
 - OS: Linux
 - Commit : 58c924ec569afceaa96d8e771a91c797b978c81b

JiaLWang, 2023-01-11:
Hi @cvinayak , Anything update?

cvinayak, 2023-01-11:
Zephyr Controller does not yet support Connected ISO feature.

JiaLWang, 2023-01-11:
OK, Thanks for you help.

closed 2023-01-11 by JiaLWang