zephyrproject-rtos/zephyr #53636: cdc_acm fails on lpcxpresso55s69 board

DerekSnell, 2023-01-09:
**Describe the bug**
Running the cdc_acm sample on this board fails.  

**To Reproduce**
Tested with this build command
```
west build -b lpcxpresso55s69_cpu0 samples/subsys/usb/cdc_acm
```

**Logs and console output**
With the latest main branch, the USB device does not enumerate properly in Windows.  The debug console shows no error:
```
*** Booting Zephyr OS build zephyr-v3.2.0-3235-g871d6d26eeb0 ***
[00:00:00.000,000] <inf> cdc_acm_echo: Wait for DTR
```

Using v3.2.0, the USB device enumerates, but gives this error:
```
*** Booting Zephyr OS build zephyr-v3.2.0  ***
[00:00:00.004,000] <inf> cdc_acm_echo: Wait for DTR
[00:00:07.560,000] <err> usb_dc_mcux: Failed to fill ep 0x80 buffer
```

Testing in v3.1, the app crashes with a hard fault:
```
*** Booting Zephyr OS build zephyr-v3.1.0  ***
[00:00:00.004,000] <err> os: ***** HARD FAULT *****
[00:00:00.004,000] <err> os:   Bus fault on vector table read
[00:00:00.004,000] <err> os: r0/a1:  0xfefa125b  r1/a2:  0x2280ff63  r2/a3:  0x4
0000000
[00:00:00.004,000] <err> os: r3/a4:  0x300007b0 r12/ip:  0x300007b0 r14/lr:  0x3
00039b0
[00:00:00.004,000] <err> os:  xpsr:  0xc33cc200
[00:00:00.004,000] <err> os: Faulting instruction address (r15/pc): 0x5ac3c35a
[00:00:00.004,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:00.004,000] <err> os: Current thread: 0x30000cc8 (unknown)
[00:00:00.061,000] <err> os: Halting system
```

**Environment (please complete the following information):**
 - tested on Windows10
 - Toolchain Zephyr SDK 0.15.1

henrikbrixandersen, 2023-01-09:
I recently ran into issues on `main` getting a `frdm_k64f` to enumerate on certain Windows USB hosts as well (Dell workstation + Windows 10 failed, Lenovo Laptop + Windows 10 works). Did you test on multiple hosts?

DerekSnell, 2023-01-09:
> I recently ran into issues on `main` getting a `frdm_k64f` to enumerate on certain Windows USB hosts as well (Dell workstation + Windows 10 failed, Lenovo Laptop + Windows 10 works). Did you test on multiple hosts?

Thanks @henrikbrixandersen , no I only tested on 1 Windows 10 machine, also a Dell

k0d, 2023-01-10:
I've looked further into this issue, and the usb_dc_mcux driver doesn't support the 512 byte buffer of USB_HS. It sets it as 64, and then it fails to send the configuration descriptor as that is greater than 64 bytes. I don't know how to fix this though.

PR #53969 (2023-01-30): drivers: usb: Fix NXP driver for CDC test cases
The cdc_acm tests were failing. Remove a length check in the write function as this is handled by the HAL driver. Remove unused type member and improve the error checking.

Fixes #53636 , #52625
closed 2023-01-30 by fabiobaltieri