zephyrproject-rtos/zephyr #53642: Display driver sample seems to mix up RGB565 and BGR565

danieldegrasse, 2023-01-10:
**Describe the bug**
It appears that `samples/drivers/display` has the RGB565 and BGR565 color formats swapped. The function implementation to get RGB565 colors calls `get_rgb565_color` (https://github.com/zephyrproject-rtos/zephyr/blob/de313d748ad22d7f429d425a0f1beb0505ad5e21/samples/drivers/display/src/main.c#L131), then performs bitshifting which appears to convert the color from RGB565 to BGR565. The `get_rgb565_color` function appears to return a red color for the top left, green color for the top right, and blue color for the bottom right, all formatted in RGB565 (as expected by the [README](https://github.com/zephyrproject-rtos/zephyr/blob/de313d748ad22d7f429d425a0f1beb0505ad5e21/samples/drivers/display/README.rst)

Additionally, it appears that the NXP eLCDIF peripheral driver treats BGR565 mode as RGB565:
https://github.com/zephyrproject-rtos/zephyr/blob/de313d748ad22d7f429d425a0f1beb0505ad5e21/drivers/display/display_mcux_elcdif.c#L223

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 I encountered this issue while writing support for the LCDIF peripheral present on the RT595 EVK, and testing #53641. The same workaround used by the eLCDIF driver worked for me, but I'm unsure if the sample is formatting colors as expected.

**Expected behavior**
RGB565 color format should be presented like the following:
[image]
(image source: http://www.barth-dev.de/online/rgb565-color-picker/)

jeronimoagullo, 2023-01-10:
Hi everyone, 

I hope that my comment helps to solve the bug. 

I confirm the bug that has described @danieldegrasse. I am using the `stm32f746g_disco` board resulting in the same behaviour. 

Actually, I am displaying an image from an C array in RGB565 format. However, if I display it directly using the image array as the buffer of  `display_write` function the colours are not correct since the 2 bytes of RGB565 format are flipped. Thus, I need to create a new buffer of the same length as the image array and flip the bytes each 2 bytes. I use the following code:
```c
for (size_t idx = 0; idx < IMAGE_LENGTH; idx += 2) {
		*(buf + idx + 0) = (*(image_array+ idx + 1));
		*(buf + idx + 1) = (*(image_array+ idx + 0));
	}
```
I don't know how the buffer is processed by the `display_write` function, otherwise, I would try to fix it.  

If I can help let me know.

danieldegrasse, 2023-01-10:
After some more investigation of Zephyr's handling of RGB565, the implementation in `samples/drivers/display` is indeed consistent with the documentation. The implementation of `get_rgb565_color` byte swaps the pixel data, which is expected since the data is stored in big endian format and the host system is little endian.

My confusion was because Zephyr's implementation of RGB565 is different than most definitions available. LVGL writes display data in little endian format, as seen here: https://docs.lvgl.io/8.0/overview/color.html. By setting `LV_COLOR_16_SWAP` in `lv_conf.h`, LVGL would write data in the format Zephyr describes as RGB565. This resulted in only the display sample being broken in my testing, which was led me to suspect the sample had an issue.

Although there still may be value in aligning the meaning of RGB565 with implementations present on hardware and in LVGL (ie little endian), the documentation is currently consistent, so I am going to close this issue.

@jeronimoagullo I am not aware of what display you are using, but you may want to investigate changing the pixel format with [display_set_pixel_format](https://docs.zephyrproject.org/latest/hardware/peripherals/display/index.html#c.display_set_pixel_format)

closed 2023-01-10 by danieldegrasse

jeronimoagullo, 2023-01-11:
Hi @danieldegrasse 

Thank you very much for the clarification. As you said, it is correct. I have read also in the documentation that Zephyr uses big endian for all formats that need more than 1 byte. It seems that the display that I am using or the specific driver (I mean, the host system) uses little endian instead of big endian.

In any case, I didn't use the LVGL library so far, thus, I will try it out setting the `LV_COLOR_16_SWAP` macro as you have said.

Thank you very much

djixon, 2023-11-26:
Guys, you ARE making mistake!!! There are RGB565 and BGR565 formats and that has nothing to do with byte order (endianness). In last LVGL 8.3 version there is mess with those two. SWAPPING is just changing order in channel order,  so without swapping it is RGB565. When swapping is ON, only R and B channels exchange positions so you have BGR565 format. In both cases, direction is from highest to lowest significant bit as in names RGB or BGR. Once 16bit value is formed then you apply endianness (which can be any little-endian or big-endian) and has nothing to do with channel swapping introduced to support both RGB565 and BGR565 formats