zephyrproject-rtos/zephyr #53654: SPI2 not working on STM32L412

kimbotto, 2023-01-10:
Dear all, SPI2 seems not to work on STM32L412.
I tried the spi_flash sample on a custom board based on L412 and it works fine if I use SPI1 but fails if I use SPI2.
Specifically,

1 - the mcu sends the first byte to detect if memory is ready (0x05)
2 - the mcu enter an infinite loop in function spi_stm32_shift_m, file spi_ll_stm32.c, lines 291-293

```
	while (!ll_func_rx_is_not_empty(spi)) {
		/* NOP */
	}
```

Since it works correctly with SPI1, I checked possible differences among the registers.
It turned out that the only difference was CR2 Register, Bit 12 FRXTH: FIFO reception threshold:

SPI2 was 0: RXNE event is generated if the FIFO level is greater than or equal to 1/2 (16-bit)
SPI1 was 1: RXNE event is generated if the FIFO level is greater than or equal to 1/4 (8-bit)

Since the data width is 8bit, I expected for both to have such a bit set.

FRXTH is set in the file spi_ll_stm32.h (lines 189-207) inside a define block

```
#if DT_HAS_COMPAT_STATUS_OKAY(st_stm32_spi_fifo)
..
#end
```

Unfortunately, SPI2 in file stm32l412.dtsi hasn't got such compatibility

```
		spi2: spi@40003800 {
			compatible = "st,stm32-spi";
			#address-cells = <1>;
			#size-cells = <0>;
			reg = <0x40003800 0x400>;
			interrupts = <36 5>;
			clocks = <&rcc STM32_CLOCK_BUS_APB1 0x00004000>;
			status = "disabled";
		};
```

In fact, modifying such file with 

``` compatible = "st,stm32-spi-fifo", "st,stm32-spi"; ```

fixes the problem.
Is there a reason for that peripheral not to have such compatibility? 
Am I the only one having problem with such a mcu?
Best,

Mirko

gautierg-st, 2023-01-11:
Hello,
SPI1 and SPI2 are functionally identical so there is no reason to not have the fifo on SPI2. It seems like an oversight.
Can you make a PR for this change?

kimbotto, 2023-01-11:
Hi, I have just done it, PR #53705

FRASTM, 2023-01-12:
Why closing the PR ?

kimbotto, 2023-01-12:
According to the slides that erwango shared with me, creating a second commit to fix the first was not a good idea.
So I created a new PR with a correct commit.
If I did wrong, sorry about that, it is my first contribution.

FRASTM, 2023-01-12:
No pb, 
https://github.com/zephyrproject-rtos/zephyr/pull/53735 is the new one

closed 2023-01-13 by FRASTM