zephyrproject-rtos/zephyr #53663: Error while trying to flash nucleo_f446re: TARGET: stm32f4x.cpu - Not halted

moritz-geier, 2023-01-10:
**Bug Description**
I'm trying out zephyr for the first time, and I followed the [getting started guide](https://docs.zephyrproject.org/latest/develop/getting_started/index.html) along when arrived at the last step, the flashing I got two errors, the first one was that I hadn't installed openocd, since I'm on Windows I installed it via choco, now openocd started and outputs:
```
(.venv) C:\zephyrproject\zephyr>west flash
-- west flash: rebuilding
ninja: no work to do.
-- west flash: using runner openocd
-- runners.openocd: Flashing file: C:/zephyrproject/zephyr/build/zephyr/zephyr.hex
Open On-Chip Debugger 0.11.0 (2021-11-18) [https://github.com/sysprogs/openocd]
Licensed under GNU GPL v2
libusb1 09e75e98b4d9ea7909e8837b7a3f00dda4589dc3
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 2000 kHz
Info : STLINK V2J39M27 (API v2) VID:PID 0483:374B
Info : Target voltage: 3.259513
Warn : target stm32f4x.cpu examination failed
Info : starting gdb server for stm32f4x.cpu on 3333
Info : Listening on port 3333 for gdb connections
    TargetName         Type       Endian TapName            State
--  ------------------ ---------- ------ ------------------ ------------
 0* stm32f4x.cpu       hla_target little stm32f4x.cpu       unknown

Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
TARGET: stm32f4x.cpu - Not halted
FATAL ERROR: command exited with status 1: C:/ProgramData/chocolatey/bin/openocd.exe -s
'C:/zephyrproject/zephyr/boards/arm/stm32f411e_disco\support' -s 'C:/Program Files/zephyr-sdk-0.15.2/sysroots/AMD64-pokysdk-
linux/usr/share/openocd/scripts' -f 'C:/zephyrproject/zephyr/boards/arm/stm32f411e_disco\support\openocd.cfg' '-c init'
 '-c targets' - c 'reset init' -c 'flash write_image erase C:/zephyrproject/zephyr/build/zephyr/zephyr.hex' -c 'reset init' 
-c 'verify_image C:/zephyrproject/zephyr/build/zephyr/zephyr.hex' -c 'reset run' -c shutdown
```
I searched a bit but all I found was this [Issue #50306](https://github.com/zephyrproject-rtos/zephyr/issues/50306). Which didn't really help. I tried to compare the openocd.cfg file of the f466re and the one mentioned in the Issue but they are the same and since I'm new to zephyr I don't really know how to solve this, I know there are many easy work around, but I'd like to use west if possible.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p auto -b nucleo_f446re samples/basic/blinky
2. west flash

**Expected behavior**
To be able to flash the Nucleo F446RE with `west flash`

**Impact**
annoyance

**Environment (please complete the following information):**
 - OS: Windows Build 19044
 - Toolchain Zephyr SDK 15.1
 - openocd 0.11.0 via choco

moritz-geier, 2023-01-10:
Curiosly its working now, I tried to build for one of my other boards, the stm32f411e_disco, but I couldn't flash it because it seems to be broken, so I returned back to the nucleo board and `west flash` instantly worked. I didn't install or update anything between the failed flashes and the successful ones, not even closed the terminal, seems wierd

erwango, 2023-01-10:
In the log provided, I can read `'C:/zephyrproject/zephyr/boards/arm/stm32f411e_disco\support`.
If your board is a `nucleo_f446re`, it would be better using matching target.

This being said. That won't solve the issue you're seeing.
Can you try updating ST-Link firmware (using STM32CubeProgrammer or ST-Link utility) ?

moritz-geier, 2023-01-10:
Oh sorry, it seems i uploaded the wrong log:
```
(.venv) C:\zephyrproject\zephyr>west flash
-- west flash: rebuilding
ninja: no work to do.
-- west flash: using runner openocd
-- runners.openocd: Flashing file: C:/zephyrproject/zephyr/build/zephyr/zephyr.hex
Open On-Chip Debugger 0.11.0 (2021-11-18) [https://github.com/sysprogs/openocd]
Licensed under GNU GPL v2
libusb1 09e75e98b4d9ea7909e8837b7a3f00dda4589dc3
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 2000 kHz
Info : STLINK V2J39M27 (API v2) VID:PID 0483:374B
Info : Target voltage: 3.253193
Warn : target stm32f4x.cpu examination failed
Info : starting gdb server for stm32f4x.cpu on 3333
Info : Listening on port 3333 for gdb connections
    TargetName         Type       Endian TapName            State
--  ------------------ ---------- ------ ------------------ ------------
 0* stm32f4x.cpu       hla_target little stm32f4x.cpu       unknown

Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
TARGET: stm32f4x.cpu - Not halted
FATAL ERROR: command exited with status 1: openocd -s 'C:/zephyrproject/zephyr/boards/arm/nucleo_f446re\support' -f 'C:/zephyrproject/zephyr/boards/arm/nucleo_f446re\support\openocd.cfg' '-c init' '-c targets' -c 'reset init' -c 'flash write_image erase C:/zephyrproject/zephyr/build/zephyr/zephyr.hex' -c 'reset init' -c 'verify_image C:/zephyrproject/zephyr/build/zephyr/zephyr.hex' -c 'reset run' -c shutdown
```
What I tried was draging the zephyr.bin file into the ST-Link folder which mounts on connecting the board, and that flashes successfully.

erwango, 2023-01-10:
> What I tried was draging the zephyr.bin file into the ST-Link folder which mounts on connecting the board, and that flashes successfully.

I would try to update ST-Link fw anyway.

nordicjm, 2023-01-11:
I see this all the time on nucleo boards, flashing the first time fails, flashing the second time works. Boards I have lying around are nucleo f401re, nucleo f746zg

moritz-geier, 2023-01-11:
I had to try even more time than twice to get it working, and as i said trying to flash with anonher board first seemed to help

closed 2023-01-11 by moritz-geier

erwango, 2023-01-11:
> I see this all the time on nucleo boards, flashing the first time fails, flashing the second time works. Boards I have lying around are nucleo f401re, nucleo f746zg

This is an actual issue. Likely due to the openocd version you're using. The main issue that was fixed around that is https://github.com/zephyrproject-rtos/zephyr/issues/50590.
I've just verify on the 2 boards you mentioned, first flash is successful.