zephyrproject-rtos/zephyr #53674: net: lwm2m: senml cbor formatter relying on implementation detail / inconsistency of lwm2m_path_to_string

broglep-work, 2023-01-10:
In order for SenML CBOR format to work, the base name and name need to adhere to special handling of leading and trailing slashes so that when base name and name are concatenated, a proper lwm2m path is the result (no double or missing slashes) 

A the moment, the SenML CBOR relies on the fact that `lwm2m_path_to_string` produces leading & trailing slashes for paths with size 1 & 2, but NO trailing slashes for paths with size 3 & 4.

Changing the implementation of `lwm2m_path_to_string` to be consistent (as one would normally expect from a general purpose helper method), breaks the SenML CBOR format.

SenML CBOR formatter should not rely on such details and must itself implement the special handling for path strings when they are used as SenML base name or names

broglep-work, 2023-01-10:
The problem can be traced back to #43662. It appears that the special path sting handling was extracted and deemed general purpose and was re-used at some other places

SeppoTakalo, 2023-01-13:
Thanks. I'll take  this into our backlog and assign someone to work with it.

jarlamsa, 2023-01-19:
Now as the https://github.com/zephyrproject-rtos/zephyr/pull/53771 has been merged, the `lwm2m_rw_senml_cbor.c` is the only caller for `lwm2m_path_to_string` inside zephyr asides from `lwm2m_observation.c` unit tests which should be converted to use the new APIs as well. Should the whole function now be moved to be a static one inside the `lwm2m_rw_senml_cbor`?

carlescufi, 2023-02-08:
@SeppoTakalo @jarlamsa any progress on this issue?

SeppoTakalo, 2023-02-08:
Minor progress. `lwm2m_path_to_string()` is used only inside SenML CBOR content formatter anymore, so this issue has been isolated. LwM2M API does not use string formats anymore to describe a resource paths.

Next step for us would be to move this function inside the SenML content format C file, so it is not a generic helper function anymore and may freely contain tweaks specific to one content.

PR #56606 (2023-04-06): net: lwm2m: Remove lwm2m_path_to_string
This function had only one use in SenML CBOR formatter and it contained some specific tweaks, so move the function to be a static member of that module.

Fixes #53674
commit 3a241592 (2023-04-06): net: lwm2m: Remove lwm2m_path_to_string

This function had only one use in SenML CBOR formatter and it
contained some specific tweaks, so move the function to be a
static member of that module.

Fixes #53674
closed 2023-04-06 by carlescufi