zephyrproject-rtos/zephyr #53680: HawkBit Metadata Error

liamclarkza, 2023-01-10:
**Describe the bug**

I am currently trying to perform OTA updates using HawkBit on a custom board with an STM32L471RGT6 microcontroller and a SIMCOM SIM800C GSM modem (which I use to connect to the internet using a ppp connection). Upon establishing a network connection, the device registers with a HawkBit server I am running in the cloud and polls for an update at which stage I receive the following error in the console:
```
[00:00:34.110,000] <inf> hawkbit: Polling target data from Hawkbit
[00:00:35.524,000] <err> hawkbit: HTTP response len mismatch, expected 0, got 174
[00:00:35.530,000] <inf> hawkbit: Metadata error
```

I can verify that the device connects to the server as it appears as a target on the HawkBit dashboard:
[image]

Unfortunately, even with HawkBit's log level set to debug, there isn't any additional information I have been able to find to diagnose what the issue is. I don't think the issue is hardware related.

**To Reproduce**
Steps to reproduce the behavior:

This may be a bit difficult as I have a custom board; however, the most important things to replicate would be:
1. Flash MCUBoot, I have the partitions laid out according to the following in my devicetree:
```cpp
partitions {
		compatible = "fixed-partitions";
		#address-cells = <1>;
		#size-cells = <1>;

		boot_partition: partition@0 {
			label = "mcuboot";
			reg = <0x00000000 DT_SIZE_K(48)>;
			read-only;
		};
		storage_partition: partition@C000 {
			label = "storage";
			reg = <0x0000C000 DT_SIZE_K(48)>;
		};
		slot0_partition: partition@18000 {
			label = "image-0";
			reg = <0x00018000 DT_SIZE_K(416)>;
		};
		slot1_partition: partition@80000 {
			label = "image-1";
			reg = <0x00080000 DT_SIZE_K(416)>;
		};
		scratch_partition: partition@e8000 {
			label = "image-scratch";
			reg = <0x000e8000 DT_SIZE_K(96)>;
		};
		
	};
```

2. Setup your prj.conf as follows:
```
# GSM modem support
CONFIG_MODEM=y
CONFIG_MODEM_GSM_PPP=y
CONFIG_MODEM_GSM_APN="<your APN here>"
CONFIG_MODEM_GSM_SIMCOM=y
CONFIG_MODEM_GSM_FACTORY_RESET_AT_BOOT=y

# PPP networking support
CONFIG_NET_DRIVERS=y
CONFIG_NET_PPP=y
CONFIG_NET_L2_PPP=y
CONFIG_NETWORKING=y
CONFIG_NET_L2_PPP_TIMEOUT=60000
CONFIG_NET_L2_PPP_DELAY_STARTUP_MS=7500
# Network management events
CONFIG_NET_CONNECTION_MANAGER=y
CONFIG_NET_MGMT=y
CONFIG_NET_MGMT_EVENT=y
# IPv4 enables PPP IPCP support
CONFIG_NET_IPV4=y
CONFIG_NET_IPV6=n

#Shell commands:
CONFIG_SHELL=y
CONFIG_KERNEL_SHELL=y
CONFIG_NET_SHELL=y
CONFIG_MODEM_SHELL=y
CONFIG_MCUBOOT_SHELL=y
CONFIG_FLASH_SHELL=y
CONFIG_SHELL_STACK_SIZE=4096

# Bootloader
CONFIG_BOOTLOADER_MCUBOOT=y
CONFIG_MCUBOOT_EXTRA_IMGTOOL_ARGS="--version 1.0.0"
CONFIG_MCUBOOT_GENERATE_CONFIRMED_IMAGE=y
CONFIG_MCUBOOT_SIGNATURE_KEY_FILE="~/zephyrproject/bootloader/mcuboot/root-rsa-2048.pem"
# Image management
CONFIG_IMG_MANAGER=y
CONFIG_MCUBOOT_IMG_MANAGER=y

# Hawkbit
CONFIG_HAWKBIT=y
CONFIG_HAWKBIT_SHELL=y
CONFIG_MAIN_STACK_SIZE=4096
#Minimal Heap mem pool size for the Hawkbit working
CONFIG_HEAP_MEM_POOL_SIZE=4096
CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=4096
CONFIG_HAWKBIT_POLLING=y
CONFIG_HAWKBIT_POLL_INTERVAL=1
CONFIG_HAWKBIT_SERVER="<your ip address for hawkbit here>"

# Logging config:
CONFIG_LOG=y
CONFIG_SHELL_LOG_BACKEND=y

# Debug Logs:
CONFIG_HAWKBIT_LOG_LEVEL_DBG=y

#General Config
CONFIG_NEWLIB_LIBC=y
CONFIG_LOG_BUFFER_SIZE=8192
```

3. Use the following code in your main.c:
```cpp
#include <zephyr/kernel.h>
#include <zephyr/sys/printk.h>
#include <zephyr/sys/reboot.h>
#include <zephyr/shell/shell.h>
#include <zephyr/drivers/uart.h>
#include <zephyr/net/net_mgmt.h>
#include <zephyr/net/net_event.h>
#include <zephyr/net/net_conn_mgr.h>
#include <zephyr/drivers/modem/gsm_ppp.h>
#include <zephyr/devicetree.h>
#include <zephyr/drivers/gpio.h>
#include <zephyr/logging/log.h>
#include <zephyr/dfu/mcuboot.h>
#include <zephyr/mgmt/hawkbit.h>

LOG_MODULE_REGISTER(sample_gsm_ppp, LOG_LEVEL_DBG);

#define GSM_MODEM_NODE DT_COMPAT_GET_ANY_STATUS_OKAY(zephyr_gsm_ppp)
#define UART_NODE DT_BUS(GSM_MODEM_NODE)

static const struct device *const gsm_dev = DEVICE_DT_GET(GSM_MODEM_NODE);
static struct net_mgmt_event_callback mgmt_cb;
static bool starting = IS_ENABLED(CONFIG_GSM_PPP_AUTOSTART);
static const struct gpio_dt_spec gsm_reg_en = GPIO_DT_SPEC_GET(DT_NODELABEL(gsm_reg_en), gpios);
static bool first_connect = true;

static void event_handler(struct net_mgmt_event_callback *cb, uint32_t mgmt_event, struct net_if *iface) {
    ARG_UNUSED(cb);
    ARG_UNUSED(iface);

    if ((mgmt_event & (NET_EVENT_L4_CONNECTED | NET_EVENT_L4_DISCONNECTED)) != mgmt_event) {
        return;
    }

    if (mgmt_event == NET_EVENT_L4_CONNECTED) {
        LOG_INF("Network connected");
        if (first_connect) {
            LOG_INF("Starting hawkbit polling mode");
	        hawkbit_autohandler();
            first_connect = false;
        }
        return;
    }

    if (mgmt_event == NET_EVENT_L4_DISCONNECTED) {
        LOG_INF("Network disconnected");
        return;
    }
}

static void modem_on_cb(const struct device *dev, void *user_data) {
    ARG_UNUSED(dev);
    ARG_UNUSED(user_data);
    LOG_INF("GSM modem on callback fired");
}

static void modem_off_cb(const struct device *dev, void *user_data) {
    ARG_UNUSED(dev);
    ARG_UNUSED(user_data);
    LOG_INF("GSM modem off callback fired");
}

int main(void) {
    const struct device *const uart_dev = DEVICE_DT_GET(UART_NODE);
    gpio_pin_configure_dt(&gsm_reg_en, GPIO_OUTPUT_INACTIVE);
    k_msleep(1000);
    gpio_pin_set_dt(&gsm_reg_en, 1);

    /* Register modem power callbacks */
    gsm_ppp_register_modem_power_callback(gsm_dev, modem_on_cb, modem_off_cb, NULL);

    LOG_INF("Board '%s' APN '%s' UART '%s' device %p (%s)", CONFIG_BOARD, CONFIG_MODEM_GSM_APN, uart_dev->name, uart_dev, gsm_dev->name);

    net_mgmt_init_event_callback(&mgmt_cb, event_handler, NET_EVENT_L4_CONNECTED | NET_EVENT_L4_DISCONNECTED);
    net_mgmt_add_event_callback(&mgmt_cb);

    int ret = -1;
    ret = hawkbit_init();
	if (ret < 0) {
		LOG_ERR("Failed to init hawkbit");
	}
    return 0;
}

``` 

**Expected behavior**
I expect that when HawkBit polls for an update that it does so successfully without the "Metadata error".

**Impact**
What impact does this issue have on your progress: **showstopper**

**Logs and console output**
```
[00:00:02.001,000] <inf> sample_gsm_ppp: Board 'sensify_mcu' APN 'Justworx' UART 'serial@40004400' device 0x803a14c (gsm-modem)
[00:00:02.005,000] <inf> fs_nvs: 3 Sectors of 2048 bytes
[00:00:02.005,000] <inf> fs_nvs: alloc wra: 0, 7e8
[00:00:02.005,000] <inf> fs_nvs: data wra: 0, 0
[00:00:02.005,000] <dbg> hawkbit: hawkbit_init: Action id: current 536890120
[00:00:02.005,000] <inf> hawkbit: Image is confirmed OK
[00:00:07.290,000] <inf> modem_gsm: Manufacturer: SIMCOM_Lt
[00:00:07.344,000] <inf> modem_gsm: Model: SIMCOM_SIM800C
[00:00:07.401,000] <inf> modem_gsm: Revision: Revision:1418B08SIM800C24_BT
[00:00:07.456,000] <inf> modem_gsm: IMEI: xxxxxxxxxxxxxxxxx
[00:00:07.511,000] <inf> modem_gsm: IMSI: xxxxxxxxxxxxxxxxx
[00:00:07.567,000] <inf> modem_gsm: ICCID: xxxxxxxxxxxxxxxxx
[00:00:32.750,000] <inf> net_ppp: Initializing PPP to use serial@40004400
[00:00:33.515,000] <inf> sample_gsm_ppp: Network connected
[00:00:33.515,000] <inf> sample_gsm_ppp: Starting hawkbit polling mode
[00:00:34.110,000] <inf> hawkbit: Polling target data from Hawkbit
[00:00:35.524,000] <err> hawkbit: HTTP response len mismatch, expected 0, got 174
[00:00:35.530,000] <inf> hawkbit: Metadata error
[00:05:36.885,000] <inf> hawkbit: Polling target data from Hawkbit
[00:05:40.781,000] <err> hawkbit: HTTP response len mismatch, expected 0, got 174
[00:05:40.787,000] <inf> hawkbit: Metadata error
[00:10:42.071,000] <inf> hawkbit: Polling target data from Hawkbit
[00:10:43.670,000] <err> hawkbit: HTTP response len mismatch, expected 0, got 174
[00:10:43.675,000] <inf> hawkbit: Metadata error
...
```

**Environment (please complete the following information):**
 - OS: MacOS
 - Toolchain: Zephyr SDK
 - Commit SHA or Version used: zephyr-v3.2.0

ycsin, 2023-01-16:
Thanks for the report. I'm not sure why and how, but it seems like the 174 bytes data that the device received is a complete payload (since it is `HTTP_DATA_FINAL`) but somehow doesn't set the `http_content_size` in [L651](https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/mgmt/hawkbit/hawkbit.c#L651), probably you can set a breakpoint over there to examine the payload, and hopefully leads you to something?

liamclarkza, 2023-01-16:
Thanks for the advice @ycsin - I'm actually struggling to debug this. Using west debug with MCUboot doesn't seem to work (or at least I haven't managed to get it to work yet). Do you have any recommendations on how to debug this?

ycsin, 2023-01-20:
@liamclarkza I only had experience with debugging using cortex debug (a vscode's pluggin), I imagine that I probably would a sleep delay at the beginning of the software and 'attach' to the target, so that jumping from the bootloader to application won't diconnects the debugger and the delay gives me enough time to attach to the mcu before the hawkbit part runs, or config the hawkBit such that it runs only when you enter the shell command

ycsin, 2023-01-31:
@liamclarkza not sure if you had any findings about the issue?

liamclarkza, 2023-02-02:
@ycsin I have managed to get it working - I was trying some of the networking examples and found that I was having issues with net_pkt buffers which led me down the road of changing buffer sizes and timeouts to get the networking samples going with my GSM modem. Once I had that fixed, Hawkbit worked fine. The changes I added to my prj.conf were:
```
# fix for <err> net_tcp: net_pkt alloc failure
CONFIG_NET_PKT_RX_COUNT=16
CONFIG_NET_PKT_TX_COUNT=16
CONFIG_NET_BUF_RX_COUNT=64
CONFIG_NET_BUF_TX_COUNT=64

# fix for <err> uart_mux: Ring buffer full, drop 1 bytes
CONFIG_UART_MUX_RINGBUF_SIZE=1024

# fix for <err> net_ppp: Rx buffer doesn't have enough space. Bytes pending: 8, written: 0
CONFIG_NET_PPP_RINGBUF_SIZE=1024

# fix for <err> OTA_LOGGER: Socket failed to connect (-116) 
CONFIG_NET_SOCKETS_CONNECT_TIMEOUT=60000
```

Thanks for the help.

liamclarkza, 2023-02-07:
@ycsin It seems that unfortunately the issue has come back even with the adjustments made above.
I checked out the line below which should be setting `http_content_size` as you mentioned but `rsp->content_length` is 0: https://github.com/zephyrproject-rtos/zephyr/blob/802600da7b0396ee515ddf68abf69d95ba33ecf5/subsys/mgmt/hawkbit/hawkbit.c#L651

Seems like there is an issue where the content length is not being set by the http client.

closed 2023-04-24 by github-actions[bot]

mkabaoglu, 2024-02-19:
Hi i am getting the same error any progress on this @liamclarkza 
[image]