zephyrproject-rtos/zephyr #53697: Can not run the usb mass storage demo on NXP mimxrt595_evk_cm33

hvbon2010, 2023-01-11:
**Describe the bug**
I built and ran the usb mass storage demo with zephyr v3.2.0 on mimxrt595_evk_cm33, but when I connect to a Linux host, the PC can not mount to this USB.
I tested this demo successfully on other evaluation board (nrf52840dk_nrf52840). So, I think this issue from the usb driver, not from Zephyr usb stack.

**To Reproduce**
Steps to reproduce the behavior:
1. build: `west build -b mimxrt595_evk_cm33 zephyr/samples/subsys/usb/mass --pristine`
2. connect usb port to Linux host.

**Logs and console output**
dmesg:
```
usb 9-6.4: new high-speed USB device number 20 using xhci_hcd
usb 9-6.4: config 1 interface 0 altsetting 0 bulk endpoint 0x81 has invalid maxpacket 64
usb 9-6.4: config 1 interface 0 altsetting 0 bulk endpoint 0x1 has invalid maxpacket 64
usb 9-6.4: New USB device found, idVendor=2fe3, idProduct=0008, bcdDevice= 3.02
usb 9-6.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
usb 9-6.4: Product: Zephyr MSC sample
usb 9-6.4: Manufacturer: ZEPHYR
usb 9-6.4: SerialNumber: 1B3D2BB2B8434626
usb-storage 9-6.4:1.0: USB Mass Storage device detected
scsi host2: usb-storage 9-6.4:1.0
scsi 2:0:0:0: Direct-Access     ZEPHYR   ZEPHYR USB DISK  0.01 PQ: 0 ANSI: 0 CCS
sd 2:0:0:0: Attached scsi generic sg0 type 0
sd 2:0:0:0: [sda] 64 512-byte logical blocks: (32.8 kB/32.0 KiB)
sd 2:0:0:0: [sda] Write Protect is off
sd 2:0:0:0: [sda] Mode Sense: 03 00 00 00
sd 2:0:0:0: [sda] No Caching mode page found
sd 2:0:0:0: [sda] Assuming drive cache: write through
usb 9-6.4: reset high-speed USB device number 20 using xhci_hcd
usb 9-6.4: reset high-speed USB device number 20 using xhci_hcd
usb 9-6.4: reset high-speed USB device number 20 using xhci_hcd
usb 9-6.4: reset high-speed USB device number 20 using xhci_hcd
usb 9-6.4: reset high-speed USB device number 20 using xhci_hcd
usb 9-6.4: reset high-speed USB device number 20 using xhci_hcd
sd 2:0:0:0: [sda] Unaligned partial completion (resid=4032, sector_sz=512)
sd 2:0:0:0: [sda] tag#0 CDB: Read(10) 28 00 00 00 00 00 00 00 08 00
sd 2:0:0:0: [sda] tag#0 FAILED Result: hostbyte=DID_ERROR driverbyte=DRIVER_OK cmd_age=1s
sd 2:0:0:0: [sda] tag#0 CDB: Read(10) 28 00 00 00 00 00 00 00 08 00
blk_update_request: I/O error, dev sda, sector 0 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
Buffer I/O error on dev sda, logical block 0, async page read
```
```
$lsblk
sda           8:0    1    32K  0 disk
```
```
$sudo fdisk -l /dev/sda
fdisk: cannot open /dev/sda: Input/output error
```
```
$sudo mkdir -p /media/usb_zephyr
$sudo mount /dev/sda /media/usb_zephyr
mount: /media/usb_zephyr: can't read superblock on /dev/sda.
```

**Environment (please complete the following information):**
 - OS: Linux (Ubuntu 22.04.1 LTS).
 - Toolchain: Zephyr SDK 0.15.1

desowin, 2023-01-16:
> usb 9-6.4: config 1 interface 0 altsetting 0 bulk endpoint 0x81 has invalid maxpacket 64
usb 9-6.4: config 1 interface 0 altsetting 0 bulk endpoint 0x1 has invalid maxpacket 64

This suggests the issue is related to lack of `CONFIG_USB_DC_HAS_HS_SUPPORT`. Enabling `CONFIG_USB_DC_HAS_HS_SUPPORT` should help.

DerekSnell, 2023-01-16:
Hi @hvbon2010 ,
@desowin's suggestion to select `CONFIG_USB_DC_HAS_HS_SUPPORT` works for me when testing this.  I tried the same sample on the MIMXRT595-EVK board with Ubuntu using Zephyr v3.2.0 (and this also works with the latest main branch at c819f0f8d63bee3b1a7cc446a16e2e9db46739ac).  This is my build command:
```
west build -b mimxrt595_evk_cm33 samples/subsys/usb/mass --pristine -- -DCONFIG_USB_DC_HAS_HS_SUPPORT=y
```

Here is the terminal output when testing on Ubuntu:
```
$ sudo dmesg
[18409.064282] usb 3-3: new high-speed USB device number 12 using xhci_hcd
[18409.213558] usb 3-3: New USB device found, idVendor=2fe3, idProduct=0008, bcdDevice= 3.02
[18409.213569] usb 3-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[18409.213572] usb 3-3: Product: Zephyr MSC sample
[18409.213575] usb 3-3: Manufacturer: ZEPHYR
[18409.213578] usb 3-3: SerialNumber: A3E8E0B9718A6B31
[18409.215919] usb-storage 3-3:1.0: USB Mass Storage device detected
[18409.216406] scsi host1: usb-storage 3-3:1.0
[18410.233402] scsi 1:0:0:0: Direct-Access     ZEPHYR   ZEPHYR USB DISK  0.01 PQ: 0 ANSI: 0 CCS
[18410.233935] sd 1:0:0:0: Attached scsi generic sg0 type 0
[18410.234670] sd 1:0:0:0: [sdb] 64 512-byte logical blocks: (32.8 kB/32.0 KiB)
[18410.234882] sd 1:0:0:0: [sdb] Write Protect is off
[18410.234886] sd 1:0:0:0: [sdb] Mode Sense: 03 00 00 00
[18410.235077] sd 1:0:0:0: [sdb] No Caching mode page found
[18410.235083] sd 1:0:0:0: [sdb] Assuming drive cache: write through
[18410.247648] sd 1:0:0:0: [sdb] Attached SCSI removable disk

$ sudo fdisk -l /dev/sdb 
Disk /dev/sdb: 32 KiB, 32768 bytes, 64 sectors
Disk model: ZEPHYR USB DISK 
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes

$ sudo mkdir /media/usb_drive/
$ sudo mount /dev/sdb /media/usb_drive/
mount: /media/usb_drive: wrong fs type, bad option, bad superblock on /dev/sdb, missing codepage or helper program, or other error.

$ sudo mkfs.minix /dev/sdb
32 inodes
32 blocks
Firstdatazone=5 (5)
Zonesize=1024
Maxsize=268966912

$ cd /media/usb_drive/
$ sudo nano test_file.txt
$ cat test_file.txt 
hello from file
```

hvbon2010, 2023-01-17:
Hi @DerekSnell , @desowin,

Thanks for your suggestions, after adding `CONFIG_USB_DC_HAS_HS_SUPPORT`, this sample is worked!

jfischer-no, 2023-01-17:
This issue is not resolved and is a bit annoying. STM32_USBHS and SAM_USBHS have a solution to pull maximum-speed to Kconfig, https://github.com/zephyrproject-rtos/zephyr/blob/122c7be703ebf1d1f78f20fc7936ea263a82d345/drivers/usb/device/Kconfig#L90 and https://github.com/zephyrproject-rtos/zephyr/blob/122c7be703ebf1d1f78f20fc7936ea263a82d345/drivers/usb/device/Kconfig#L55, unfortunately NXP uses different nodelabels for USB device controllers, but something like that should be done to fix this issue..

mmahadevan108, 2023-01-18:
We will submit a PR shortly.

PR #53903 (2023-01-24): drivers: usb: Enable High speed Kconfig for NXP controllers
Enable the USB_DC_HAS_HS_SUPPORT Kconfig for NXP controllers that support High speed.

Fixes #53697
closed 2023-01-24 by laurenmurphyx64