zephyrproject-rtos/zephyr #53715: mec15xxevb_assy6853: broken UART console output

golowanow, 2023-01-11:
**Describe the bug**
mec15xxevb_assy6853 rev.A broken (binary) output at its UART console.
Bisect points to commit c7e0d727 introduced with #52303 

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p always -b mec15xxevb_assy6853 samples/hello_world
2. flash the board as usual
3. check console output at another shell with e.g. `picocom -b 115200 /dev/ttyUSB1` 

**Expected behavior**
correct console output

**Impact**
all our test cases fail to get expected console output

**Logs and console output**
corrupted output (with commit c7e0d727)
[image]
correct output (for previous commit)
[image]

**Environment (please complete the following information):**
 - OS: Fedora Linux kernel 5.10.12-200.fc33.x86_64
 - Toolchain zephyr-sdk-0.15.0
 - Commit c7e0d727

**Additional context**
n/a

golowanow, 2023-01-12:
@jvasanth1, could you please have a look why this issue happens after your recent #52303 ?

jvasanth1, 2023-01-16:
@golowanow we are unable to reproduce the issue so far. Can you let us know the MEC15xx silicon version on your board?

golowanow, 2023-01-16:
> @golowanow we are unable to reproduce the issue so far. Can you let us know the MEC15xx silicon version on your board?

do you mean `mec15xxevb_assy6853 rev.A` as 'silicon version' of the board or something else (then where to read what you need) ?

jvasanth1, 2023-01-16:
It should be written on the main silicon (soc chip) on the EVB below the string MICROCHIP

golowanow, 2023-01-16:
@jvasanth1 can it be `144WFBGA` ?

jvasanth1, 2023-01-16:
it should be something beginning with MEC15

golowanow, 2023-01-16:
> it should be something beginning with MEC15

aha, `MEC1501`

jvasanth1, 2023-01-16:
great, Thanks!

golowanow, 2023-01-16:
@jvasanth1, I've also noticed it is `rev.B` not `rev.Ą`
[image]

golowanow, 2023-01-16:
> I think you are looking at the wrong place. The chip will be square like this image. The example is a MEC1521 (written below the microchip logo)

@jvasanth1 in our case the chip is on CPU daughter board and it has pretty weak label. Not sure we are able to read it.
Is there a way to read silicon version another way, e.g. from registers ?

jvasanth1, 2023-01-16:
can you provide us the value of the 32-bit register at 0x400FFF1C?

golowanow, 2023-01-17:
> it should be something beginning with MEC15

@jvasanth1 

[image]

jvasanth1, 2023-01-17:
@golowanow can you try the draft PR to see if it solves the problem
https://github.com/zephyrproject-rtos/zephyr/pull/53869

jvasanth1, 2023-01-17:
Thanks @golowanow 

this PR https://github.com/zephyrproject-rtos/zephyr/pull/53874 should address the issue

golowanow, 2023-01-20:
@jvasanth1, thank you !  I close the issue as resolved.

closed 2023-01-20 by golowanow