zephyrproject-rtos/zephyr #53728: Sensor API documentation: no mention of blocking behaviour

cdfh, 2023-01-12:
**Describe the bug**
The documentation for the Sensor API describes `sensor_sample_fetch()` and `sensor_sample_get()`, but it does not describe whether either function blocks. It isn't obvious from the context whether calling each in turn without waiting is guaranteed to block as necessary to produce results, or whether an error may be produced if `sensor_sample_get()` is called before a value is ready, or whether `sensor_sample_get()` is allowed to take a sample-and-hold approach (i.e., return the most recently received value, regardless of whether it resulted from the most recent call to `sensor_sample_fetch()`). The latter would seem counter intuitive, but it's consistent with the documentation, which states that the function should "Return a useful value for a particular channel, from the driver’s internal data." 

**Expected behavior**
The documentation should explicitly label for both functions whether they block or return immediately. If this is implementation defined, then the documentation should specify whether implementations are required to ensure that `sensor_sample_fetch()` followed by `sensor_sample_get()` will overall block as necessary to produce a result (i.e., because one or the other blocked, which one being left to the implementation to decide). 

Assuming that `sensor_sample_fetch()` is not allowed to take a sample-and-hold approach, the documentation should be more explicit and state that the function "returns a value corresponding to that fetched by the most recent invocation of sensor_sample_fetch()", or something to that effect.

**Impact**
Without knowing the synchronous nature of the functions, I've had to dig into the code of the specific driver I'm using to try and figure it out. If I can't find it out, then I'll need to take an empirical approach to find out. I can't program to the API without knowing.

gmarull, 2023-01-12:
If I'm not mistaken, all Zephyr APIs are, in general, blocking unless specified (e.g. `isr-ok`, `_async` calls, etc.).

cdfh, 2023-01-12:
@nordicjm

> Firstly I think you have the wrong function names

Whoops, many thanks, I've corrected.

> and secondly it does say say this,

Yep, seen that, but it doesn't make it explicit whether it's blocking or not. The most suggestive part is perhaps

> Since the function communicates with the sensor device, it is unsafe to call it in an ISR if the device is connected via I2C or SPI.

However, that's far from clear, especially to a newcomer. It would greatly facilitate rapid understanding if the synchronous/asynchronous nature were stated upfront.

@gmarull 

> If I'm not mistaken, all Zephyr APIs are, in general, blocking unless specified (e.g. isr-ok, _async calls, etc.).

That's very good to know, but in that case I very much believe it would make the API much more approachable if each page stated as much. You've contributed to the Zephyr project, so you're far more knowledgeable than many who will be reading the docs, but yet you mentioned the possibility that you may be mistaken. That's perhaps proof that adding such a note in the docs would be beneficial: if a contributor isn't sure, then a new-comer will certainly not be sure.

gmarull, 2023-01-13:
> That's very good to know, but in that case I very much believe it would make the API much more approachable if each page stated as much. You've contributed to the Zephyr project, so you're far more [...]

PRs welcome :-)

PR #55343 (2023-03-02): drivers: sensor: Document blocking behavior of sample fetch functions
Updates the sensor driver API documentation to explicitly state that sample fetch functions block until the fetch operation is complete.

Fixes #53728
commit a87e4c25 (2023-03-02): drivers: sensor: Document blocking behavior of sample fetch functions

Updates the sensor driver API documentation to explicitly state that
sample fetch functions block until the fetch operation is complete.

Fixes #53728
closed 2023-03-02 by nashif