zephyrproject-rtos/zephyr #53729: Can not build for ESP32 sample program - Zephyr using CMake build

Vaishnavibhavaa, 2023-01-12:
I tried build sample program -Hello world using CMake commands ie., without west . Im getting the below error . I tried setting ESP_IDF_PATH by set command too , even that dint work.

CMake Error at C:/Users/zephyrproject1/zephyr/cmake/modules/extensions.cmake:2090 (message):
  Assertion failed: ESP_IDF_PATH is not set
Call Stack (most recent call first):
  C:/Users/zephyrproject1/zephyr/boards/common/esp32.board.cmake:14 (assert)
  C:/Users/zephyrproject1/zephyr/boards/xtensa/esp32/board.cmake:8 (include)
  C:/Users/zephyrproject1/zephyr/cmake/modules/kernel.cmake:169 (include)
  C:/Users/zephyrproject1/zephyr/cmake/modules/zephyr_default.cmake:122 (include)
  C:/Users/zephyrproject1/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  C:/Users/zephyrproject1/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:97 (include_boilerplate)
  CMakeLists.txt:5 (find_package)

-- Configuring incomplete, errors occurred!
See also "C:/Users/zephyrproject1/zephyr/samples/hello_world/build/CMakeFiles/CMakeOutput.log".
See also "C:/Users/zephyrproject1/zephyr/samples/hello_world/build/CMakeFiles/CMakeError.log".

sylvioalves, 2023-01-12:
@Vaishnavibhavaa, A have a few questions:  
1) What is the command you have used?  
2) Which Zephyr version and SDK are you working with?
3) Any reasons not to use `west`?

Vaishnavibhavaa, 2023-01-13:
Command Used to build - cmake -Bbuild -GNinja -DBOARD=esp32
Zephyr Version - 0.16.0 Latest Zephyr Version
Since we dont want to have dependency with west , we are not using west

sylvioalves, 2023-01-13:
@Vaishnavibhavaa, have you set `export ZEPHYR_TOOLCHAIN_VARIANT="zephyr"`?

Vaishnavibhavaa, 2023-01-18:
Yeah , tried still not working .

Vaishnavibhavaa, 2023-01-18:
PS C:\Users\zephyrproject1\zephyr\samples\hello_world> set ZEPHYR_TOOLCHAIN_VARIANT="zephyr"
PS C:\Users\zephyrproject1\zephyr\samples\hello_world> cmake -Bbuild -GNinja -DBOARD=esp32
Loading Zephyr default modules (Zephyr base (cached)).
-- Application: C:/Users/zephyrproject1/zephyr/samples/hello_world
-- CMake version: 3.25.1
-- Cache files will be written to: C:/Users/BVV1COB/zephyrproject1/zephyr/.cache
-- Zephyr version: 3.2.99 (C:/Users/BVV1COB/zephyrproject1/zephyr)
-- Board: esp32
-- Found host-tools: zephyr 0.15.2 (C:/zephyr/zephyrproject/zephyr-sdk-0.15.2)
-- Found toolchain: zephyr 0.15.2 (C:/zephyr/zephyrproject/zephyr-sdk-0.15.2)
-- Could NOT find Dtc (missing: DTC) (Required is at least version "1.4.6")
-- Found BOARD.dts: C:/Users/zephyrproject1/zephyr/boards/xtensa/esp32/esp32.dts
-- Generated zephyr.dts: C:/Users/zephyrproject1/zephyr/samples/hello_world/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: C:/Users/zephyrproject1/zephyr/samples/hello_world/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: C:/Users/zephyrproject1/zephyr/samples/hello_world/build/zephyr/dts.cmake
Parsing C:/Users/BVV1COB/zephyrproject1/zephyr/Kconfig
Loaded configuration 'C:/Users/zephyrproject1/zephyr/samples/hello_world/build/zephyr/.config'
No change to configuration in 'C:/Users/zephyrproject1/zephyr/samples/hello_world/build/zephyr/.config'
No change to Kconfig header in 'C:/Users/zephyrproject1/zephyr/samples/hello_world/build/zephyr/include/generated/autoconf.h'
CMake Error at C:/Users/zephyrproject1/zephyr/cmake/modules/extensions.cmake:2090 (message):
  Assertion failed: ESP_IDF_PATH is not set
Call Stack (most recent call first):
  C:/Users/zephyrproject1/zephyr/boards/common/esp32.board.cmake:14 (assert)
  C:/Users/zephyrproject1/zephyr/boards/xtensa/esp32/board.cmake:8 (include)
  C:/Users/zephyrproject1/zephyr/cmake/modules/kernel.cmake:169 (include)
  C:/Users/zephyrproject1/zephyr/cmake/modules/zephyr_default.cmake:122 (include)
  C:/Users/zephyrproject1/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  C:/Users/zephyrproject1/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:97 (include_boilerplate)
  CMakeLists.txt:5 (find_package)

-- Configuring incomplete, errors occurred!

sylvioalves, 2023-01-18:
I see, you are working using Windows. I haven't tested that. Would you try the following:  
`-- Could NOT find Dtc (missing: DTC) (Required is at least version "1.4.6")`
1) You are missing dtc, can you check Zephyr's documentation to get your Windows env ready?  
2) Can you set the `ESP_IDF_PATH` to `C:/Users/zephyrproject1/modules/hal/hal_espressif` manually? It should be automatic, but for some reason it isn't working in there.

Vaishnavibhavaa, 2023-01-24:
Hi , 
1.Not able to install dtc - Getting as failure 
_Failures
 _- dtc-msys2 - dtc-msys2 not installed. The package was not found with the source(s) listed.__
 
2.I have set ESP_IDF_PATH to the mentioned location manually and tried still , im facing the same issue , Still its not build.

  _Assertion failed: ESP_IDF_PATH is not set_

Vaishnavibhavaa, 2023-01-24:
[image]

closed 2023-04-09 by github-actions[bot]

PragatiGarg-eaton, 2024-11-04:
Hi @Vaishnavibhavaa have you made any progress on this issue? I’m facing the same problem and have tried several solutions, but nothing seems to be working so far.
cc: @laurenmurphyx64

sylvioalves, 2024-11-04:
@PragatiGarg-eaton can you provide more informations about which revision, SDK, SoC and sample code?

PragatiGarg-eaton, 2024-11-04:
> @PragatiGarg-eaton can you provide more informations about which revision, SDK, SoC and sample code?

I’m working on the zephyr\samples\hello_world sample code for the ESP32 using Zephyr v3.6.0 and the Zephyr SDK version 0.16.5-1. 
After several attempts and a fresh installation of Zephyr, everything is functioning perfectly now! 
Thanks for your prompt response.