zephyrproject-rtos/zephyr #53751: RFC: mgmt/mcumgr: Named Groups

de-nordic, 2023-01-12:
**Is your enhancement proposal related to a problem? Please describe.**

Currently it is possible to provide user defined groups with use of `MGMT_GROUP_ID_PERUSER`
https://github.com/zephyrproject-rtos/zephyr/blob/077273add7d7db411133c74510093448a97ecd78/include/zephyr/mgmt/mcumgr/mgmt/mgmt.h#L77
as a base group and Zephyr uses it to add non-standard MCUmgr groups
https://github.com/zephyrproject-rtos/zephyr/blob/077273add7d7db411133c74510093448a97ecd78/include/zephyr/mgmt/mcumgr/mgmt/mgmt.h#L83
giving them numbers below the `MGMT_GROUP_ID_PERUSER` - this is legacy behavior intended to keep compatibility with existing MCUmgr implementation outside of Zephyr tree.
Problem with using the `MGMT_GROUP_ID_PERUSER` for both purposes is that it also may be used  as base for group identifiers by some vendor provided code, 3rd party library and user application. This causes collision as in the end that is hard to solve, because any change in base of group codes provided by any of the above will also require changes in tools supporting the commands, if there are any provided.

**Describe the solution you'd like**
Solution is to provide named groups and group discovery mechanism.
It would look like that:
 1) Vendors, app, and 3rd party library providers use dynamic group registration functions to register them with use of vendor/group key pair;
 2) system would then allocate group ID, at run-time, for each such group, which also be event subsystem group.
 3) To use named group, the first time request is sent a client software has to do group discovery by sending group ID request to system providing vendor name and group name.
 4) After doing name discovery the obtained group ID will be used by client to do requests, the same way that static groups work.

The difference between static groups and named groups would be additional token being sent with each request. The token is provided on group discovery and should change each time groups are re-organized, and is individual for each group.
The token is needed to make sure that client app is sending requests to correct group, for example in case when system gets restarted and group IDs change do to new registration sequence.

**Describe alternatives you've considered**
Not touching this.

**Additional context**
New group registration functions will be needed to allow "vendor" group discovery registration.
This has to work with callbacks subsystem:
https://github.com/zephyrproject-rtos/zephyr/blob/main/include/zephyr/mgmt/mcumgr/mgmt/callbacks.h
For vendor names the
https://github.com/zephyrproject-rtos/zephyr/blob/main/dts/bindings/vendor-prefixes.txt
registered prefixes could be used to reserve prefixes for vendors, otherwise any name may be used (yes I know that it is possible to make collision to, but why would you make your 3rd party stuff unwanted by using name of somebody else?)

de-nordic, 2023-01-12:
FYI @nordicjm @balaji-nordic

nordicjm, 2023-01-12:
> Problem with using the MGMT_GROUP_ID_PERUSER for both purposes is that it also may be used as base for group identifiers by some vendor provided code, 3rd party library and user application. This [...]

The use of PERUSER is that it is designed for users to set up, so this would be equivalent to 128-bit UUIDs in bluetooth. If you connect to a device, you have registered 16-bit UUID services - and they must work as the specification allows (so this would be groups under 64 for MCUmgr), but you also have 128-bit UUIDs. A device has no clue how to use a 128-bit UUID because it is entirely custom to the vendor, they will add support for it in their custom applications/apps but a generic bluetooth application will be unable to use it, and many vendors can use the same 128-bit UUID (they are not reserved) - this is equivalent to groups 64 and above in MCUmgr. So with this in mind, I'm not understanding this issue or why this would be needed, the registration and ordering of groups >= 64 is down to vendors and their applications, we don't need to worry what they do with those numbers or if multiple vendors reuse the same numbers, because those functions are specific to those vendor's implementations. And vendors can decide if they want to use common custom handlers and keep the same IDs for a  bunch of products, or use different IDs (for the same handlers) for each product, because it would be their firmware/apps that would have to deal with identification of a service and how to use it. Likewise, generic MCUmgr utilities would not have support for these handlers, because they are not specified groups.

de-nordic, 2023-01-12:
> The use of PERUSER is that it is designed for users to set up, so this would be equivalent to 128-bit UUIDs in bluetooth.

blabla TLDR, does not matter in the context.

> So with this in mind, I'm not understanding this issue or why this would be needed, the registration and ordering of groups >= 64 is down to vendors and their applications, we don't need to worry [...]

You use 3rd party chip for mode, and vendor provided SMP commands to update it to include into your app - these take PERUSER + 1 , your app also uses PERUSER + 1. Which one works?

> Likewise, generic MCUmgr utilities would not have support for these handlers, because they are not specified groups.

We already, as also users do, have tools that can generate specific commands. The only generic tool we have now, the mcumgr-cli, is already problematic because it does not support some of commands, or changes to commands, that we have implemented.

nordicjm, 2023-01-12:
> You use 3rd party chip for mode, and vendor provided SMP commands to update it to include into your app - these take PERUSER + 1 , your app also uses PERUSER + 1. Which one works?

You need to fix your code so it uses a unique ID, if you register the same ID then it should fail (not sure if that's been tested but it shouldn't be hard to do that with the iterable section)

de-nordic, 2023-01-12:
> You need to fix your code so it uses a unique ID, if you register the same ID then it should fail (not sure if that's been tested but it shouldn't be hard to do that with the iterable section)

You also have to reflect the changes in client software. When your client software can just ask about group name it simplifies a lot.

nordicjm, 2023-01-12:
> You also have to reflect the changes in client software. When your client software can just ask about group name it simplifies a lot.

You'd need to set it anyway when you create the application, vendor A is not going to be making an app for vendor A's product which also works with vendor B's product.

de-nordic, 2024-02-08:
Lost interest.

closed 2024-02-08 by de-nordic