zephyrproject-rtos/zephyr #53756: esp32 - Wrong value for the default cpu freq -> crash on assert

rillbert, 2023-01-12:
The CONFIG_ESP32_DEFAULT_CPU_FREQ_MHZ is off by a million. For my esp32, this config value expands to 240000000 instead of the expected 240.

This triggers the assert at line 185 in `clk.c`.
```
...
  const uint32_t old_freq_mhz = old_config.freq_mhz;
  const uint32_t new_freq_mhz = CONFIG_ESP32_DEFAULT_CPU_FREQ_MHZ;

  bool res = rtc_clk_cpu_freq_mhz_to_config(new_freq_mhz, &new_config);
  assert(res);
...
```

My guess is that this is an effect of commit zephyrproject-rtos/hal_espressif@1a2292632131d095

**On a side note concerning asserts**
I was in the situation where I had two applications, one that crashed and one that did not. Neither of them had `CONFIG_ASSERT` enabled. 

App 1 (crashed but without any output to the console) - used newlib via `CONFIG_NEWLIB_LIBC=y`
App 2 (did not crash) - did not use newlib.

When enabling `CONFIG_ASSERT` on App 2, it also crashed and it also gave the nice output to the console shown below.

How come that asserts seem to be enabled (and another version of the assert function as well) when using newlib even if CONFIG_ASSERT is not set?

**Target platform**
 - esp32 WROOM module

**To Reproduce**
1. build a sample application and enable asserts
2. run it

Example output:

```
[esp32] [INF] DRAM segment: start=0x1a70, size=0x4d0, vaddr=0x3ffb0000
[esp32] [INF] IRAM segment: start=0x1f40, size=0x3c60, vaddr=0x40080000
[esp32] [INF] start=0x40083ae0
ASSERTION FAIL [res] @ WEST_TOPDIR/modules/hal/espressif/components/esp_system/port/soc/esp32/clk.c:185
```

**Impact**
Not good

**Environment:**
 - OS: Ubuntu 22.04
 - Zephyr SDK 0.15.2
 - Zephyr sha de313d748ad

sylvioalves, 2023-01-12:
@rillbert, that's a known issue and has already been fixed and merged on hal_espressif: https://github.com/zephyrproject-rtos/hal_espressif/pull/171. It is pending merge in Zephyr's now.

closed 2023-01-12 by sylvioalves

rillbert, 2023-01-12:
Brilliant!

@sylvioalves do you have any insight why I see the, to me, strange behavior of the assert macros that I try to explain in the issue?

sylvioalves, 2023-01-12:
I can't say exactly. Can you update and use latest main with `west update` and verify if this happens again in both scenarios? If it still happens, are you able to share some code sample?

rillbert, 2023-01-17:
> I can't say exactly. Can you update and use latest main with `west update` and verify if this happens again in both scenarios? If it still happens, are you able to share some code sample?

@sylvioalves ok, I have used zephyr sha ef203fd8aeb7175 to build the same project with four different configurations. The project is a slightly modified `blinky` sample and its complete `prj.conf` file looks as:

```
CONFIG_BOOTLOADER_MCUBOOT=y
CONFIG_ASSERT=X
CONFIG_NEWLIB_LIBC=X
```
where 'X' is either 'n' or 'y' depending on the build configurations below. Note that the project itself is not dependent on newlib to build correctly. I would guess that the same behaviour would be seen ni all sample applications.

To me, case 3 below looks just wrong and case 4 is a bit confusing since a different assert function is being used. 
Do I miss any configurations or otherwise fail to enable something for this to work as expected?

Here are the generated code snippets for the code around line 185 in file `clk.c´ for the four cases:

### 1 (CONFIG_NEWLIB_C=n, CONFIG_ASSERT=n)
The `assert` seems to be a no-op as expected.
```
    bool res = rtc_clk_cpu_freq_mhz_to_config(new_freq_mhz, &new_config);
400d0ac2:	004d25        	call8	400d0f94 <rtc_clk_cpu_freq_mhz_to_config>
400d0ac5:	0a3d      	mov.n	a3, a10
    assert(res);

    // Wait for UART TX to finish, otherwise some UART output will be lost
    // when switching APB frequency
#ifdef CONFIG_ESP_CONSOLE_UART
    esp_rom_uart_tx_wait_idle(CONFIG_ESP_CONSOLE_UART_NUM);
400d0ac7:	0a0c      	movi.n	a10, 0
400d0ac9:	fd6b81        	l32r	a8, 400d0078 <_image_irom_vaddr+0x78> (40009278 <esp_rom_uart_tx_wait_idle>)
400d0acc:	0008e0        	callx8	a8
#endif
```

### 2 (CONFIG_NEWLIB_C=n, CONFIG_ASSERT=y)
The ´assert´ calls a function ´assert_print´ which seems ok.
```
    bool res = rtc_clk_cpu_freq_mhz_to_config(new_freq_mhz, &new_config);
400d0d6b:	004ea5        	call8	400d1254 <rtc_clk_cpu_freq_mhz_to_config>
    assert(res);
400d0d6e:	6adc      	bnez.n	a10, 400d0d88 <esp_clk_init+0x5c>
400d0d70:	fcdbb1        	l32r	a11, 400d00dc <_image_irom_vaddr+0xdc> (3f400d71 <esp32_gpio_ports_addrs+0x401>)
400d0d73:	fcd7c1        	l32r	a12, 400d00d0 <_image_irom_vaddr+0xd0> (3f400cf6 <esp32_gpio_ports_addrs+0x386>)
400d0d76:	fcd8a1        	l32r	a10, 400d00d8 <_image_irom_vaddr+0xd8> (3f4009e2 <esp32_gpio_ports_addrs+0x72>)
400d0d79:	b9a0d2        	movi	a13, 185
400d0d7c:	ffc4e5        	call8	400d09cc <assert_print>
400d0d7f:	b9a0b2        	movi	a11, 185
400d0d82:	fcd3a1        	l32r	a10, 400d00d0 <_image_irom_vaddr+0xd0> (3f400cf6 <esp32_gpio_ports_addrs+0x386>)
400d0d85:	ffc3a5        	call8	400d09c0 <assert_post_action>

    // Wait for UART TX to finish, otherwise some UART output will be lost
    // when switching APB frequency
```

### 3 (CONFIG_NEWLIB_C=y, CONFIG_ASSERT=n)
The ´assert´ macro now **calls a function ´__assert_func´ despite asserts being set to disabled**.
```
    bool res = rtc_clk_cpu_freq_mhz_to_config(new_freq_mhz, &new_config);
400d0b2c:	004ea5        	call8	400d1018 <rtc_clk_cpu_freq_mhz_to_config>
    assert(res);
400d0b2f:	dacc      	bnez.n	a10, 400d0b40 <esp_clk_init+0x4c>
400d0b31:	fd54d1        	l32r	a13, 400d0084 <_image_irom_vaddr+0x84> (3f400d79 <blanks$1+0x111>)
400d0b34:	fd52c1        	l32r	a12, 400d007c <_image_irom_vaddr+0x7c> (3f400d7d <__func__$0>)
400d0b37:	b9a0b2        	movi	a11, 185
400d0b3a:	fd51a1        	l32r	a10, 400d0080 <_image_irom_vaddr+0x80> (3f400d2c <blanks$1+0xc4>)
400d0b3d:	b243a5        	call8	40082f78 <__assert_func>

    // Wait for UART TX to finish, otherwise some UART output will be lost
    // when switching APB frequency
```

### 4 (CONFIG_NEWLIB_C=y, CONFIG_ASSERT=y)
The ´assert´ macro calls the same function ´__assert_func´  as when asserts where disabled.
```
    bool res = rtc_clk_cpu_freq_mhz_to_config(new_freq_mhz, &new_config);
400d0da0:	004ea5        	call8	400d128c <rtc_clk_cpu_freq_mhz_to_config>
    assert(res);
400d0da3:	dacc      	bnez.n	a10, 400d0db4 <esp_clk_init+0x4c>
400d0da5:	fccdd1        	l32r	a13, 400d00dc <_image_irom_vaddr+0xdc> (3f40107f <blanks$1+0x417>)
400d0da8:	fccbc1        	l32r	a12, 400d00d4 <_image_irom_vaddr+0xd4> (3f401083 <__func__$0>)
400d0dab:	b9a0b2        	movi	a11, 185
400d0dae:	fccaa1        	l32r	a10, 400d00d8 <_image_irom_vaddr+0xd8> (3f401032 <blanks$1+0x3ca>)
400d0db1:	b33925        	call8	40084144 <__assert_func>

    // Wait for UART TX to finish, otherwise some UART output will be lost
    // when switching APB frequency
```