zephyrproject-rtos/zephyr #53775: LE Audio: Remove use do k_work in PACS notify

Thalley, 2023-01-13:
**Is your enhancement proposal related to a problem? Please describe.**
In pacs.c all notifications are done from the system workqueue due to the use of k_work. This has an issue with not knowing whether the request to send the notification is accepted by GATT, which could/should perhaps prevent the PACS server from changing the value.

**Describe the solution you'd like**
Remove the use of `k_work` in `pacs.c` and just do the calls to `bt_gatt_notify` directly. 

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

szymon-czapracki, 2023-03-07:
[WIP] https://github.com/zephyrproject-rtos/zephyr/pull/55572

Thalley, 2023-03-07:
> [WIP] #55572

You can just add the `fixes` keyword in the draft PR :) That gives it a nice link that is easier to follow (added it)

closed 2023-07-04 by fabiobaltieri