zephyrproject-rtos/zephyr #53780: sysbuild with custom board compilation failed to find the board

VinyFerrero1, 2023-01-13:
**Describe the bug**
During a build with the **_sysbuild_** option the option to use a specific path for a custom board is failed.

Please also mention any information which could help others to understand
the problem you're facing:
 - target nrf52840dk_nrf52840
 - project hello_world from zephyr/samples
 - using the option --sysbuild with the option -DBOARD_ROOT then 
 - Dhello_world_BOARD_ROOT, -Dhello_world_BOARD_ROOT_CONFIG, -Dhello_world_CONFIG_BOARD_ROOT

**To Reproduce**
Steps to reproduce the behavior:
1.  west build -c -p auto -b plank --sysbuild zephyr/samples/hello_world/ -- -DSB_CONFIG_BOOTLOADER_MCUBOOT=y -DBOARD_ROOT=/../path_of_the_custom_boards
2. CMake Error at /../zephyrproject/zephyr/cmake/modules/boards.cmake:163 (message):
  Invalid BOARD; see above.

**Expected behavior**
The build is successful for the custom board

**Impact**
It is not possible to use the sysbuild and so all the domain mechanism if the boards is not describes into the folder zephyr/

**Logs and console output**
 west build -c -p auto -b plank -d build/hello_world --sysbuild zephyr/samples/hello_world/ -- -DSB_CONFIG_BOOTLOADER_MCUBOOT=y -DBOARD_ROOT=/home/viny/Workspace/01-Repository/zephyr-3.2.0/zephyrproject/test/
-- west build: generating a build system
Loading Zephyr module(s) (Zephyr base): extensions sysbuild_extensions python west root zephyr_module boards shields sysbuild_kconfig
-- Found Python3: /usr/bin/python3.10 (found suitable exact version "3.10.6") found components: Interpreter 
-- Cache files will be written to: /home/viny/.cache/zephyr
-- Found west (found suitable version "0.14.0", minimum required is "0.7.1")
-- Board: plank
Parsing /home/viny/Workspace/01-Repository/zephyr-3.2.0/zephyrproject/zephyr/share/sysbuild/Kconfig
Loaded configuration '/home/viny/Workspace/01-Repository/zephyr-3.2.0/zephyrproject/build/hello_world/empty.conf'
Merged configuration '/home/viny/Workspace/01-Repository/zephyr-3.2.0/zephyrproject/build/hello_world/empty.conf'
Merged configuration '/home/viny/Workspace/01-Repository/zephyr-3.2.0/zephyrproject/build/hello_world/zephyr/misc/generated/extra_kconfig_options.conf'
Merged configuration '/home/viny/Workspace/01-Repository/zephyr-3.2.0/zephyrproject/build/hello_world/empty.conf'
Configuration saved to '/home/viny/Workspace/01-Repository/zephyr-3.2.0/zephyrproject/build/hello_world/zephyr/.config'
Kconfig header saved to '/home/viny/Workspace/01-Repository/zephyr-3.2.0/zephyrproject/build/hello_world/autoconf.h'
-- 
   *********************************
   * Running CMake for hello_world *
   *********************************

Loading Zephyr default modules (Zephyr base).
-- Application: /home/viny/Workspace/01-Repository/zephyr-3.2.0/zephyrproject/zephyr/samples/hello_world
-- CMake version: 3.22.1
-- Found Python3: /usr/bin/python3.10 (found suitable exact version "3.10.6") found components: Interpreter 
-- Cache files will be written to: /home/viny/.cache/zephyr
-- Zephyr version: 3.2.99 (/home/viny/Workspace/01-Repository/zephyr-3.2.0/zephyrproject/zephyr)
-- Found west (found suitable version "0.14.0", minimum required is "0.7.1")
-- Board: plank
No board named 'plank' found.

Please choose one of the following boards:

arc:
  em_starterkit
  ...
CMake Error at /home/viny/Workspace/01-Repository/zephyr-3.2.0/zephyrproject/zephyr/cmake/modules/boards.cmake:163 (message):
  Invalid BOARD; see above.
Call Stack (most recent call first):
  /home/viny/Workspace/01-Repository/zephyr-3.2.0/zephyrproject/zephyr/cmake/modules/zephyr_default.cmake:113 (include)
  /home/viny/Workspace/01-Repository/zephyr-3.2.0/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /home/viny/Workspace/01-Repository/zephyr-3.2.0/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:5 (find_package)

-- Configuring incomplete, errors occurred!
CMake Error at cmake/modules/sysbuild_extensions.cmake:243 (message):
  CMake configure failed for Zephyr project: hello_world

  Location:
  /home/viny/Workspace/01-Repository/zephyr-3.2.0/zephyrproject/zephyr/samples/hello_world
Call Stack (most recent call first):
  CMakeLists.txt:59 (ExternalZephyrProject_Add)

-- Configuring incomplete, errors occurred!
See also "/home/viny/Workspace/01-Repository/zephyr-3.2.0/zephyrproject/build/hello_world/CMakeFiles/CMakeOutput.log".
FATAL ERROR: command exited with status 1: /usr/bin/cmake -DWEST_PYTHON=/usr/bin/python3 -B/home/viny/Workspace/01-Repository/zephyr-3.2.0/zephyrproject/build/hello_world -GNinja -DBOARD=plank -DSB_CONFIG_BOOTLOADER_MCUBOOT=y -DBOARD_ROOT=/home/viny/Workspace/01-Repository/zephyr-3.2.0/zephyrproject/test/ -S/home/viny/Workspace/01-Repository/zephyr-3.2.0/zephyrproject/zephyr/share/sysbuild -DAPP_DIR:PATH=/home/viny/Workspace/01-Repository/zephyr-3.2.0/zephyrproject/zephyr/samples/hello_world

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain : zephyr 3.2.99, gcc-arm-none-eabi-10-2020-q4-major-x86_64-linux.tar.bz2, zephyr-sdk-0.15.2_linux-x86_64.tar.gz

**Additional context**

tejlmand, 2023-01-18:
Please note that this issue is not specific to `BOARD_ROOT` but exists for all `x_ROOT`s in Zephyr, ARCH, SOC, BOARD, etc.

So a solution to this issue should preferable address all handling of `x_ROOT`.

FYI @nordicjm

closed 2023-05-03 by carlescufi

liweifeng81, 2023-10-09:
it's still not working on v3.4.0

nordicjm, 2023-10-10:
> it's still not working on v3.4.0

It is working in 3.4, just tested and works. Note that you must use an absolute path for the BOARD_ROOT, not a relative path

liweifeng81, 2023-10-11:
Thanks  for your fast response.
And yes, it works with absolute path.
Is it possible to make it work with the relative path?

liweifeng81, 2023-10-11:
> > it's still not working on v3.4.0
> 
> It is working in 3.4, just tested and works. Note that you must use an absolute path for the BOARD_ROOT, not a relative path

Thanks for your fast response.
And yes, it works with absolute path.
Is it possible to make it work with the relative path?

nordicjm, 2023-10-11:
No, relative path will not be supported because what's the path relative to? To the sysbuild folder? To the main application folder? To the current image folder? It's indeterminate, therefore absolute paths are required

liweifeng81, 2023-10-11:
> No, relative path will not be supported because what's the path relative to? To the sysbuild folder? To the main application folder? To the current image folder? It's indeterminate, therefore absolute paths are required

IMHO, it should relative to the path that run the command.

nordicjm, 2023-10-11:
Well the path that ran the command is not used, not even without sysbuild, it uses the application that is being configured as the base path for relative paths. But really that's the point, every person would give a different answer, so the easy solution is to now allow relative paths and require absolute paths, then no-one gets confused.

PR #56801 (2023-05-03): sysbuild: Fix issue with *_ROOT values not propagating
Fixes an issue where variables like BOARD_ROOT would be provided to sysbuild but would then be lost on target images.

Fixes #53780