zephyrproject-rtos/zephyr #53786: BLE:DF: slot_plus_us is not set properly

saleh-unikie, 2023-01-13:
**Describe the bug**
I was reviewing the source code and I noticed to update the sync slot, an improper number is used.
As you can find here:
https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/bluetooth/controller/ll_sw/ull_df.c#L494
```
...
slot_plus_us = BT_HCI_LE_CTE_LEN_MAX;
...
err = ull_sync_slot_update(sync, slot_plus_us, slot_minus_us);
...
```
And it's expected that ``slot_plus_us``be in microseconds unit, according to this line which converts us to ticks:
https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/bluetooth/controller/ll_sw/ull_sync.c#L1212
```
int ull_sync_slot_update(...)
{
	...
	ret = ticker_update(...,
			    HAL_TICKER_US_TO_TICKS(slot_plus_us),
			    HAL_TICKER_US_TO_TICKS(slot_minus_us),
			    ...);
```

while the variable is not in microseconds:
https://github.com/zephyrproject-rtos/zephyr/blob/main/include/zephyr/bluetooth/hci.h#L1602
```
/* Min and max Constant Tone Extension length in 8us units */
#define BT_HCI_LE_CTE_LEN_MIN                  0x2
#define BT_HCI_LE_CTE_LEN_MAX                  0x14
```

It should be multiplied to 8 to be converted to the microseconds unit.

**To Reproduce**
Check the source code

**Expected behavior**
N/A

**Impact**
Unknown

**Logs and console output**
N/A

**Environment (please complete the following information):**
N/A

carlescufi, 2023-01-13:
Could you send a Pull Request?

saleh-unikie, 2023-01-13:
Yes, of cource

PR #53837 (2023-01-17): Bluetooth: Controller: Fix slot_plus_us by converting max CTE len to us
Fixes #53786
commit a4890f95 (2023-01-17): Bluetooth: Controller: Fix slot_plus_us by converting max CTE len to us

Fixes #53786
The variable slot_plus_us is assumed to be a number in us unit.
To assign the maximum possible CTE length to this variable,
BT_HCI_LE_CTE_LEN_MAX has been used while this parameter is defined
in a 8us units (it's 0x14=20 that corresponds to 160us).
To convert this number to us, it's needed to be multiplied by 8.