zephyrproject-rtos/zephyr #53790: Flash Init fails when CONFIG_SPI_NOR_IDLE_IN_DPD=y

JoeEtacompute, 2023-01-13:
**Describe the bug**
spi_nor_configure function fails when CONFIG_SPI_NOR_IDLE_IN_DPD and CONF_SPI_NOR_SFDP_MINIMAL are enabled. 

acquire_device must be called before calling spi_nor_process_sfdp. After the call to spi_nor_process_sfdp call release_device.

We are using a nrf52832. 
The nrf connect version is v2.1.3.
The zephyr version is:
    VERSION_MAJOR = 3
    VERSION_MINOR = 1
    PATCHLEVEL = 99
   VERSION_TWEAK = 0

Here is the modified code in the function:  static int spi_nor_configure(const struct device *dev)
The modified code snipit starts at line 1093 in spi_nor.c

#else /* CONFIG_SPI_NOR_SFDP_MINIMAL */
	/* For devicetree and runtime we need to process BFP data and
	 * set up or validate page layout.
	 */
	acquire_device(dev);
	rc = spi_nor_process_sfdp(dev);
	release_device(dev);

	if (rc != 0) {
		LOG_ERR("SFDP read failed: %d", rc);
		return -ENODEV;
	}

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 - What have you tried to diagnose or workaround this issue?
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
 - ...

**To Reproduce**
Build for eta compute nrf52832 platform. 

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
This was a show stopper for us.

**Logs and console output**

**Environment (please complete the following information):**
 - OS:  Windows
 - Toolchain nrf connect v2.1.3

**Additional context**
Using a mx25u6432f external flash, with spi communication.

nordicjm, 2023-01-17:
This mentions nRF connect SDK, not zephyr, can you reproduce this in current `main` of zephyr, if it does not reproduce please close this issue and raise this on https://devzone.nordicsemi.com/ instead

JoeEtacompute, 2023-01-17:
In addition to the nrf connect version, I also documented the zephyr
version:

We are using a nrf52832.
The nrf connect version is v2.1.3.
The zephyr version is:
VERSION_MAJOR = 3
VERSION_MINOR = 1
PATCHLEVEL = 99

I checked the zephyr main branch and the issue exists.

Regards,
Joe

-- 
This email (including its attachments) is intended only for the person or 
entity to which it is addressed and may contain information that is 
privileged, confidential or otherwise protected from disclosure. 
Unauthorized use, dissemination, distribution or copying of this email or 
the information herein or taking any action in reliance on the contents of 
this email or the information herein, by anyone other than the intended 
recipient, or an employee or agent responsible for delivering the message 
to the intended recipient, is strictly prohibited. If you are not the 
intended recipient, please do not read, copy, use or disclose any part of 
this e-mail to others. Please notify the sender immediately and permanently 
delete this e-mail and any attachments if you received it in error. 
Internet communications cannot be guaranteed to be timely, secure, 
error-free or virus-free. The sender does not accept liability for any 
errors or omissions.

carlescufi, 2023-01-17:
@JoeEtacompute can you reproduce this in vanilla Zephyr? nRF Connect SDK is a different project and questions about this should go to https://devzone.nordicsemi.com/

carlescufi, 2023-03-06:
Closing due to lack of response from reporter.

closed 2023-03-06 by carlescufi