zephyrproject-rtos/zephyr #53795: Provide way to seamlessly integrate Kconfig from modules into Zephyr

SkryptX, 2023-01-13:
Currently I haven't found any way to seamlessly integrate a module's Kconfig into the standard Zephyr menuconfig tree. For example I have a custom display driver in a module, but instead of having all the settings for it under `Device Drivers -> Display Drivers` like in-tree drivers, they are split between `Device Drivers -> Display Drivers` for general stuff like the log level and more specific stuff in `Modules -> [[Module name]] -> Device Drivers -> Display Drivers`.

**The solution?**
I'm not dead set on a solution, but I think some kind of hooking system could already solve most of the problems. Since all the specific device drivers are just appended to the Kconfig of the respective driver (i.e. `zephyr/drivers/display/Kconfig`). There would have to be various hooks all across the Kconfig files and a module Kconfig could register itself to one of those and be inserted at this place. 

I have no idea how hard something like this would be or maybe there are easier ways to achieve this, but I want to start the discussion, since it would be a next step in making modules more like first-class citizens.

**Additional context**
Integrating into the CMake library for a specific driver is already possible with` zephyr_library_amend()`, but Kconfig there's nothing similar for Kconfig.

henrikbrixandersen, 2023-01-17:
Having the module Kconfigs end up in a different tree in e.g. menuconfig has its benefits, IMO. Otherwise it would be very difficult to spot from where a given Kconfig originates.

SkryptX, 2023-01-17:
@henrikbrixandersen I do see your point, but you can always just expand the menuconfig with the details. In my use of Zephyr I found it more annoying to have to switch back and forth between these two disjoint locations since especially drivers do depend on the base variables a lot. You enable something and have to the module, change there, go back to verify and so on. It just feels clunky and if you are not the person writing the driver/module/whatever yourself you might just miss that an option in the module got unlocked by you toggling something in Zephyr Kconfig tree.

closed 2024-11-19 by nordicjm