zephyrproject-rtos/zephyr #53799: Info request: SMP hash definition

pmagnuson, 2023-01-14:
**Describe the bug**
The context of this is the SMP protocol for Application/software image management group.

I have downloaded a file to a nRF52840DK board.  The file is our own build and includes the Zephyr OS with SMP enabled.  
The SHA256 of this file is 

```
shasum -a 256 merged.hex 
c63c0f507d6a7d886b9c3c591e01bdd16bc4f37bf429cf0585cceda2bf4c6e40  merged.hex
```
A request is made to the `Get state of images request` (https://docs.zephyrproject.org/latest/services/device_mgmt/smp_groups/smp_group_1.html#mcumgr-smp-group-1)

The reported hash comes back as:

```json
{"images":[{"slot":0,"version":"0.0.0","hash":"M0n_yjkSlchlyur5Eesz8-BkMTh2mjWtJj5wlBs9CyY","bootable":true,"pending":false,"confirmed":true,"active":true,"permanent":false}],"splitStatus":0}}
```

Question: 
- is there another encoding that should be applied to the SHA256 before it is transmitted via SMP?
- is there a different way the SHA256 should be calculated?  

Thank you,

nordicjm, 2023-01-16:
Ah interesting, seems the wrong field was updated... The description for full SHA256 hash should actually be in the upload command, not the image state command, will submit a PR to fix that.

The 2 hashes are different - when uploading, you use a full sha256 of the image, this is so it can be resumed if one side disconnects and also so that the entire image can be verified. When you get the state of images though, you are no longer dealing with sha256 hashes of the full image, you are dealing with sha256 hashes though but it is the sha256 has that is part of the MCUboot headers in the TLV section, essentially, it is the sha256 hash of the image's data which needs signature verification only, and is the hash that is signed with the private key. So to get that hash, you need to search for the TLV in the image and get the right identifier then extract it.

Details are on https://docs.mcuboot.com/design.html - IMAGE_TLV_SHA256 is the key you want

de-nordic, 2023-01-16:
You can verify the sha, that the image list should return, with `imgtool verify`, for example
$$ imgtool verify ../hello/zephyr/zephyr.signed.bin
Image was correctly validated
Image version: 0.0.0+0
Image digest: 4966f2d97dcb7ccda76aa0787a7df38a52ec96496a9f9e555f21749fe551aafc

closed 2023-01-17 by carlescufi

pmagnuson, 2023-01-18:
I really appreciate all the responses to this request.  It's helping a lot.  
I have a file, `app_update_z_ot1.bin`, that I loaded onto the nRF dev board.  
`imgtool` reports:

```
imgtool verify app_update_z_ot1.bin
Image was correctly validated
Image version: 0.0.0+0
Image digest: ebe393b41171036e223fc2686bd03ab7211fc2e6d0f86415750f9dadf06c7283
```

When the code does an SMP request for State of Images, the response includes:
```
"hash":"6-OTtBFxA24iP8Joa9A6tyEfwubQ-GQVdQ-drfBscoM"
```

I'm not sure why the reported hash is different than the `imgtool` report.  Is there another transformation performed on the digest hash?

PR #53817 (2023-01-17): doc: mgmt: img_mgmt: Fix description of hash and sha
Fixes an issue with the documentation whereby the wrong hash/sha sections were described, also adds a link to upstream MCUboot documentation for further information.

Fixes #53799