zephyrproject-rtos/zephyr #53801: k_busy_wait adds 1us delay unnecessarily

girishji, 2023-01-14:
**Describe the bug**
 
k_busy_wait(x) is supposed to wait ~x us. But specifying parameter x=.1 gives 1us delay, x=1us gives 2us delay, x=2 us gives 3us delay, etc. I have attached oscilloscope screen for x=0.1us where it results in 1us delay. k_busy_wait() cannot be relied on for values under 10us.

I am using nrf52840, specifically Seeed XIAO BLE board. 

**To Reproduce**

west build -b xiao_ble

I am switching a GPIO from high to low continuously with k_busy_wait() between each switch. See the screen capture of oscilloscope probe attached to GPIO pin.

**Expected behavior**

k_busy_wait should not add 1us extra delay to value specified in the parameter.

**Impact**

annoyance. I have to implement my own routine for this.

**Logs and console output**

See oscilloscope screen capture

**Environment (please complete the following information):**
MacOS, Zephyr SDK directly (not nrf connect) zephyr-sdk-0.15.2_macos-x86_64

**Additional context**
Nothing special. I used "getting started" guide to install  Zephyr and used a short program to setup GPIO.

[image]

girishji, 2023-01-15:
I can understand 0.1us rounded up to 1us. But 2us becoming 3us is a bug.

andyross, 2023-01-20:
Am I not understanding that scope output?  It's reporting the period of the square wave is 2.172us, so the time between software-directed transitions is indeed "1us plus some slop due to loop overhead".

girishji, 2023-01-22:
I specified 0.1us in that scope output, and it delays by >1us. The scope output I captured is not a good example. If you specify 2us for instance, it will give you more like 3.5us delay. It adds >1us to every number you specify.

andyross, 2023-01-22:
How exactly did you specify 0.1?  k_busy_wait() takes an integer argument...  Some exerciser code might help.

I think what you're seeing here is just round up in the nrf handler.  The way k_sleep() et. al. work is that they're defined as sleeping for "at least" the argument period.  They don't wake up early (by default, though you can interrupt the call via other APIs!).  So for timer-based calls, you always get your sleep time rounded up to the nearest OS tick.

The k_busy_wait() docs are less strict, but it makes sense that they'd have similar behavior.  On Nordic, k_busy_wait() is not implemented with the timer driver, but with an instruction counted loop at CPU frequency.  There, the round up could plausibly have been done in the argument space of microseconds instead of in units of loop iterations.  That would be a bugglet, I guess.

nashif, 2023-05-30:
@girishji  @64mhz can you please reply to the above comment

closed 2023-08-13 by github-actions[bot]