zephyrproject-rtos/zephyr #53810: TaskList for Supporting RaspberryPi Pico peripherals

soburi, 2023-01-16:
A task list for hardware support for the RaspberryPi Pico. (Inspired by #38657.)
This list is categorized by title of the table of contents of Pico-SDK.

## peripherals

- [x] hardware_adc
  #45363
  #53591 
- [ ] hardware_base ❓May not need in zephyr?
- [ ] hardware_claim
- [ ] hardware_clocks
- [ ] hardware_divider
    #52896
- [ ] hardware_dma
- [ ] hardware_exception ❓May not need in zephyr?
- [x] hardware_flash
  #47806
  #53431
- [x] hardware_gpio
  #34835 
  #42495
- [x] hardware_i2c
  #42513
  #44046
- [ ] hardware_interp
- [ ] hardware_pio
  #44316 
- [ ] hardware_pll
- [x] hardware_pwm
  #47578

  #50876
  #50877
  #51546
  #52348
  #52692
- [ ] hardware_resets
  #43073
  #46057
  #48969
- [ ] hardware_rtc
- [ ] hardware_spi
  #45131
- [ ] hardware_sync
- [ ] hardware_timer
- [x] hardware_uart
 #34835
   #42421
  #49883
- [ ] hardware_vreg
- [x] hardware_watchdog
  #43393
- [ ] hardware_xosc

## peripherals (Not listed in ToC of Pico-SDK)

- [x] hwinfo
  #42558
- [x] usb
  #42506
  #49772
  #52145
- [ ] clock control driver
  #52901 

## Runtime Infrastructure	

- [ ] boot_stage2
- [ ] pico_base	❓May not need in zephyr?
- [ ] pico_binary_info	
- [ ] pico_bit_ops	
- [ ] pico_bootrom	
- [ ] pico_bootsel_via_double_reset ❓May not need in zephyr?
- [ ] pico_cxx_options ❓May not need in zephyr?
- [ ] pico_divider
- [ ] pico_double	
  #52896
- [ ] pico_float	
  #52896
- [ ] pico_int64_ops	
  #52896
- [ ] pico_malloc	❓May not need in zephyr?
- [ ] pico_mem_ops	
- [ ] pico_platform ❓May not need in zephyr?
- [ ] pico_printf ❓May not need in zephyr?
- [ ] pico_runtime ❓May not need in zephyr?
- [ ] pico_stdio ❓May not need in zephyr?
- [ ] pico_standard_link ❓May not need in zephyr?

## Boards

- [x] rpi_pico
 #34835
 #45137
 #49563
 #49944
 #50772
- [x] adafruit_kb2040
 #51098
- [x] sparkfun_pro_micro_rp2040
 #53600

soburi, 2023-01-16:
@yonsch @andrei-edward-popa 
I made a list to check the support status.
Please comment if there is something that should be fixed.

andrei-edward-popa, 2023-01-16:
@soburi Here is also the counter driver for Pico which uses RTC. https://github.com/zephyrproject-rtos/zephyr/pull/52312

petejohanson, 2023-01-16:
And this PR I opened last night: #53811 for reboot support into bootloader.

ajf58, 2023-01-19:
I've been working on adding support for `hardware_timer` as I'm generally interested in adding a bit more support for the RP2040 and the RPi Pico and Pico W to Zephyr. Is there a way to throw my hat into the ring on this?

DaAwesomeP, 2023-01-19:
@soburi issue for MPU support: https://github.com/zephyrproject-rtos/zephyr/issues/44318

soburi, 2023-01-19:
Hi, @ajf58 

> Is there a way to throw my hat into the ring on this?

Thank you for your interest.

The best way is creating PR as draft and add the the number of it to this list.
I think this list can edit by anyone, please tell me if not so.
I'll update it.

soburi, 2023-01-21:
@DaAwesomeP 

I created a PR to fix the issue.
https://github.com/zephyrproject-rtos/zephyr/pull/53973

yonsch, 2023-01-26:
@soburi Thank you for this list. I've edited this list to add a few items, but mainly to remove entries that seemed redundant. Here are the ones removed with the reasons for the removal:

- hardware_base - Necessary headers for the HAL, does not require any work on our side
- hardware_claim - A util library for resource management, not required
- hardware_exception - Zephyr already handles it for us, not required
- hardware_pll - Already covered by clock control
- hardware_sync - A util library for resource management, used by the hal but does not require any work on out side
- hardware_xosc - Already covered by clock control
- bootstage2 - Already in use
- pico_base - Could not find it in pico-sdk
- pico_binary_info - Can be handled by Zephyr
- pico_divider/double/float/int64_ops - Already covered by divider

DaAwesomeP, 2023-01-30:
@yonsch The pico version seems to [make use of the ROSC](https://github.com/raspberrypi/pico-sdk/pull/1111/files#diff-01bd4769cf3328e6be7f955a13f7c7c794c65bff480d71d1dd8a6e0b068e2980R30-R35) as an entropy source as well as the [Bus Performance Counter](https://github.com/raspberrypi/pico-sdk/pull/1111/files#diff-01bd4769cf3328e6be7f955a13f7c7c794c65bff480d71d1dd8a6e0b068e2980R37-R38). I'm not certain if the latter is pico-specific or an M0+ feature.

I agree that the Zephyr implementation is probably best, but I do wonder if there is a way to input additional sources of entropy to the Zephyr methods. At a quick glance, it doesn't seem like the Zephyr version has a way to make use of that as a source of entropy (this was a very surface level check). I can look into this more later if there is interest.

soburi, 2023-02-01:
> @soburi Thank you for this list. I've edited this list to add a few items, but mainly to remove entries that seemed redundant. Here are the ones removed with the reasons for the removal:

Thank you for updating this list.
It was helpful because I was not confident in my judgment.

We got many functions are now available.
It seems PIO is a difficult piece. I wish to discuss and make progress about it.

yonsch, 2023-06-01:
> Are there any plans to support bluetooth functionality on the pi pico wireless?
> Thanks

@rp1231 Once the interface to the cyw43 (SPI over PIO) is supported, support for both wifi and Bluetooth will be possible.

ThreeEights, 2023-07-28:
@soburi - Can you add this pull request for SPI via PIO to the list?  [#60395](https://github.com/zephyrproject-rtos/zephyr/pull/60395)  I'm not sure whether it should be listed under SPI or PIO.

PatrickE94, 2023-07-29:
Hey everyone,

I'm also interested in both Zephyr and Pico W support. I jumped over a few steps in creating proper support and wrote a small wifi driver using the cyw43-driver library from pico sdk (via the hal module). I say jumped over since I mostly ported the SPI over PIO from pico sdk.

This is all working, at least in some sense. Running the `samples/net/wifi` sample successfully with DHCP and communication over both TCP and UDP. I'm currently fiddling with the interrupt for the wake-pin.

I would be happy to contribute to this in any way for proper support. I'm completely new to Zephyr, my main experience is not mainly embedded but I have a few years of professional embedded on FreeRTOS and many years C programming.

Can anyone point me in a good direction forward? Should I push my early driver as a draft PR, or begin switching to @ThreeEights SPI over PIO implementation?

Thanks

ThreeEights, 2023-07-31:
Hi, @PatrickE94 - Of course, I'd be most pleased if you work with my SPI via PIO implementation.  :-)  You should know that it currently doesn't support the 3-wire SPI configuration needed to access the WiFi chip on the Pico W.  I'm trying to debug that today.  We also have a project in our team to complete a Zephyr driver for the CYW43439 chip, though that's still some weeks away from being ready.

Meanwhile ... welcome to Zephyr!
Steve

PatrickE94, 2023-07-31:
> Hi, @PatrickE94 - Of course, I'd be most pleased if you work with my SPI via PIO implementation. :-) 
> You should know that it currently doesn't support the 3-wire SPI configuration needed to access the WiFi chip on the Pico W. I'm trying to debug that today. 

Neat, then perhaps I will start at the other end :-)

> We also have a project in our team to complete a Zephyr driver for the CYW43439 chip, though that's still some weeks away from being ready.

I found that out a few hours ago. Since that's the Infineon guys doing a more generic implementation (with more features, AMDPU etc) than the one in pico sdk, I agree that's the right way! Quickly checking their wifi-host-driver concept, it doesn't look to intimidating to implement a SPI HAL layer which should enable their driver on the pico.
They've done a good job documenting it.

> Meanwhile ... welcome to Zephyr! Steve

Many thanks!
Patrick

JohnConnett, 2023-10-27:
Hi @ThreeEights, @PatrickE94,

I'm also interested in WiFi support on Pico W. What's the current state of play?

ThreeEights, 2023-10-27:
@JohnConnett - I have a PIO SPI driver that works on everything except the Pico W, for some reason I'm trying to debug, and a pull request is pending for the CYW43xxx WiFi driver.  We're trying to pull it all together.

PatrickE94, 2023-10-27:
@JohnConnett I have just put together some glue for the cyw driver and the PIO SPI. It looks like it should work, given that @ThreeEights gets three wire SPI to work. But I haven't tested anything since I have no communication...

I have an old branch where I've mashed in the official pico sdk driver which works, but it's not very neat.

I've been pretty busy with life recently, but I might just take a slow evening to see if I can support @ThreeEights. Let me know if there's anything you'd like me to take a peek at. I have access to Salae logic analysers and some other fine equipment at work.

JohnConnett, 2023-10-29:
> I've been pretty busy with life recently, but I might just take a slow evening to see if I can support @ThreeEights. Let me know if there's anything you'd like me to take a peek at. I have access to Salae logic analysers and some other fine equipment at work.

I currently have a single **Pico W** (no headers), with a **Pico WH** and **Pi Debug Probe** on order.

The plan is to use a **Pico W** with a [PicoADC16](https://www.8086.net/product/picoadc16) and [Photo Transistor Light Sensors](https://thepihut.com/products/photo-transistor-light-sensor) to monitor the state of LEDs on an underfloor heating controller and report them using **MQTT** to [OpenEnergyMonitor](https://openenergymonitor.org/). The LEDs indicate if the electrothermal actuators and circulation pump are energized.

We will probably use [MicroPython](https://www.raspberrypi.com/documentation/microcontrollers/micropython.html) initially. I was interested in **Zephyr** as a possible alternative.

Thanks for the comments. I might be interested in exploring **Zephyr** on **Pico W** for other things when WiFi is available.

DaAwesomeP, 2024-07-01:
Something to potentially add to the list: PIO-based Ethernet MAC
Example: https://github.com/holysnippet/pico_eth

DaAwesomeP, 2024-07-01:
The "binary info" feature was I think completed through here: #54464 (can be checked off).

vulpes2, 2024-12-29:
Would be great if @soburi can update this list to reflect the recently merged SPI support for the AIROC WiFi driver and the RP2350 support.

soburi, 2025-01-07:
@vulpes2 
Done. Thanks for the notification.

flyfish30, 2025-02-20:
The cyw43439 wifi module on the pico 2 w still doesn't works fine. It is failed to set country code to cyw43439. When country code is WHD_COUNTRY_UNITED_STATES, the set IOVAR command failed with timeout. When country code is WHD_COUNTRY_UNITED_STATES_REV4, the set IOVAR command failed with return status -2. I also try set country code with WHD_COUNTRY_WORLD_WIDE_XX and WHD_COUNTRY_WORLD_WIDE_XX_REV17, the set IOVAR command also failed with timeout or return status -2.
I had checked the buffer of set IOVAR command, it is same as buffer of the set IOVAR command in pico-sdk.
Please fix this issue.

soburi, 2025-03-08:
@flyfish30 

> The cyw43439 wifi module on the pico 2 w still doesn't works fine. It is failed to set country code to cyw43439. When country code is WHD_COUNTRY_UNITED_STATES, the set IOVAR command failed with timeout. When country code is WHD_COUNTRY_UNITED_STATES_REV4, the set IOVAR command failed with return status -2. I also try set country code with WHD_COUNTRY_WORLD_WIDE_XX and WHD_COUNTRY_WORLD_WIDE_XX_REV17, the set IOVAR command also failed with timeout or return status -2. I had checked the buffer of set IOVAR command, it is same as buffer of the set IOVAR command in pico-sdk. Please fix this issue.[@soburi](https://github.com/soburi)

At this time, Zephyr has not provided any support for Pico2W.
It is best to create an issue to request a new feature.
If you have already worked a port of pico2w,
you can also create a PR and discuss it through review.

flyfish30, 2025-03-17:
Hi, @soburi 

I have already worked a port of pico 2 w by myself. I will create a PR to zephyr project. And by use third-party wifi driver of cyw43 from beechwoods-software, I get the wifi support in pico 2 w, it work well.
I saw a statement from beechwoods-software saying that the Infineon official WHD Wi-Fi driver integrated in the new code of Zephyr 4.0 is already available. Why can't it work properly on the Pico 2 W board?

soburi, 2025-03-18:
@flyfish30 

Simply put, just because it works on Pico1W doesn't mean it will work on Pico2W. Unless someone investigates, we won't know why.
Someone has to do that work.

keks, 2025-03-19:
Just asking because neither the docs nor this issue explicitly mention this: There is Wifi support for the Pico W (Version 1) but not BLE, right? What would be needed to make that work?

rt-2pm2, 2025-03-22:
> Just asking because neither the docs nor this issue explicitly mention this: There is Wifi support for the Pico W (Version 1) but not BLE, right? What would be needed to make that work?
> 
> Edit: Looks like in theory there are BLE drivers for that chip: https://docs.zephyrproject.org/latest/build/dts/api/bindings/bluetooth/infineon%2Ccyw43xxx-bt-hci.html#dtbinding-infineon-cyw43xxx-bt-hci9

At least from the zephyr pico documentation [page](https://docs.zephyrproject.org/latest/boards/raspberrypi/rpi_pico/doc/index.html), the bluetooth is not mentioned. @keks You are right, there is a driver for the cyw43xx-bt, but from that page it looks like it is on UART, whilst that chip on the RPi seems to be on SPI.
 
I am also interested to know more about enabling the bluetooth on the pico.

soburi, 2025-03-23:
@keks @rt-2pm2 

I haven't heard of anyone working on it yet.

It might be a lot of work, as it will require some changes to the bus, just like @ThreeEights did.
Of course, collaborators are welcome.

rt-2pm2, 2025-03-23:
@soburi 

Thanks very much for your reply. I don't consider myself an expert, so I guess I will have fun reading datasheet and documentation, but, from a first glance, I am concerned about the connections. For this reason, please, correct me if I get something wrong, I don't want to mislead anyone here ;-).

The RPi Pico [datasheet ](https://datasheets.raspberrypi.com/picow/pico-w-datasheet.pdf), page 4, reports about it:
> The wireless interface is connected via SPI to the RP2040

Looking at the CW43439 datasheet v3.0 (I don't understand why the link to the v5.0 on the Infineon site is broken...) it seems that that device has different buses for the two subsystems. The WLAN communicates via SPI and the Bluetooth is via UART. 

If the RPi Pico has only the SPI connection with the CW43439, how to manage the Bluetooth? Would this problem be addressed with the "some changes to the bus" you were referring to? 
I will look at the commits made by @ThreeEights to figure it out.

Best

vulpes2, 2025-03-23:
There has been some confusion about Bluetooth on the Pico W so I'll attempt to provide some details on the hardware and explain what's required to make Bluetooth support (the controller is dual mode) happen.

On this board the only available HCI transport is SPI, and the same SPI bus is already used by the AIROC WiFi driver. UART on the cypress chip is not connected to the micro and you cannot use it. This HCI over SPI transport is completely undocumented in the datasheet, and from what I can tell the RPi Pico W (and the Pico 2 W) are the only boards that use it. The packet format is pretty standard as far as I can tell.

To add Bluetooth support someone needs to come up with a way to multiplex the SPI bus and make it usable for both the WiFi and HCI driver simultaneously. There is currently no mechanism in Zephyr to facilitate that so a new shim driver is likely required. pico-sdk has such a [driver](https://github.com/raspberrypi/pico-sdk/tree/2.1.1/src/rp2_common/pico_cyw43_driver) and parts of it is within pico-sdk (BSD-3-Clause license), the other part (cyw43-driver) is proprietary source-available software so porting portions of it is not possible.

Amritha-124, 2025-05-16:
how to add bluetooth support in pico w ,is there any way to multiplex the spi bus

dsseng, 2025-07-23:
Could we maybe track some of these bits? All go to the generic System category, as they can be adapted to RP2040 by a person with hardware to test on.

- RAM/flashless boot, useful on its own and needed for AMP: https://github.com/zephyrproject-rtos/zephyr/pull/91200
- Inter-processor communication, part of the AMP work: https://github.com/zephyrproject-rtos/zephyr/pull/92923
- CPU1 spinup, part of the AMP work: https://github.com/zephyrproject-rtos/zephyr/pull/93379

r-c-n, 2025-08-15:
Hi, I worked on adding support for the inter-processor mailbox for eventual AMP support: https://github.com/zephyrproject-rtos/zephyr/pull/94502

This targets the same peripheral as https://github.com/zephyrproject-rtos/zephyr/pull/92923 but it provides a [mailbox interface](https://docs.zephyrproject.org/apidoc/latest/group__mbox__interface.html) that can be used as a backend for [zephyr,mbox-ipm](https://docs.zephyrproject.org/latest/hardware/peripherals/ipm.html) for inter-processor communication. It was tested on RP2350 but it's meant to work for the RP2040 as well.

Could we consider adding it to the tasklist as an alternative approach? Thanks

r-c-n, 2025-08-21:
Hi @soburi , are you aware of any ongoing effort to add USB Host Controller support for the RP2350/2040? If not, I may give it a try.

soburi, 2025-08-21:
> Hi [@soburi](https://github.com/soburi) , are you aware of any ongoing effort to add USB Host Controller support for the RP2350/2040? If not, I may give it a try.

@jfischer-no is the leading USB expert, so ask. It's probably hard work, but worth it.

@jfischer-no
Could we hear your opinion?