zephyrproject-rtos/zephyr #53821: Support for dynamic, portable configuration of GPIO interrupt priority through GPIO API

besmarsh, 2023-01-16:
**Is your enhancement proposal related to a problem? Please describe.**

Zephyr provides the GPIO API to allow applications to interface with GPIOs in a portable way. This includes configuration of GPIO interrupts. Features supported for GPIO interrupts include configuring the trigger, adding callbacks, and removing callbacks. One feature that is missing is configuring the priority of a GPIO-triggered interrupt.

The Zephyr IRQ API does support configuration of interrupt priorities when configuring an interrupt, either through the compile-time `IRQ_CONNECT` or the runtime `irq_connect_dynamic`. However this cannot be used to configure the priority of a GPIO interrupt as both methods require the IRQ number, which is not exposed through the GPIO API (handled internally to drivers).

Some configuration of GPIO interrupts is possible. For example, for STM32, the Kconfig options `EXTI_STM32_EXTI*_IRQ_PRI` can be used to configure EXTI interrupt priorities. However, this is not portable, dynamic, or consistent across architectues/SoCs.

**Describe the solution you'd like**

I would like to see `gpio_pin_interrupt_configure` extended to take an additional parameter; the priority for the GPIO interrupt being configured. Much like `irq_connect_dynamic` this could be ignored by platforms that do not support this feature. For those that do, the underlying driver could be extended to configure the interrupt priority. As this is already implemented for many architectures - it's done by `irq_connect_dynamic` - this existing functionality could be utilised.

**Describe alternatives you've considered**

An alternative method to support this feature could be to expose the IRQ number through the GPIO API so that `irq_connect_dynamic` could be used to configure a GPIO-triggered interrupt's priority. However, this is messy as `irq_connect_dynamic` also takes a callback to configure for the interrupt, which is already supported through the GPIO API. It's also stated that `irq_connect_dynamic` does not support reconfiguring interrupts.

**Additional context**

See conversation at https://discord.com/channels/720317445772017664/733037791876874280/1063094172518715533

henrikbrixandersen, 2023-01-17:
Typically, GPIO controllers provide just one IRQ line, representing all available GPIO lines. For this kind of GPIO controller, the interrupt priority can (statically) be configures via devicetree.

The STM32 EXTI is special, as it allows for configuring the priority of individual GPIO pin interrupts. One option could be to extend the STM32 GPIO controller driver with support for custom flags for settings the EXTI line IRQ priority.

/cc @erwango

erwango, 2023-01-17:
> One option could be to extend the STM32 GPIO controller driver with support for custom flags for settings the EXTI line IRQ priority.

This is already available today (with Kconfig, moving to dts in https://github.com/zephyrproject-rtos/zephyr/pull/53825).
The tricky part of the question here is to configure priority **dynamically**.

henrikbrixandersen, 2023-01-17:
The upper 8 bits or `gpio_flags_t` are reserved for GPIO controller driver specific flags. Could the EXTI interrupt priority be encoded there?

besmarsh, 2023-01-17:
I've started working on a PR for this using the approach described in my description of this issue. Do you see an issue with that approach @henrikbrixandersen?

henrikbrixandersen, 2023-01-17:
> I've started working on a PR for this using the approach described in my description of this issue. Do you see an issue with that approach @henrikbrixandersen?

I'd like to see a list of more than one vendor supplying a GPIO controller with support for IRQ priority on individual lines before extending the generic GPIO API to support this.

besmarsh, 2023-01-17:
> I'd like to see a list of more than one vendor supplying a GPIO controller with support for IRQ priority on individual lines before extending the generic GPIO API to support this.

Fair enough, it would be better to have multiple vendors supporting this to have it in the generic API. However wouldn't it be useful to add the interface and have it supported by STM and ready for any other vendors if support comes along? It would also allow dynamic configuration of GPIO interrupt priority right now for those vendors where just one IRQ line is used, just in a more limited way than on STM, but still more flexible than static-only configuration.

henrikbrixandersen, 2023-01-17:
> Fair enough, it would be better to have multiple vendors supporting this to have it in the generic API. However wouldn't it be useful to add the interface and have it supported by STM and ready for [...]

We already support vendor-specific GPIO API extensions through `gpio_flags_t`, hence my question:

> The upper 8 bits of `gpio_flags_t` are reserved for GPIO controller driver specific flags. Could the EXTI interrupt priority be encoded there?

besmarsh, 2023-01-18:
> We already support vendor-specific GPIO API extensions through `gpio_flags_t`, hence my question:

Maybe I'm missing something, but I see `gpio_dt_flags_t` used to hold GPIO devicetree flags, with "The lower 8 bits are used for standard flags. The upper 8 bits are reserved for SoC specific flags." in the description. However, `gpio_flags_t`, used to hold GPIO configuration flags and configure GPIOs at runtime doesn't state this format, and appears to use the top 8 bits.

besmarsh, 2023-01-18:
cc @cfriedt

henrikbrixandersen, 2023-01-18:
> Maybe I'm missing something, but I see `gpio_dt_flags_t` used to hold GPIO devicetree flags, with "_The lower 8 bits are used for standard flags. The upper 8 bits are reserved for SoC specific [...]

Right, but `gpio_dt_flags_t` is a subset of `gpio_flags_t`. _If_ the custom GPIO line IRQ priority can be encoded in the vendor-specific bits, this would allow for configuring the priority using devicetree _and_ allow for dynamic configuration via regular, existing GPIO API calls.

besmarsh, 2023-01-18:
> Right, but `gpio_dt_flags_t` is a subset of `gpio_flags_t`. _If_ the custom GPIO line IRQ priority can be encoded in the vendor-specific bits, this would allow for configuring the priority using [...]

Right okay, I didn't realise `gpio_dt_flags_t` was a subset of `gpio_flags_t`. I think this approach is possible and would work but to me it seems less straightforward to use. The application would have to rely on the format of `gpio_flags_t` and encode the priority rather than simply passing it in its plain form. I think it's also a little confusing to encode priority in a _flags_ parameter, as it's not really a flag.

besmarsh, 2023-01-18:
I've tested my method on an STM32 and it seems to work. Modifying the `gpio_pin_interrupt_configure` API would require many changes throughout the Zephyr tree though.

besmarsh, 2023-01-18:
@erwango @cfriedt Any thoughts on the ideas proposed here by myself and @henrikbrixandersen?

cfriedt, 2023-01-18:
@besmarsh - I'm kind of inclined to agree with Brix on this one.

Assuming STM does not already use the vendor specific bits, that would probably be the best place. It should be possible to create STM specific macros to encode the priority in the flags field for both compile-time const and runtime code.

How many bits are required to encode STM GPIO IRQ priority?

besmarsh, 2023-01-18:
I don't think NVIC priorities are ever more than 8 bits, so assuming STM doesn't already use the vendor-specific bits then it should fit. That solution is definitely less disruptive to the API than mine, too.

To me though, using `gpio_dt_flags_t` ("DT", "flags") to encode a multi-bit integer set at runtime seems wrong. It doesn't seem logical and to an application developer I imagine it would be rather counter-intuitive.

besmarsh, 2023-01-19:
After moving on to looking at UART, I realised I was going to have the same problem there (no UART API for configuring priority dynamically). Here though, the IRQ line can be retrieved from devicetree fairly easily. I'm realizing now just how much the STM32 GPIO/EXTI is special/unusual, and it may be that for every other peripheral and on every other device (I don't know if there are any other special cases) the IRQ line can be easily retrieved from the devicetree.

I'm thinking now that this whole approach of implementing dynamic priority configuration within the GPIO API is wrong, as this is generic functionality that could be useful for any peripheral and that doesn't need to be implemented in every driver. Dynamic interrupt priority configuration is already done by `irq_connect_dynamic`, so the only change needed would be a new API function in `zephyr/include/irq.h`; `irq_priority_set(irqn, prio)` or `irq_set_priority(irqn, prio)`, which redirects to the already-implemented architecture-specific `irq_priority_set` functions. No changes needed to any drivers.

Of course this would also provide dynamic priority re-configuration for any interrupts as well, which would be a useful enhancement in general.

After @erwango's #53825, hopefully the same method could be used for GPIO on STM32s, but going via the exti node which maps GPIO EXTI lines to IRQ lines.

Thoughts? @henrikbrixandersen @cfriedt @erwango

henrikbrixandersen, 2023-01-19:
Could you please elaborate a bit on the use-case for changing peripheral IRQ priorities like this at runtime?

besmarsh, 2023-01-20:
> Could you please elaborate a bit on the use-case for changing peripheral IRQ priorities like this at runtime?

In very general terms, consider any application that has multiple operating modes/states, and between these different modes/states the priority of peripherals changes. For example, an application that uses GPIO and UART, where in one operating mode the UART communication is higher priority than the GPIO, but in another mode the GPIO is higher priority than the UART.

As this solution would now apply to _any_ interrupt priorities (not just peripherals) it adds extra flexibility there too and is no longer just a peripheral-centric change. Interrupt priorities are already configured at runtime anyway (even in `IRQ_CONNECT`  priority is set at runtime) so I feel this enhancement would be a small change that provides further flexibility.

go2sh, 2024-06-12:
Hey, I just came around here during my porting of TC3x  devices. TC3x devices also support multiple IRQ lines for GPIO IRQs and it has at least two way to generate gpio interrupts. I'm currently having separate configuration for the IRQs in a different devicetree module.