zephyrproject-rtos/zephyr #53823: Bluetooth init failed on nrf5340_audio_dk_nrf5340_cpuapp

justin-nxp, 2023-01-16:
**Describe the bug**
Bluetooth init failed on nrf5340_audio_dk_nrf5340_cpuapp 
samples\bluetooth\iso_receive
samples\bluetooth\iso_broadcast

Please also mention any information which could help others to understand
the problem you're facing:
 - nrf5340_audio_dk_nrf5340_cpuapp 
 - What have you tried to diagnose or workaround this issue?
 - not

**To Reproduce**
Steps to reproduce the behavior:
1.in zephyr repo “git checkout main”
2.build target
 C:\Users\Desktop\zephyrproject\zephyr> west build -b nrf5340_audio_dk_nrf5340_cpuapp .\samples\bluetooth\iso_broadcast --pristine
3. flash board
west flash --recover
4.connect board to serial 
[image]
5. See error
 Bluetooth init failed (err -11)

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: windows
 - commit 3d6f3a52cf075d8d9f3a94fe2c30156423fc1066
 - SDK 0.15.0

cvinayak, 2023-02-06:
You need the hci_rpmsg sample be built and flashed to the network core without which there is no Bluetooth LE Controller hence you will get the said failure.

carlescufi, 2023-03-06:
Please follow the instructions provided by @cvinayak

closed 2023-03-06 by carlescufi