zephyrproject-rtos/zephyr #53834: stm32 sdmmc doesn't work correctly with power management enabled

coran21, 2023-01-16:
**Describe the bug**
I'm new to Zephyr RTOS so it's possible that I'm doing something incorrectly or forgot for some configuration, I'm sorry in that case.

In disk access driver for STM32 `sdmmc_stm32.c` is function `static int stm32_sdmmc_clock_enable(struct stm32_sdmmc_priv *priv)`, which sets and initialize PLLSAI1 clock for SDMMC.

After sdmmc clocks are configured, there is `HAL_Delay(2);` in the stm32 sdmmc hal driver, which calls `k_msleep(Delay);` and trigger whole suspend - resume. The problem is that the sdmmc is using `PLLSAI1` clock which is not re-enabled during resume.

```
/**
  * @brief  Set SDMMC Power state to ON.
  * @param  SDMMCx Pointer to SDMMC register base
  * @retval HAL status
  */
HAL_StatusTypeDef SDMMC_PowerState_ON(SDMMC_TypeDef *SDMMCx)
{
  /* Set power state to ON */
#if defined(STM32L4P5xx) || defined(STM32L4Q5xx) || defined(STM32L4R5xx) || defined(STM32L4R7xx) || defined(STM32L4R9xx) || defined(STM32L4S5xx) || defined(STM32L4S7xx) || defined(STM32L4S9xx)
  SDMMCx->POWER |= SDMMC_POWER_PWRCTRL;
#else
  SDMMCx->POWER = SDMMC_POWER_PWRCTRL;
#endif /* STM32L4P5xx || STM32L4Q5xx || STM32L4R5xx || STM32L4R7xx || STM32L4R9xx || STM32L4S5xx || STM32L4S7xx || STM32L4S9xx */

  /* 1ms: required power up waiting time before starting the SD initialization
  sequence */
  HAL_Delay(2);

  return HAL_OK;
}
```

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 STM32L4R5

**To Reproduce**
Steps to reproduce the behavior:
```
CONFIG_PM=y
CONFIG_DISK_ACCESS=y
CONFIG_DISK_DRIVER_SDMMC=y
CONFIG_SDMMC_STM32=y
CONFIG_FILE_SYSTEM=y
CONFIG_FAT_FILESYSTEM_ELM=y
```
```
&sdmmc1 {
	pinctrl-0 = <&sdmmc1_d0_pc8 &sdmmc1_d1_pc9
		     &sdmmc1_d2_pc10 &sdmmc1_d3_pc11
		     &sdmmc1_ck_pc12 &sdmmc1_cmd_pd2>;
	pinctrl-names = "default";
	bus-width = <4>;
	status = "okay";
};

```
**Expected behavior**
`PLLSAI1` clock is re-enabled after resume

**Impact**
SDMMC inicialization doesn't work

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
Linux
 - Toolchain (e.g Zephyr SDK, ...)
Zephyr SDK
 - Commit SHA or Version used
3.2

coran21, 2023-01-17:
@erwango 
Thanks for pointing out important things, it was very helpful. I implemented PD_DEVICE mode support into SDMMC driver, but that doesn't help with preventing the OS to go into suspend (STOP2) state at all as I need to have the SDMMC clock running during read/write and from what I understand non of the A] B] C] you wrote about would prevent OS to go into suspend (STOP2) state which disable the clocks or I'm I wrong?

erwango, 2023-01-18:
A] and B]  don't relate to PM indeed.

Though, in C], use of `pm_device_busy_set(dev); ` should prevent entering stop mode (if `CONFIG_PM_DEVICE=y`).
Otherwise `pm_policy_state_lock_get(PM_STATE_SUSPEND_TO_IDLE, PM_ALL_SUBSTATES);` should prevent the whole system entering `PM_STATE_SUSPEND_TO_IDLE`.
PM_ALL_SUBSTATES: all stop modes. 
Use '3' instead to only prevent STOP MODE 2 in L4 case. (Abstracting for all STM32 series might require some thinking)

coran21, 2023-01-18:
> Though, in C], use of pm_device_busy_set(dev);  should prevent entering stop mode (if CONFIG_PM_DEVICE=y).
I would say that this isn't true. I though that too so I implement the whole PM_DEVICE mode support into the driver, but when I tested it, the system went to suspend mode even when the device was busy.
This is the only code in suspend function related to devices:
`#if defined(CONFIG_PM_DEVICE) && !defined(CONFIG_PM_DEVICE_RUNTIME_EXCLUSIVE)
	if ((z_cpus_pm_state[id].state != PM_STATE_RUNTIME_IDLE) &&
			(atomic_sub(&z_cpus_active, 1) == 1)) {
		if (pm_suspend_devices()) {
			pm_resume_devices();
			z_cpus_pm_state[id].state = PM_STATE_ACTIVE;
			(void)atomic_add(&z_cpus_active, 1);
			SYS_PORT_TRACING_FUNC_EXIT(pm, system_suspend, ticks,
						   z_cpus_pm_state[id].state);
			return false;
		}
	}
#endif`

Unfortunately  `if (pm_suspend_devices())` is non zero only if the device suspend failed. It skips busy device suspend phase, but it continue to suspend the whole system.

Is it ok to use `pm_policy_state_lock_get(PM_STATE_SUSPEND_TO_IDLE, PM_ALL_SUBSTATES);` in the device driver?

erwango, 2023-01-18:
Ok, right, PM busy is not really what we want.
So yes `pm_policy_state_lock_get(PM_STATE_SUSPEND_TO_IDLE, PM_ALL_SUBSTATES);` could be used in device drivers.

erwango, 2023-01-20:
@coran21 A] is ow impemented: https://github.com/zephyrproject-rtos/zephyr/pull/53898. So now, MSI is used as domain clock which should solve part of the issue.

coran21, 2023-01-23:
@erwango
I tested #https://github.com/zephyrproject-rtos/zephyr/pull/53898 and it actually breaks SDMMC at least on my stm32l4r5 board. The problem is that HSI48 is set as a clock for SDMMC, but 48MHz it's too much (at least without DMA) when the uC runs at 80MHz and I'm getting FIFO overrun error. The previous PLLSA1 clock was set to 16MHz and that worked fine. Did somebody actually tested that new stm32_sdmmc.c driver? It's not possible to build it because of this:

```
	.pclken = {
		.bus = DT_INST_CLOCKS_CELL(0, bus),
		.enr = DT_INST_CLOCKS_CELL(0, bits),
	},
```

`pclken` was changed in that same commit to the pointer and this won't work anymore.

erwango, 2023-01-23:
@coran21 This PR breaks a lot of things indeed. Likely I had a really bad last minute rebase issue. This should be fixed by https://github.com/zephyrproject-rtos/zephyr/pull/54011

coran21, 2023-01-23:
#54011 fixes most of the problems, but from what I saw it won't fix stm32l4r5 configuration. Also, for MSI clock is necessary to set `CONFIG_SDMMC_STM32_CLOCK_CHECK=n`  otherwise it returns error.

I'm sorry for using this for a question, but I'm straggling with adding DMA into device tree for SDMMC, could you point me to some documentation where can I find how to do it? 
So far I found only this from Zephyr documentation: (https://docs.zephyrproject.org/latest/build/dts/api/bindings/mmc/st,stm32-sdmmc.html) 

```
Optional DMA channel specifier. If DMA should be used, specifier should
hold a phandle reference to the dma controller, the channel number,
the slot number, channel configuration and finally features.

Only priority bits are used in the configuration. There are no feature
flags.

For example dmas for TX/RX on SDMMC
   dmas = <&dma2 4 6 0x30000 0x00>, <&dma2 4 3 0x30000 0x00>;
```
But I don't know from where this numbers came (except the dma number and channel number) and what kind of configuration is it.

I tried to just copy and paste it into my device tree that I can debug the code little bit, but I got these error:

`devicetree error: dma-names property in /soc/sdmmc@50062400 in /home/peta/prace/zephyrproject/zephyr/misc/empty_file.c has 2 strings, expected 4 strings
`

Here is my sdmmc in dts file:
```
&sdmmc1 {
	clocks = <&rcc STM32_CLOCK_BUS_AHB2 0x400000>,
		 	<&rcc STM32_SRC_MSI CLK48_SEL(3)>;
	pinctrl-0 = <&sdmmc1_d0_pc8 &sdmmc1_d1_pc9
		     &sdmmc1_d2_pc10 &sdmmc1_d3_pc11
		     &sdmmc1_ck_pc12 &sdmmc1_cmd_pd2>;
	pinctrl-names = "default";
 	dmas = 	<&dma2 4 6 0x30000 0x00>,
 			<&dma2 4 3 0x30000 0x00>;
 	dma-names = "tx", "rx";
 	bus-width = <4>;
	status = "okay";
};
```

erwango, 2023-01-23:
DMA bindings are described here: dts/bindings/dma/st,stm32-dma-v1.yaml.
Also check: https://github.com/zephyrproject-rtos/zephyr/pull/53486

erwango, 2023-01-23:
About: 
> https://github.com/zephyrproject-rtos/zephyr/pull/54011 fixes most of the problems, but from what I saw it won't fix stm32l4r5 configuration. 

I'm not able to test sdmmc on L4R5 target, but USB works fine of L4S5 target (USB and SDMMC share same clock domain, and L4R5 is same as L4S5 from clock dts perspective). Can you elaborate?

> Also, for MSI clock is necessary to set CONFIG_SDMMC_STM32_CLOCK_CHECK=n otherwise it returns error.

Ok, there was indeed a bug around this. I'll fix it.

coran21, 2023-01-23:
Thanks for that link and pointing me to the DMA binding description.

>I'm not able to test sdmmc on L4R5 target, but USB works fine of L4S5 target (USB and SDMMC share same clock domain, and L4R5 is same as L4S5 from clock dts perspective). Can you elaborate?

sdmmc for L4R5 now use HSI48 clock, but L4R5 is not fast enough to handle such a fast read from sdmmc and there will be rx fifo overrun - happened to me with HSI48 clock and works fine with MSI at 16MHz

erwango, 2023-01-23:
Ok, thanks for the insights. Note that there is a hardware flow control option for SDMMC, which you can enable with `CONFIG_SDMMC_STM32_HWFC=y`. This doesn't prevent using DMA, but can help in some cases.

coran21, 2023-01-25:
Ok, it works fine with hardware flow enabled. When I looked more into adding DMA support into my board I found that STM32L4plus series has internal DMA which works without any configuration and it's handled by SDMMC HAL driver. I created a pull request to add this functionality into Zephyr #54091 , would be great if you can look at it.

erwango, 2023-01-25:
> Ok, it works fine with hardware flow enabled. 

Great. We should think about enabling it by default. That's not the first time it's shown to be useful.

> When I looked more into adding DMA support into my board I found that STM32L4plus series has internal DMA which works without any configuration and it's handled by SDMMC HAL driver. I created a pull request to add this functionality into Zephyr #54091 , would be great if you can look at it.

Ok, will do

coran21, 2023-01-25:
one off topic question:

Next thing I need to get to work is RTC. I found that there is PR for new RTC API #52618 and also some old PR for STM32 RTC driver #8405. Could you recommend the direction to go? In current counter implementation the RTC is zeroed during reset and is missing "set time" feature which doesn't work for me. I would need to modify the counter driver which is probably not the way I want to go or add my own RTC driver probably based on PR #8405 and modify it to work with new RTC API #52618 (would this be eventually merged?).
I would appreciate any comments of suggestions.

coran21, 2023-01-25:
regarding the PM for SDMMC driver, I implemented PD_DEVICE feature, but I'm not sure if it worth for PR. The problem is that PM for SDMMC is much more complex - in case that SDMMC is suspended and SD card itself is powered down with power pin, just resuming the peripheral won't work as we need to initialize the SD card itself after power up (this is currently done by disk_access init) before we want to use it. One option could  be to call this SD card init function if the card was initialized in the past with disk_access init. 
I will look into this more when I get to work basic peripheral I need to use and start working on power management.

erwango, 2023-01-26:
> Next thing I need to get to work is RTC. 

I suggest to ping people on https://github.com/zephyrproject-rtos/zephyr/pull/52618 to ask where this is going or if a new PR is goign to make it progress.

erwango, 2023-01-26:
> regarding the PM for SDMMC driver, 

I suggest to open a dedicated issue on this topic and discuss this with subsystem maintainer. Don't tag it STM32 as it's not implementation specific.

coran21, 2023-01-26:
> I suggest to open a dedicated issue on this topic and discuss this with subsystem maintainer. Don't tag it STM32 as it's not implementation specific.

I opened new issue on this topic #54125, maybe you can look at it if it makes sense to you. Also I made a mistake and opened it as a bug and realized later that it's a bug because I implemented the power management myself, maybe you can fix the label.

erwango, 2023-06-20:
@coran21 Since there were a lot of topics addressed and solved in this issue, I propose to close it.
#54125 will allow to track the remaining issue.
If there are remaining points to be addressed, I suggest to open a new issue (and tag it as a follow up of current issue if required).

closed 2023-06-20 by erwango