zephyrproject-rtos/zephyr #53851: Add Pigweed Moudle

yperess, 2023-01-17:
## Origin

Pigweed
https://pigweed.dev/
https://pigweed.googlesource.com/pigweed/pigweed

## Purpose

Pigweed is a collection of embedded specific utilities that aim at making embedded application development easier/faster.

## Mode of integration

I would like this to be integrated as a module. Pigweed is highly complimentary to Zephyr and provides many great utilities for our developers. It should help reduce the overhead of new projects as well as offload some feature (mainly subsystem features) by leveraging what other teams are building. I would suggest putting this in github.com/zephyrproject-rtos/pigweed

## Maintainership

Maintainer:
- [TBD by Pigweed Team]

Collaborators:
- @yperess 
- @rbbrns

## Pull Request

https://github.com/zephyrproject-rtos/zephyr/pull/53760

## Description

Pigweed provides a slew of modules that would make application development in Zephyr faster and more modern. While some modules do have quite a bit of overlap with Zephyr, they're necessary for Pigweed in order to avoid having the RTOS implementation be done all in one module. For example, `pw_rpc.synchronous_client_api` depends on `pw_chrono.system_clock`. For Zephyr, there isn't a direct benefit coming from `pw_chrono.system_clock`, but there is a very real benefit coming from `pw_rpc.*` libraries. The full list of the modules can be found at https://pigweed.dev/module_guides.html

Pigweed will primarily solve the problem of trying to add generic (non Kernel level) features to Zephyr. These libraries include HDLC, `pw_snapshot`, `pw_tokenizer`, `pw_rpc` and a few others that can be found in the module guide page.

Pigweed is the right module so solve these issues because it gets us many implementations of what would have later been subsystems in Zephyr for nearly free. It also leverages a whole dedicated team to support these subsystems/modules.

## Dependencies

Pigweed will not be adding any new dependencies for Zephyr in order to support it.

Zephyr will not have any direct dependencies on Pigweed. Adding Pigweed as a module will help increase discoverability for Pigweed and will allow Zephyr to offload some components that would otherwise require quite a bit of engineering.

## Revision

Pigweed is currently used in several Google products with Zephyr and has been tested with `main:HEAD` so it should suffice to take the latest commit from `main`.

## License

Apache 2.0

yperess, 2023-01-17:
I added the TSC label to make sure we get some eyes on this tomorrow.

yperess, 2023-01-25:
The Pigweed team is still working on slides to present. There are some delays due to recent turmoil at Google but we'll post them here once ready.

yperess, 2023-08-08:
Architecture review:

Need feedback about duplicate logic. Pigweed provides a C++ interface or implement to Zephyr constructs which means some duplication. Does anyone object?

Examples:
* pw_assert
* pw_base64
* pw_byes
* pw_checksum

chadnorvell, 2023-08-09:
Hey @galak, yes, it appears that particular page is a bit out of date, since just about everything marked as "Planned" for Zephyr actually has modules now, as you can see [here](https://pigweed.dev/docs/os/zephyr/). We'll get that fixed.

henrikbrixandersen, 2023-08-15:
If Pigweed is not supposed to be used in-tree I see no reason for it to be included in the default west manifest. We really need to move towards not including _everything_ in the upstream west manifest, but become better at advertising external modules that work with Zephyr. Our current approach does not scale.

fabiobaltieri, 2023-08-15:
I think that the module management situation should be treated separately, it's not like one module less or more would make much of a difference at this point.

henrikbrixandersen, 2023-08-15:
> I think that the module management situation should be treated separately, it's not like one module less or more would make much of a difference at this point.

I disagree. We need to start changing the situation at some point. This might as well be that point (instead of accepting it first only to remove it again later).

fabiobaltieri, 2023-08-15:
> I disagree. We need to start changing the situation at some point. This might as well be that point (instead of accepting it first only to remove it again later).

Do you think that the fix is to drop modules? If that's the case sure, but I'd imagine more likely we'd come up with a way of managing what gets downloaded by default.

henrikbrixandersen, 2023-08-15:
> Do you think that the fix is to drop modules? If that's the case sure, but I'd imagine more likely we'd come up with a way of managing what gets downloaded by default.

Yes, I do believe the fix is to drop modules that are not used in-tree/do not require in-tree glue code from the upstream west manifest and become better at advertising these as external projects ("Works with Zephyr").

nashif, 2023-08-15:
Agree we need to do something about this, will put this topic in the Arch WG meeting today (https://github.com/zephyrproject-rtos/zephyr/issues/54276)