zephyrproject-rtos/zephyr #53867: kconfig: Linked code into external SEMC-controlled memory without boot header

MikhailSiomin, 2023-01-17:
It is not clear to me why the following check for CODE_FLEXSPI and CODE_FLEXSPI2 configs was added at NXP_IMX_RT_BOOT_HEADER config (at PR https://github.com/zephyrproject-rtos/zephyr/pull/41579). Since the boot ROM cannot copy and then run code linked into non-FlexSPI controlled memory.
https://github.com/zephyrproject-rtos/zephyr/blob/426c9cbc90fb33cab56d2705e40a8a855df1785c/soc/arm/nxp_imx/rt/Kconfig.soc#L652-L657

It is also unclear to me why there was no imply for CODE_SEMC (in PR https://github.com/zephyrproject-rtos/zephyr/pull/22864), CODE_ITCM and other CODE_LOCATION. https://github.com/zephyrproject-rtos/zephyr/blob/426c9cbc90fb33cab56d2705e40a8a855df1785c/soc/arm/nxp_imx/rt/Kconfig.soc#L710-L711

I think these checks should be removed and imply for CODE_SEMC, CODE_ITCM, CODE_SRAM0, CODE_OCRAM added.
`zephyr/soc/arm/nxp_imx/rt/Kconfig.soc`:
```k
...
menuconfig NXP_IMX_RT_BOOT_HEADER
	bool "Boot header"
	depends on (!BOOTLOADER_MCUBOOT)
	help
	  Enable data structures required by the boot ROM to boot the
	  application from an external flash device.
...
config CODE_SEMC
	bool "Link code into external SEMC-controlled memory"
	imply NXP_IMX_RT_BOOT_HEADER if !BOOTLOADER_MCUBOOT

config CODE_ITCM
	bool "Link code into internal instruction tightly coupled memory (ITCM)"
	imply NXP_IMX_RT_BOOT_HEADER if !BOOTLOADER_MCUBOOT
...
config CODE_SRAM0
	bool "Link code into RAM_L memory (RAM_L)"
	imply NXP_IMX_RT_BOOT_HEADER if !BOOTLOADER_MCUBOOT

config CODE_OCRAM
	bool "Link code into OCRAM memory (OCRAM-M4)"
	imply NXP_IMX_RT_BOOT_HEADER if !BOOTLOADER_MCUBOOT
...
```

MikhailSiomin, 2023-02-08:
@danieldegrasse, I can make a PR if you can confirm that my assumptions above are correct.

danieldegrasse, 2023-02-08:
Hi @MikhailSiomin, Sorry for the late response. Currently, `CONFIG_IMX_RT_BOOT_HEADER` only enables the presence of the boot header. Although the iMX.RT bootloader should support loading the image into RAM using this feature, this use case is not currently tested or supported. If you are interested in enabling this feature, a PR to do so would be welcomed, I just wanted to make you aware that the effort to do so may be more than changing the KConfig settings here.

MikhailSiomin, 2023-02-08:
@danieldegrasse, I used a custom bootloader on 2.7 Zephyr that linked into semc and it worked. But faced the above problem when switching to Zephyr version 3.2. I will check what the changes affect and prepare a PR.

danieldegrasse, 2023-02-08:
@MikhailSiomin, great- thank you for taking this on!

PR #54864 (2023-02-23): soc: arm: nxp_imx: rt: Allow to include boot header.
Allow to include boot header for code linked into
not only FlexSPI controlled memory.
Fixes #53867
commit 8310f837 (2023-02-23): soc: arm: nxp_imx: rt: Allow to include boot header.

Allow to include boot header for code linked into
not only FlexSPI controlled memory.
Fixes #53867
closed 2023-02-23 by carlescufi