zephyrproject-rtos/zephyr #53876: The handle of att indication violates the spec

shiniiiiiiiiing, 2023-01-18:
According to spec 5.3, 3.4.7.2 ATT_HANDLE_VALUE_IND,
_"If the attribute value is longer than (ATT_MTU-3) octets, then only the first
(ATT_MTU - 3) octets of this attributes value can be sent in an indication.
Note: For a client to get a long attribute, it must use the
ATT_READ_BLOB_REQ PDU (see Section 3.4.4.5) following the receipt of
this indication."_
Both the handle of sending att_indication and receving att_indication violate the above statement. the ATT_MTU isn't considered.

stigo-nordic, 2023-01-23:
On the receiving side, l2cap will ensure the received message is within max SDU size, which in the case of ATT messages is the ATT_MTU, see https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/bluetooth/host/l2cap.c#L2512 before calling the channel handler.

On the sending side, ATT indicate will be allocated by  bt_att_chan_create_pdu which will ensure the message does not exceed the ATT_MTU size. See https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/bluetooth/host/att.c#L622 

In other words, I think this should already be handled correctly outside of the core processing of the messages.

stigo-nordic, 2023-01-23:
@shiniiiiiiiiing Do you have any steps to reproduce the issue which are not caught by the regular flow I outlined below, please provide more details for us to reproduce.

closed 2023-02-15 by jori-nordic