zephyrproject-rtos/zephyr #53885: Ethernet TCP Client Issue description with iperf/zperf

Desvauxm-st, 2023-01-18:
Bug description:

The board STM32 used are: Nucleo_F767zi and Nucleo_F429zi 
We used Iperf ( samples\net\zperf)  when Zephyr is in  TCP client and when we try to upload frame we have the following issue into the Zephyr console:
<err> net_pkt: Data buffer (68) allocation failed.  
[image]
  

How to  Reproduce

On the host side  (Windows PC) in cmd windows  type :
iperf -s -p 5001 -B 192.168.0.110 
[image]

On the Zephyr console  type :
zperf tcp upload2 v4 10 1024 100M 
[image]

Expected behavior

No error when we the client TCP test is  performed 

Impact

may be all the board due to the memory allocation failed in TCP client

Logs and console output

I made many test With iperf (samples\net\zperf)
At the beginning the goal was to compare the Ethernet performance in STM32 when  the switch  ETH_STM32_HW_CHECKSUM is Activated
All test results can be seen in the attached excel file
The board used are: Nucleo_F767zi and Nucleo_F429zi
With these test one issue have been found in TCP client Zephir.

See the result file in excel format

[iperf test ETH_STM32_HW_CHECKSUM_Issue.xlsx](https://github.com/zephyrproject-rtos/zephyr/files/10444184/iperf.test.ETH_STM32_HW_CHECKSUM_Issue.xlsx)

[Ethernet TCP Client Zephyr Issue.docx](https://github.com/zephyrproject-rtos/zephyr/files/10444179/Ethernet.TCP.Client.Zephyr.Issue.docx)

rlubos, 2023-01-18:
@Desvauxm-st I don't think the issue you see should be considered a bug, it looks more like a misconfiguration for the specific board you use. My guess here is that TCP might be generating too much traffic for your boards to swallow, hence the allocation errors (given poor TCP performance, TCP retransmissions take place, so the buffers may be hogged at the TCP level).

Try decreasing the TX window value so that TCP stack generates less traffic (by setting `CONFIG_NET_TCP_MAX_SEND_WINDOW_SIZE` in `prj.conf` file). Unless you've modified the default configuration of the sample, the default value for TX window should be `14000` bytes, please try lowering it with the config I've mentioned and see if it helps.

Desvauxm-st, 2023-01-24:
@rlubos
 I change the value of CONFIG_NET_PKT_RX_COUNT, CONFIG_NET_PKT_TX_COUNT, CONFIG_NET_BUF_RX_COUNT
CONFIG_NET_BUF_TX_COUNT and when I increase all the values it works better
Thanks you for your help.

rlubos, 2023-01-24:
@Desvauxm-st can we close the issue then?

Desvauxm-st, 2023-01-24:
@rlubos
Yes

Desvauxm-st, 2023-01-24:
I change the value of CONFIG_NET_PKT_RX_COUNT, CONFIG_NET_PKT_TX_COUNT, CONFIG_NET_BUF_RX_COUNT
CONFIG_NET_BUF_TX_COUNT and when I increase all the values it works better.

closed 2023-01-24 by Desvauxm-st