zephyrproject-rtos/zephyr #53889: Support for LE Audio profiles in Bluetooth tester application and autopts

sjanc, 2023-01-18:
This issue is to track scope and progress of adding support for LE Audio profiles (and their dependencies) in automated qualification test environment (autopts).

This list is (tentative) priority order:
(checked - in progress,  strike-through - completed)

- [ ] BAP
    - [ ] PACS
    - [ ] ASCS
    - [ ] BASS
        - [ ] GAP PADV
        - [ ] GAP PAST
- [ ] VCP
    - [ ] IAS
    - [ ] VCS
    - [ ] AICS
    - [ ] VOCS
- [ ] MCS (GMCS)
- [ ] MCP
- [ ] TBS (GTBS)
- [ ] CCP
- [ ] CSIS
- [ ] CSIP
- [ ] MICS
- [ ] MICP
- [ ] CAS
- [ ] CAP
- [ ] TMAP
- [ ] HAS
- [ ] HAP
- [ ] PBP
- [ ] OTS

sjanc, 2023-01-18:
@mkasenberg FYI

Thalley, 2023-04-18:
@sjanc I'd suggest to move CAP and CAS further up the list, and MCS/MCP down. Perhaps just switch the two?

rasmbach, 2023-04-18:
I agree, let's swap the two...

Thalley, 2023-04-18:
@sjanc I also suggest to swap MICP/S and CCP/TBS, as CCP/TBS might need some more work, whereas MICP/S is closer to production-ready

larsgk, 2023-05-02:
> @sjanc @larsgk Based on the talks from the F2F last week, I think Demant wanted to have a TMAP test application. @MariuszSkamra suggested to implement it as part of the `bttester` where we have a defined interface (BTP) to instruct it.
either that (BTP) or another format that might be easier to interface with for both manual and automatic testing (e.g. JSON over UART could be an option)

> 
> If this is still the case, then I suggest that we move TMAP as far up as we can, but that will of course have dependencies with other profiles.
> 
> @larsgk Would you prefer a TMAP or HAP tester? The former will more like what Pixel is doing now, but the latter is more similar to what a phone _should_ do for hearing aids.

I think we should go for TMAP first, then add HAP (and HAP could be a configurable option).
Maybe @runegrunnet has more input on this?

MSEEHenrik, 2023-05-26:
I'm working on CSIS.

jsplyy, 2023-07-26:
Hi @larsgk , @sp3030 team has enabled TMAP on the bt_tester app, and 
I am working on the TMAP auto-pts enablement.
currently,  
- TMAP profile test framework enablement in auto-pts is ready and TMA server and client test cases pass now. 
- For the TMAP device discovery test cases, they required ext_adv function remote calls, but which have not  been supported in BTP protocols, will continue after this PR merged, https://github.com/zephyrproject-rtos/zephyr/pull/60500.
- For the other test cases, 4.2-4.4 (role and configuration, Audio Stream Transitions, Multi-stream synchronization) do not seem to involve the TMAP APIs, and only rely on the low layer profiles, which may require much efforts.

expect that we have no duplicated works on this for now, 
do you require this urgently?
If yes, I could work with @sp3030 to create some PRs to submit all the current local commits.

[image]
[image]

I will enable

mkasenberg, 2023-09-15:
> I'm working on HAS, CSIS, HAP, CAS & CAP.

@MSEEHenrik , I've almost finished support for BAP test cases. Have you already started CAS/CAP test cases?

MSEEHenrik, 2023-09-15:
Yes - I'm done with CAS & CAP.

Thalley, 2024-01-16:
@sjanc added GMAP to the tasklist

Thalley, 2024-01-30:
TMAP is marked as in progress, but I don't see any PR or effort on this. From https://github.com/zephyrproject-rtos/zephyr/issues/57682 it seems like @sp3030 wanted to do this?

mkasenberg, 2024-01-30:
@agata-ponitka has recently started working on TMAP

jsplyy, 2024-02-06:
Hi @mkasenberg and @agata-ponitka , I also have worked on TMAP, and about 12 test cases have been done on my side, expect to finish in February, what is your status ? could we reallocate some profiles? such as GMAP ?

mkasenberg, 2024-02-06:
@agata-ponitka @jsplyy, which test cases have you completed each of you? Do the tests overlap? Maybe Agata could merge her part and then start the GMAP test cases?

mkasenberg, 2024-02-06:
@sjanc , the CCP seems to be done. Maybe the HRS should be removed from the list as non-audio profile.

piotrnarajowski, 2024-02-06:
@mkasenberg @sjanc CCP, MICS, GMCS, VCP are finished, TBS, GTBS and OTS are finished but not merged yet, and I've just started working on CSIP

agata-ponitka, 2024-02-06:
Hi @jsplyy, I also have 12 tests completed. 
TMAP/SR/SGGIT/SER/BV-01-I  
TMAP/SR/SGGIT/CHA/BV-01-I 
TMAP/CT/TRC/BV-01-I  
TMAP/UMR/TRC/BV-01-I
TMAP/BMS/TRC/BV-01-I 
TMAP/BMR/TRC/BV-01-I
TMAP/BMR/VRC/BV-02-I
TMAP/BMR/ASC/BV-03-I
TMAP/BMR/ASC/BV-04-I
TMAP/BMR/ASC/BV-05-I
TMAP/CL/TMAS/BV-01-C
TMAP/CL/TMAS/BI-01-C

jsplyy, 2024-02-06:
> @agata-ponitka @jsplyy, which test cases have you completed each of you? Do the tests overlap? Maybe Agata could merge her part and then start the GMAP test cases?

5/12 are different, @agata-ponitka , is it ok for you with @mkasenberg 's suggestion ? I could rebase your TMAP change and then go ahead.
FYI @sp3030 

TMAP/SR/SGGIT/SER/BV-01-I
TMAP/SR/SGGIT/CHA/BV-01-I
**TMAP/CL/CGGIT/SER/BV-01-I
TMAP/CL/CGGIT/CHA/BV-01-I**
TMAP/CL/TMAS/BV-01-C
TMAP/CL/TMAS/BI-01-C
**TMAP/CG/TRC/BV-01-I**
TMAP/CT/TRC/BV-01-I 
**TMAP/UMS/TRC/BV-01-I
TMAP/UMR/TRC/BV-01-I**
TMAP/BMS/TRC/BV-01-I
TMAP/BMR/TRC/BV-01-I

agata-ponitka, 2024-02-07:
@jsplyy yes, it's ok for me :) here are my PRs:
https://github.com/zephyrproject-rtos/zephyr/pull/68675
https://github.com/auto-pts/auto-pts/pull/1109

sjanc, 2025-12-11:
with exception of GMAP this is mostly completed, I think we can close this and further work can be tracked with individual tickets