zephyrproject-rtos/zephyr #53894: Move Kconfiglib to Zephyr's GitHub org

carlescufi, 2023-01-18:
## Introduction

As per the conclusions in https://github.com/ulfalizer/Kconfiglib/issues/121, Kconfiglib needs a new home and a new maintainer.

### Problem description

Given that the pypi

### Proposed change

A brief summary of the proposed change - the 10,000 ft view on what it will
change once this change is implemented.

## Detailed RFC

In this section of the document the target audience is the dev team. Upon
reading this section each engineer should have a rather clear picture of what
needs to be done in order to implement the described feature.

### Proposed change (Detailed)

This section is freeform - you should describe your change in as much detail
as possible. Please also ensure to include any context or background info here.
For example, do we have existing components which can be reused or altered.

By reading this section, each team member should be able to know what exactly
you're planning to change and how.

### Dependencies

Highlight how the change may affect the rest of the project (new components,
modifications in other areas), or other teams/projects.

### Concerns and Unresolved Questions

List any concerns, unknowns, and generally unresolved questions etc.

## Alternatives

List any alternatives considered, and the reasons for choosing this option
over them.

carlescufi, 2023-01-18:
@marc-hb @mbolivar-nordic @tejlmand @nordicjm @stephanosio @fschrempf @gmsotavio

nashif, 2023-01-18:
one more... 
python-kconfig

sjg20, 2023-01-18:
Do we need to rename it? Could we take over the project and continue with the same name?

carlescufi, 2023-01-18:
> > Do we need to rename it? Could we take over the project and continue with the same name?
> 
> It looks like that's something we could request: https://peps.python.org/pep-0541/#how-to-request-a-name-transfer

Let's give this a go. I will create a request today and link it.

carlescufi, 2023-01-18:
@jackrosenthal here's the request: https://github.com/pypi/support/issues/2526

galak, 2023-01-18:
pykconfig +1

fabiobaltieri, 2023-01-18:
`Kconfiglib-ng`? Traditional open source optimistic fork name.

marc-hb, 2023-01-18:
> python-kconfig

There are 103 Python packages that debian has named by adding a `python-` prefix to their original name: https://packages.debian.org/bullseye/python/

So if `python-kconfig` is ever packaged by Debian they could call `python-python-kconfig`.

Same thing for Fedora: https://docs.fedoraproject.org/en-US/packaging-guidelines/Python/#_library_naming

Of course Debian and Fedora wouldn't actually do that; I'm merely recommending that Zephyr does not "steal" a very popular naming convention already used by Linux distributions to avoid confusion.

My strong, first preference is to reclaim the existing name if possible, as already requested by @carlescufi. The least disruption by far. I'm optimistic: if this process does not work when the maintainer vanishes then when does it work?

My second preference is `pykconfig`: short, simple and to the point. No hyphen, no mixed case,...

sjg20, 2023-01-18:
It looks like the requests are being actioned, at least some of them. I added my support into the issue

gmsotavio, 2023-01-19:
Hello

My thoughts about the subject

Kconfiglib2 - This name sounds fine to me, but I agree with @jackrosenthal that it can be confusing.
Kconfig - May be confused with Linux Kconfig. So it seems like a bad idea.
python-kconfig - That's ok, but this is the default name for Debian packages. So I agree with @marc-hb that if python-kconfig is packaged by Debian they could call it python-python-kconfig. So, I think this name is not optimistic.
pykconfig - The best of the proposed names.

As pointed out by @fabiobaltieri, we could name this project Kconfiglib-ng.

Summing up, +1 vote to Kconfiglib-ng. My second option is pykconfig.

But, for now, it is best to wait for the request sent by @carlescufi to https://peps.python.org/pep-0541/#how-to-request-a-name-transfer

jackrosenthal, 2023-02-27:
So the name transfer appears to be going nowhere. Anyone opposed to just renaming to `pykconfig`?  Seems like a simple find/replace op.

keith-zephyr, 2023-03-08:
The feedback we got from one of the pypi maintainers is that requests are handled first-in/first-out.  But because everyone is a volunteer, one should expect a slow response time.

marc-hb, 2023-04-02:
From
- https://github.com/pypi/support/issues/2526

> Hi all, I will now start the reachability process for your request which can take up to 6 weeks. Thank you for gathering resources regarding the owner, hopefully we can get a response soon and proceed with the transfer.

marc-hb, 2024-07-16:
https://github.com/pypi/support/issues/2526#issuecomment-2230991181

> Owner role on kconfiglib Project granted to zephyr-project.

carlescufi, 2024-08-14:
I can confirm that the zephyr-project account in PyPi now manages the Kconfiglib project:
https://pypi.org/project/kconfiglib/
[image]

carlescufi, 2024-08-14:
@jackrosenthal can you confirm that you are still interested in maintaining Kconfiglib?

jackrosenthal, 2024-08-14:
Yeah, I'm still available to help maintain kconfiglib!

carlescufi, 2024-08-14:
Concluding on this issue:

- Repo is now available in https://github.com/zephyrproject-rtos/Kconfiglib (@jackrosenthal has write access)
- PyPi package is now owned by Zephyr

TODO: @jackrosenthal can you comment on all PRs and issues in https://github.com/ulfalizer/Kconfiglib redirecting users to https://github.com/zephyrproject-rtos/Kconfiglib?

carlescufi, 2024-08-14:
Closing this issue as this is effectively complete.

carlescufi, 2024-08-14:
Reopening because I now realize we have some commits in Zephyr's own kconfiglib copy (yes, we don't use the PyPi package) that would need to be backported to the new upstream:
https://github.com/zephyrproject-rtos/zephyr/commits/main/scripts/kconfig/kconfiglib.py

carlescufi, 2024-08-17:
Now complete.

closed 2024-08-17 by carlescufi